borglab/gtsam #456: The mechenism in the propagation of covariance.

Gaochao-hit, 2020-08-03:
Recently I run LIO_SAM which uses GTSAm for optimization. And find that  it's weird that the covariance of odometry flutuates which should increse constantly. And the author also don't know the reason, so I asked the question. the link is here [https://github.com/TixiaoShan/LIO-SAM/issues/49]

varunagrawal, 2020-08-11:
Why would the pose covariance be non-decreasing? If you provide sufficient constraints in your problem via measurements, then the covariance can go down since your certainty is going up.

Can you please provide a minimal working example? The code + data.

varunagrawal, 2020-08-11:
Also, for faster answers, I recommend asking this question on the [GTSAM Google Group](https://groups.google.com/forum/#!forum/gtsam-users).

closed 2020-08-11 by varunagrawal

jlblancoc, 2020-08-11:
As proved in [a seminar paper](https://www.semanticscholar.org/paper/A-solution-to-the-simultaneous-localization-and-map-Dissanayake-Newman/d7aa257db5f2cc40e486f677b4378b219727459e), the inferior limit in the covariance of the map is determined by the initial covariance of the pose... which in VIO is zero. Together with the fact that we have *inertial* global observations (acceleration), it *might* be correct to see the covariance of VIO to shrink over time. 
OTOH, it might be a bug in the marginalization of past variables... As @varunagrawal suggested, this is probably better a question for the mailing list.

Gaochao-hit, 2020-08-11:
> As proved in [a seminar paper](https://www.semanticscholar.org/paper/A-solution-to-the-simultaneous-localization-and-map-Dissanayake-Newman/d7aa257db5f2cc40e486f677b4378b219727459e), the inferior [...]

I agree with you. But there seems only prior constraint and  sequencial constraints in the mapoptimization.cpp of [LIO_SAM,](https://github.com/TixiaoShan/LIO-SAM) and no other redundant constraints. I have asked the author of LIO-SAM to give you some feedback.