borglab/gtsam #461: Support for super-fisheye camera (larger than 180 deg of FOV)

acloitre, 2020-08-07:
## Feature
Change the intrinsic coordinate representation of 3D points to support super-fisheye cameras that have field-of-views larger than 180 degrees.

## Motivation
GTSAM currently proposes a camera model capable of handling very large FOV in theory, but not in practice. The Unified model from Mei and Rives (2007) can handle such cameras thanks to its extra parameter, the mirror parameter. This allows to project points that are behind the camera, while also being in their field of view.

However, GTSAM models 3D points as normalized 2D points: X = [x, y, z] becomes X = [x/z, y/z, 1.0], with the 3rd coordinate being dropped. Because of this representation, a super-fisheye camera cannot be modelled in GTSAM, since a point lying at X = [1, 1, 0] would create a normalized 2D point at infinity.

## Pitch
We can keep a normalized 2D representation of a 3D point, by encoding its projection on the camera-centered unit sphere, and by using spherical coordinates (latitude + longitude). Switching from cartesian to spherical coordinates for the representation of the intrinsic coordinates will not increase the dimensionality of the intrinsic coordinates and will make them general to any central camera.

A 3D point X = [x, y, z] would not be normalized as X = [x/z, y/z, 1], but as X = [theta, phi, r = 1] with:

- theta = arccos(z / sqrt(x^2 + y^2 + z^2))
- phi = arctan2(y, x)

## Alternatives
The alternative solution I have used so far, was to create 5 virtual perspective cameras for each of my super-fisheye cameras. These 5 virtual cameras are configured and arranged to exactly cover each side of a cube, except for the 6th face which is placed directly behind the actual camera. This allows to model super-fisheye cameras with FOV up to 270 degrees. Whenever I need to project or lift a feature, I would identify which virtual camera it belonged to, and project/lift the feature using this perspective camera model. 

This solution works in practice, but at the cost of a drastically bigger graph (5 virtual cameras per real camera), and at the cost of added complexity (search for the valid cube face for each feature, and map each measurement on the real image to a virtual measurement on a virtual camera image).

## Final thoughts
I am not sure of how big of an update it would be to make this change in the code. Please let me know if there is an obvious solution to this problem that I am not seeing. I appreciate the work you have put in with GTSAM, and I thank you for making this library open source.

varunagrawal, 2021-01-03:
Hi @acloitre. Apologies for the late response. You'll be happy to know that we recently streamlined all the camera calibration models so that there is less duplication and clutter, so I imagine this feature wouldn't be too big of an issue. Can you please take a look at `Cal3_S2.h` and the related files and start a PR?

Since you already have the equations figured out, it should be straightforward and I am here to help review. :slightly_smiling_face:

acloitre, 2021-01-08:
Hello @varunagrawal ,
Thank you for your reply.
Should I implement the change as two separate functions, e.g. `calibrateSpherical` and `uncalibrateSpherical`?
Or should I modify the existing `calibrate` and `uncalibrate` functions, as well as all the other functions in gtsam that use these functions' outputs?

varunagrawal, 2021-01-08:
You could just create a subclass.

varunagrawal, 2021-01-10:
Would it be possible to simply subclass Cal3Fisheye? That way you'd have very little to do.

dellaert, 2021-01-11:
Actually, please note that there is already a normalized direction class, Unit3. And that several (if not all?) camera models *already* support a Unit3 measurement. I recommend reading the code to inform you of the shortest path here.

varunagrawal, 2021-08-01:
@acloitre any updates on this?

acloitre, 2021-08-02:
Hello @varunagrawal,
My application ended-up not needing a fisheye lens and I did not implement the change listed above. 
I expect this to change in the coming months, and in which case, I will update this post.
Best regards,
Audren

varunagrawal, 2021-08-02:
Fair enough. I'll close this for now, please reopen if and when things change.

closed 2021-08-02 by varunagrawal

acloitre, 2021-10-19:
Hello @varunagrawal,

I am back to implementing a fisheye lens on my application and restarting this issue as I think I got to the core of it.
Cameras in GTSAM are virtually all derived from the class `PinholeBase` defined in the file `CalibratedCamera.h` and implemented in the file `CalibratedCamera.cpp`.
This class has a function `Project` which projects a point to the normalized plane, at `z=1`.
Prof. @dellaert mentioned that a `Unit3` direction vector can be used in place of a 3D point, but both implementation of the `Project` function returns the normalized point `pn = [x/z, y/z]`, which is only defined for `z > 0`.

I will attempt to implement a `SphericalCameraBase` class that uses the spherical coordinate system described in my original post in order to maintain the dimensionality of the measurement to be 2, while allowing to project points at arbitrary locations with respect to the camera.

My question for you is the following: does the above make sense? Or am I still missing an important insight as to how GTSAM handles visual measurements?

Thank you for any help you can provide.

Audren