borglab/gtsam #467: Null-pointer access in NonlinearFactorGraph::saveGraph()

tmoerwald, 2020-08-13:
In contrast to line 232, the const NonlinearFactor::shared_ptr& factor is not checked whether it is null, but accessed here:
https://github.com/borglab/gtsam/blob/de49ef6ff01f46fbece16282b3af268647fce23f/gtsam/nonlinear/NonlinearFactorGraph.cpp#L209

varunagrawal, 2020-08-13:
That seems like an easy PR if you'd like to tackle it @tmoerwald. :slightly_smiling_face:

tmoerwald, 2020-08-14:
I fixed it and tried to push it, but I've got access rights issues:
`git push --set-upstream origin fix/NullPtrNonlinearFactorGraph
ERROR: Permission to borglab/gtsam.git denied to tmoerwald.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.`

Do you have to grant me permission to push?

ProfFan, 2020-08-14:
@tmoerwald You can fork the repo and do a PR?
https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request-from-a-fork

dellaert, 2020-08-17:
@tmoerwald I invited you as a collaborator so you don't have to fork. Just try to push your branch again ;-)

varunagrawal, 2020-08-31:
@tmoerwald any updates?

dellaert, 2020-08-31:
@varunagrawal if easy to fix just push a PR :-)

closed 2020-09-01 by varunagrawal