borglab/gtsam #475: Remove Appveyor

varunagrawal, 2020-08-17:
## Feature

Now that we've moved to Github Actions with a CI for Windows, we need to remove the CI from Appveyor.

Instructions:
 https://help.appveyor.com/discussions/questions/1393-how-to-turn-off-appveyor-for-a-project

## Motivation

Github will keep reporting failure since the appveyor files were removed by #472.

## Pitch

Follow the instructions in the link above. Needs admin access aka @dellaert.

ProfFan, 2020-08-17:
Just remove the app from Settings of the repo, very easy :) @dellaert

varunagrawal, 2020-08-17:
That's literally what the link says :joy:

jlblancoc, 2020-08-17:
Remember to also remove the badges from the README and add the new ones... perhaps that's already in another PR but couldn't see it, sorry if it's already WIP.

varunagrawal, 2020-08-17:
Ah thanks for the reminder @jlblancoc. I just updated #476 to ensure the correct badges.

dellaert, 2020-08-17:
Done

closed 2020-08-17 by dellaert