borglab/gtsam #479: executable bin built by qt creator can't run

Baodh, 2020-08-18:
## Description
I compile gtsam-4.0.3/examples/Pose2SLAMExample.cpp by qt creator, when run executable bin, it shows Segmentation fault (core dumped).

## Steps to reproduce

1.  add new app project in qt creator, select GCC compiler.
2.  copy gtsam-4.0.3/examples/Pose2SLAMExample.cpp to project.
3.  add include and link file in .pro file as follow.
TEMPLATE = app
CONFIG += console c++11
CONFIG -= app_bundle
CONFIG -= qt
SOURCES += Pose2SLAMExample.cpp
INCLUDEPATH += /usr/include
INCLUDEPATH += /usr/local/include
INCLUDEPATH += /usr/include/eigen3
LIBS += -L/usr/lib/x86_64-linux-gnu -ltbb
LIBS += -lboost_thread -lboost_filesystem -lboost_date_time -lboost_system
LIBS += -L/usr/local/lib -lgtsam

4.  compile project and get executable bin
5.  run executable bin and shows Segmentation fault (core dumped).

## Expected behavior
run and print result

## Environment
ubuntu 16.04 + qt creator 4.0.3

## Additional information

ProfFan, 2020-08-18:
Note that GTSAM requires a lot of flags to be defined to be built correctly. If you are using your own build tools (other than CMake), you will need to extract the correct flags manually, possibly `make VERBOSE=1` when building GTSAM. Otherwise it's is guaranteed to fail.

dellaert, 2020-08-19:
We do not support any build systems except those that can be generated by cmake. If you have used cmake to generate QT creator project and something is wrong, we still cannot help you unfortunately, because none of us use QT creator. But we welcome any suggested changes to the cmake files to support QT creator, if you find out what's wrong.

closed 2020-08-19 by dellaert