borglab/gtsam #496: Cannot build matlab toolbox due to to unresolved external symbol error

fpczk, 2020-08-26:
## Description

On Windows 10 / Visual Studio 2015, I cannot build the matlab toolbox. It always fails with the following errors:

```
ShonanAveragingCLI.obj : error LNK2019: unresolved external symbol "public: __cdecl gtsam::ShonanAveraging2::ShonanAveraging2(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,struct gtsam::ShonanAveragingParameters<2> const &)" (??0ShonanAveraging2@gtsam@@QEAA@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBU?$ShonanAveragingParameters@$01@1@@Z) referenced in function main
ShonanAveragingCLI.obj : error LNK2019: unresolved external symbol "public: __cdecl gtsam::ShonanAveraging3::ShonanAveraging3(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,struct gtsam::ShonanAveragingParameters<3> const &)" (??0ShonanAveraging3@gtsam@@QEAA@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBU?$ShonanAveragingParameters@$02@1@@Z) referenced in function main
...
gtsam_wrapper.obj : error LNK2019: unresolved external symbol "public: class Eigen::Matrix<double,-1,1,0,-1,1> __cdecl gtsam::SO<-1>::vec(class gtsam::OptionalJacobian<-1,-1>)const " (?vec@?$SO@$0?0@gtsam@@QEBA?AV?$Matrix@N$0?0$00$0A@$0?0$00@Eigen@@V?$OptionalJacobian@$0?0$0?0@2@@Z) referenced in function "void __cdecl gtsamSOn_vec_303(int,struct mxArray_tag * * const,int,struct mxArray_tag const * * const)" (?gtsamSOn_vec_303@@YAXHQEAPEAUmxArray_tag@@HQEAPEBU1@@Z)
gtsam_wrapper.obj : error LNK2019: unresolved external symbol "public: __cdecl gtsam::ShonanAveraging2::ShonanAveraging2(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,struct gtsam::ShonanAveragingParameters<2> const &)" (??0ShonanAveraging2@gtsam@@QEAA@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBU?$ShonanAveragingParameters@$01@1@@Z) referenced in function "void __cdecl gtsamShonanAveraging2_constructor_2009(int,struct mxArray_tag * * const,int,struct mxArray_tag const * * const)" (?gtsamShonanAveraging2_constructor_2009@@YAXHQEAPEAUmxArray_tag@@HQEAPEBU1@@Z)
gtsam_wrapper.obj : error LNK2019: unresolved external symbol "public: __cdecl gtsam::ShonanAveraging3::ShonanAveraging3(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,struct gtsam::ShonanAveragingParameters<3> const &)" (??0ShonanAveraging3@gtsam@@QEAA@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBU?$ShonanAveragingParameters@$02@1@@Z) referenced in function "void __cdecl gtsamShonanAveraging3_constructor_2036(int,struct mxArray_tag * * const,int,struct mxArray_tag const * * const)" (?gtsamShonanAveraging3_constructor_2036@@YAXHQEAPEAUmxArray_tag@@HQEAPEBU1@@Z)
gtsam_wrapper.obj : error LNK2019: unresolved external symbol "public: __cdecl gtsam::ShonanAveraging3::ShonanAveraging3(class std::vector<class boost::shared_ptr<class gtsam::BetweenFactor<class gtsam::Pose3> >,class std::allocator<class boost::shared_ptr<class gtsam::BetweenFactor<class gtsam::Pose3> > > > const &,struct gtsam::ShonanAveragingParameters<3> const &)" (??0ShonanAveraging3@gtsam@@QEAA@AEBV?$vector@V?$shared_ptr@V?$BetweenFactor@VPose3@gtsam@@@gtsam@@@boost@@V?$allocator@V?$shared_ptr@V?$BetweenFactor@VPose3@gtsam@@@gtsam@@@boost@@@std@@@std@@AEBU?$ShonanAveragingParameters@$02@1@@Z) referenced in function "void __cdecl gtsamShonanAveraging3_constructor_2038(int,struct mxArray_tag * * const,int,struct mxArray_tag const * * const)" (?gtsamShonanAveraging3_constructor_2038@@YAXHQEAPEAUmxArray_tag@@HQEAPEBU1@@Z)
gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::NoiseModelFactor2<class gtsam::SO<4>,class gtsam::SO<4> >::NoiseModelFactor2<class gtsam::SO<4>,class gtsam::SO<4> >(void)" (__imp_??0?$NoiseModelFactor2@V?$SO@$03@gtsam@@V12@@gtsam@@QEAA@XZ) referenced in function "public: __cdecl gtsam::BetweenFactor<class gtsam::SO<4> >::BetweenFactor<class gtsam::SO<4> >(void)" (??0?$BetweenFactor@V?$SO@$03@gtsam@@@gtsam@@QEAA@XZ)
gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::NoiseModelFactor2<class gtsam::SO<4>,class gtsam::SO<4> >::NoiseModelFactor2<class gtsam::SO<4>,class gtsam::SO<4> >(class boost::shared_ptr<class gtsam::noiseModel::Base> const &,unsigned __int64,unsigned __int64)" (__imp_??0?$NoiseModelFactor2@V?$SO@$03@gtsam@@V12@@gtsam@@QEAA@AEBV?$shared_ptr@VBase@noiseModel@gtsam@@@boost@@_K1@Z) referenced in function "void __cdecl gtsamBetweenFactorSO4_constructor_1729(int,struct mxArray_tag * * const,int,struct mxArray_tag const * * const)" (?gtsamBetweenFactorSO4_constructor_1729@@YAXHQEAPEAUmxArray_tag@@HQEAPEBU1@@Z)
gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: virtual __cdecl gtsam::NoiseModelFactor2<class gtsam::SO<4>,class gtsam::SO<4> >::~NoiseModelFactor2<class gtsam::SO<4>,class gtsam::SO<4> >(void)" (__imp_??1?$NoiseModelFactor2@V?$SO@$03@gtsam@@V12@@gtsam@@UEAA@XZ) referenced in function "public: virtual __cdecl gtsam::BetweenFactor<class gtsam::SO<4> >::~BetweenFactor<class gtsam::SO<4> >(void)" (??1?$BetweenFactor@V?$SO@$03@gtsam@@@gtsam@@UEAA@XZ)
gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: unsigned __int64 __cdecl gtsam::NoiseModelFactor2<class gtsam::SO<4>,class gtsam::SO<4> >::key1(void)const " (__imp_?key1@?$NoiseModelFactor2@V?$SO@$03@gtsam@@V12@@gtsam@@QEBA_KXZ) referenced in function "public: virtual void __cdecl gtsam::BetweenFactor<class gtsam::SO<4> >::print(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class boost::function<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl(unsigned __int64)> const &)const " (?print@?$BetweenFactor@V?$SO@$03@gtsam@@@gtsam@@UEBAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBV?$function@$$A6A?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@_K@Z@boost@@@Z)
gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: unsigned __int64 __cdecl gtsam::NoiseModelFactor2<class gtsam::SO<4>,class gtsam::SO<4> >::key2(void)const " (__imp_?key2@?$NoiseModelFactor2@V?$SO@$03@gtsam@@V12@@gtsam@@QEBA_KXZ) referenced in function "public: virtual void __cdecl gtsam::BetweenFactor<class gtsam::SO<4> >::print(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class boost::function<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl(unsigned __int64)> const &)const " (?print@?$BetweenFactor@V?$SO@$03@gtsam@@@gtsam@@UEBAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBV?$function@$$A6A?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@_K@Z@boost@@@Z)
gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::NoiseModelFactor2<class gtsam::SO<4>,class gtsam::SO<4> >::NoiseModelFactor2<class gtsam::SO<4>,class gtsam::SO<4> >(class gtsam::NoiseModelFactor2<class gtsam::SO<4>,class gtsam::SO<4> > const &)" (__imp_??0?$NoiseModelFactor2@V?$SO@$03@gtsam@@V12@@gtsam@@QEAA@AEBV01@@Z) referenced in function "public: __cdecl gtsam::BetweenFactor<class gtsam::SO<4> >::BetweenFactor<class gtsam::SO<4> >(class gtsam::BetweenFactor<class gtsam::SO<4> > const &)" (??0?$BetweenFactor@V?$SO@$03@gtsam@@@gtsam@@QEAA@AEBV01@@Z)
gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::FrobeniusFactor<class gtsam::SO<3> >::FrobeniusFactor<class gtsam::SO<3> >(unsigned __int64,unsigned __int64,class boost::shared_ptr<class gtsam::noiseModel::Base> const &)" (__imp_??0?$FrobeniusFactor@V?$SO@$02@gtsam@@@gtsam@@QEAA@_K0AEBV?$shared_ptr@VBase@noiseModel@gtsam@@@boost@@@Z) referenced in function "void __cdecl gtsamFrobeniusFactorSO3_constructor_1950(int,struct mxArray_tag * * const,int,struct mxArray_tag const * * const)" (?gtsamFrobeniusFactorSO3_constructor_1950@@YAXHQEAPEAUmxArray_tag@@HQEAPEBU1@@Z)
gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: virtual __cdecl gtsam::FrobeniusFactor<class gtsam::SO<3> >::~FrobeniusFactor<class gtsam::SO<3> >(void)" (__imp_??1?$FrobeniusFactor@V?$SO@$02@gtsam@@@gtsam@@UEAA@XZ) referenced in function "public: virtual void * __cdecl gtsam::FrobeniusFactor<class gt [... cut ...]
vs2015_runtime            14.16.27012          hf0eaf9b_3    anaconda
wheel                     0.34.2                   py36_0    anaconda
wincertstore              0.2              py36h7fe50ca_0    anaconda
xz                        5.2.5                h62dcd97_0    anaconda
zlib                      1.2.11           vc14h1cdd9ab_1  [vc14]  anaconda
```

## Additional information

Here are the full build logs from CMake and Visual Studio:

- CMake: https://pastebin.com/raw/UTBj1q7R
- Visual Studio: https://pastebin.com/raw/zdMTvDa8

fpczk, 2020-08-26:
The following change apperantly solves the first 2 errors, while the other ones remain:

```
diff --git a/gtsam/sfm/ShonanAveraging.h b/gtsam/sfm/ShonanAveraging.h
index edd9f33a2..8f52b920b 100644
--- a/gtsam/sfm/ShonanAveraging.h
+++ b/gtsam/sfm/ShonanAveraging.h
@@ -350,7 +350,7 @@ class GTSAM_EXPORT ShonanAveraging {
 // Subclasses for d=2 and d=3 that explicitly instantiate, as well as provide a
 // convenience interface with file access.

-class ShonanAveraging2 : public ShonanAveraging<2> {
+class GTSAM_EXPORT ShonanAveraging2 : public ShonanAveraging<2> {
  public:
   ShonanAveraging2(const Measurements &measurements,
                    const Parameters &parameters = Parameters());
@@ -358,7 +358,7 @@ class ShonanAveraging2 : public ShonanAveraging<2> {
                             const Parameters &parameters = Parameters());
 };

-class ShonanAveraging3 : public ShonanAveraging<3> {
+class GTSAM_EXPORT ShonanAveraging3 : public ShonanAveraging<3> {
  public:
   ShonanAveraging3(const Measurements &measurements,
                    const Parameters &parameters = Parameters());

```

Build log: https://pastebin.com/raw/AqgCZaT6

dellaert, 2020-08-26:
Thanks! I'm not a windows user, but I created a branch fix/windows_warnings just now. Hoping the export in SOn.h fixes the other errors. Give it a try and share build log? Alternatively - if it does not work yet - maybe you could fork and add a few more GTSAM_EXPORT directives to try and fix? Adding @mikesheffler  to this thread as he has more insight into this, usually :-)

dellaert, 2020-08-26:
As an aside: if you don't need Shonan or new python wrapper, you can check out our stable release 4.0.3 instead. Develop is a bit bleeding edge now.

fpczk, 2020-08-27:
Thank you for your reply! Unfortunately, those changes caused new errors that look like the following:

```
4>C:\Users\XYZ\working\gtsam\gtsam\gtsam/geometry/SOn.h(362): error C2908: explicit specialization; 'gtsam::SO<-1> gtsam::LieGroup<gtsam::SO<-1>,-1>::compose(const Class &,gtsam::OptionalJacobian<-1,-1>,gtsam::OptionalJacobian<-1,-1>) const' has already been instantiated
4>          with
4>          [
4>              Class=gtsam::SO<-1>
4>          ] (compiling source file C:\Users\XYZ\working\gtsam\gtsam\gtsam\geometry\CalibratedCamera.cpp)
```

Here is the full log: https://pastebin.com/raw/zfhABibg

As for stable release 4.0.3: I was able to compile it on Windows, but I'm particularly interested in the IMUKittiExampleGPS for Matlab, and that doesn't work in release 4.0.3 due to a type mismatch error (at least on Windows, I couldn't test it yet under Linux). On linux, I was able to compile the current develop branch, and the Matlab example worked, so I assume that the type mismatch bug was fixed; but eventually I'll need to run it on Windows.

mikesheffler, 2020-08-27:
> Thanks! I'm not a windows user, but I created a branch fix/windows_warnings just now. Hoping the export in SOn.h fixes the other errors. Give it a try and share build log? Alternatively - if it does [...]

I haven't tried to build `develop` since the release candidate; I'll try to build it today and see what sort of fireworks I encounter.

ProfFan, 2020-08-27:
@fpapi83 If you are talking about the MATLAB example in unstable, yes I fixed it in develop.

xdtl, 2020-08-28:
I cannot build matlab toolbox on Windows either. I got all the interfaces files and "gtsam_unstable_wrapper.mexw64", but cannot generate "gtsam_wrapper.mexw64". I am on Windows 10, Visual Studio Professional 2019, CMake 3.17 and Boost 1.73.0. Thanks!

```
70>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::NoiseModelFactor2<class gtsam::SO<3>,class gtsam::SO<3> >::NoiseModelFactor2<class gtsam::SO<3>,class gtsam::SO<3> >(void)" (__imp_??0?$NoiseModelFactor2@V?$SO@$02@gtsam@@V12@@gtsam@@QEAA@XZ) referenced in function "void __cdecl gtsamBetweenFactorSO3_string_deserialize_2283(int,struct mxArray_tag * * const,int,struct mxArray_tag const * * const)" (?gtsamBetweenFactorSO3_string_deserialize_2283@@YAXHQEAPEAUmxArray_tag@@HQEAPEBU1@@Z)
70>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::NoiseModelFactor2<class gtsam::SO<3>,class gtsam::SO<3> >::NoiseModelFactor2<class gtsam::SO<3>,class gtsam::SO<3> >(class boost::shared_ptr<class gtsam::noiseModel::Base> const &,unsigned __int64,unsigned __int64)" (__imp_??0?$NoiseModelFactor2@V?$SO@$02@gtsam@@V12@@gtsam@@QEAA@AEBV?$shared_ptr@VBase@noiseModel@gtsam@@@boost@@_K1@Z) referenced in function "public: __cdecl gtsam::FrobeniusFactor<class gtsam::SO<3> >::FrobeniusFactor<class gtsam::SO<3> >(unsigned __int64,unsigned __int64,class boost::shared_ptr<class gtsam::noiseModel::Base> const &)" (??0?$FrobeniusFactor@V?$SO@$02@gtsam@@@gtsam@@QEAA@_K0AEBV?$shared_ptr@VBase@noiseModel@gtsam@@@boost@@@Z)
70>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: virtual __cdecl gtsam::NoiseModelFactor2<class gtsam::SO<3>,class gtsam::SO<3> >::~NoiseModelFactor2<class gtsam::SO<3>,class gtsam::SO<3> >(void)" (__imp_??1?$NoiseModelFactor2@V?$SO@$02@gtsam@@V12@@gtsam@@UEAA@XZ) referenced in function "public: virtual __cdecl gtsam::BetweenFactor<class gtsam::SO<3> >::~BetweenFactor<class gtsam::SO<3> >(void)" (??1?$BetweenFactor@V?$SO@$02@gtsam@@@gtsam@@UEAA@XZ)
70>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: unsigned __int64 __cdecl gtsam::NoiseModelFactor2<class gtsam::SO<3>,class gtsam::SO<3> >::key1(void)const " (__imp_?key1@?$NoiseModelFactor2@V?$SO@$02@gtsam@@V12@@gtsam@@QEBA_KXZ) referenced in function "public: virtual void __cdecl gtsam::BetweenFactor<class gtsam::SO<3> >::print(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class boost::function<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl(unsigned __int64)> const &)const " (?print@?$BetweenFactor@V?$SO@$02@gtsam@@@gtsam@@UEBAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBV?$function@$$A6A?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@_K@Z@boost@@@Z)
70>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: unsigned __int64 __cdecl gtsam::NoiseModelFactor2<class gtsam::SO<3>,class gtsam::SO<3> >::key2(void)const " (__imp_?key2@?$NoiseModelFactor2@V?$SO@$02@gtsam@@V12@@gtsam@@QEBA_KXZ) referenced in function "public: virtual void __cdecl gtsam::BetweenFactor<class gtsam::SO<3> >::print(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class boost::function<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl(unsigned __int64)> const &)const " (?print@?$BetweenFactor@V?$SO@$02@gtsam@@@gtsam@@UEBAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBV?$function@$$A6A?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@_K@Z@boost@@@Z)
70>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::NoiseModelFactor2<class gtsam::SO<3>,class gtsam::SO<3> >::NoiseModelFactor2<class gtsam::SO<3>,class gtsam::SO<3> >(class gtsam::NoiseModelFactor2<class gtsam::SO<3>,class gtsam::SO<3> > const &)" (__imp_??0?$NoiseModelFactor2@V?$SO@$02@gtsam@@V12@@gtsam@@QEAA@AEBV01@@Z) referenced in function "public: virtual class boost::shared_ptr<class gtsam::NonlinearFactor> __cdecl gtsam::BetweenFactor<class gtsam::SO<3> >::clone(void)const " (?clone@?$BetweenFactor@V?$SO@$02@gtsam@@@gtsam@@UEBA?AV?$shared_ptr@VNonlinearFactor@gtsam@@@boost@@XZ)
70>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::NoiseModelFactor2<class gtsam::SO<4>,class gtsam::SO<4> >::NoiseModelFactor2<class gtsam::SO<4>,class gtsam::SO<4> >(void)" (__imp_??0?$NoiseModelFactor2@V?$SO@$03@gtsam@@V12@@gtsam@@QEAA@XZ) referenced in function "public: __cdecl gtsam::BetweenFactor<class gtsam::SO<4> >::BetweenFactor<class gtsam::SO<4> >(void)" (??0?$BetweenFactor@V?$SO@$03@gtsam@@@gtsam@@QEAA@XZ)
70>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::NoiseModelFactor2<class gtsam::SO<4>,class gtsam::SO<4> >::NoiseModelFactor2<class gtsam::SO<4>,class gtsam::SO<4> >(class boost::shared_ptr<class gtsam::noiseModel::Base> const &,unsigned __int64,unsigned __int64)" (__imp_??0?$NoiseModelFactor2@V?$SO@$03@gtsam@@V12@@gtsam@@QEAA@AEBV?$shared_ptr@VBase@noiseModel@gtsam@@@boost@@_K1@Z) referenced in function "public: __cdecl gtsam::FrobeniusFactor<class gtsam::SO<4> >::FrobeniusFactor<class gtsam::SO<4> >(unsigned __int64,unsign [... cut ...] Eigen::Matrix<double,-1,-1,0,-1,-1> > > &>)const " (?unwhitenedError@?$NoiseModelFactor2@V?$SO@$03@gtsam@@V12@@gtsam@@UEBA?AV?$Matrix@N$0?0$00$0A@$0?0$00@Eigen@@AEBVValues@2@V?$optional@AEAV?$vector@V?$Matrix@N$0?0$0?0$0A@$0?0$0?0@Eigen@@V?$allocator@V?$Matrix@N$0?0$0?0$0A@$0?0$0?0@Eigen@@@std@@@std@@@boost@@@Z)
70>...\gtsam\build\gtsam\wrap\gtsam_mex\Release\gtsam_wrapper.mexw64 : fatal error LNK1120: 24 unresolved externals
70>Done building project "gtsam_matlab_wrapper.vcxproj" -- FAILED.
'''

mikesheffler, 2020-08-29:
Hmm. Poaching the two `GTSAM_EXPORT` changes, I get slightly different errors than @fpczk , but I'd wager they're of the same root. My first error is

```
1>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "public: class Eigen::Matrix<double,-1,1,0,-1,1> __cdecl gtsam::SO<-1>::vec(class gtsam::OptionalJacobian<-1,-1>)const " (?vec@?$SO@$0?0@gtsam@@QEBA?AV?$Matrix@N$0?0$00$0A@$0?0$00@Eigen@@V?$OptionalJacobian@$0?0$0?0@2@@Z) referenced in function "void __cdecl gtsamSOn_vec_303(int,struct mxArray_tag * * const,int,struct mxArray_tag const * * const)" (?gtsamSOn_vec_303@@YAXHQEAPEAUmxArray_tag@@HQEAPEBU1@@Z)
```

On my line 281
```
using LieGroup<SO<N>, internal::DimensionSO(N)>::inverse;
```
there is a LINT notice, `invalid base class detected during instantiation of class "gtsam::SO<N> [with N=-1]"` at line 144, which is
```
return SO(SO<Eigen::Dynamic>::Random(rng, N).matrix());
```
BUT, that's part of this block:
```
  /// Random SO(N) element (no big claims about uniformity)
  template <int N_ = N, typename = IsFixed<N_>>
  static SO Random(std::mt19937& rng) {
    // By default, use dynamic implementation above. Specialized for SO(3).
    return SO(SO<Eigen::Dynamic>::Random(rng, N).matrix());
  }
```
looking at the definition for `IsFixed`, I don't think that code should be evaluated when `N` is equal to -1. I don't know exactly where the LINT message is coming from, but the error in my build at the top *also* seems to suggest that MSVC is attempting to instantiate an instance of the class with size -1 and not finding the definition.

If I had to guess, I would say that something about finding the dynamic constructor (maybe even as simple as identifying static vs. dynamic) is not working correctly in MSVC. Why that is ... I don't know. In `testSOn.cpp`, I see some construction tests, but not with `N` equal to -1. Is the wrap the only place that is happening (*i.e.* can Windows not currently handle the `SOn` code, and we're only noticing when compiling the wrap, or is the wrap generating code to be compiled that doesn't make sense)?

dellaert, 2020-08-29:
Hmm. What happens if you explicitly say, to force the compiler's hand?
```
    return SO<N>(SO<Eigen::Dynamic>::Random(rng, N).matrix());
```

mikesheffler, 2020-08-30:
```
1>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "public: class Eigen::Matrix<double,-1,1,0,-1,1> __cdecl gtsam::SO<-1>::vec(class gtsam::OptionalJacobian<-1,-1>)const " (?vec@?$SO@$0?0@gtsam@@QEBA?AV?$Matrix@N$0?0$00$0A@$0?0$00@Eigen@@V?$OptionalJacobian@$0?0$0?0@2@@Z) referenced in function "void __cdecl gtsamSOn_vec_303(int,struct mxArray_tag * * const,int,struct mxArray_tag const * * const)" (?gtsamSOn_vec_303@@YAXHQEAPEAUmxArray_tag@@HQEAPEBU1@@Z)
```
I think it's the same error 🤕