borglab/gtsam #498: Segfault in destructor of PreintegratedCombinedMeasurements

ItsTimmy, 2020-08-28:
This question started on [StackOverflow](https://stackoverflow.com/questions/63619467/gtsam-segfault-possible-alignment-issue), then moved to the [Google group](https://groups.google.com/g/gtsam-users/c/AIbXNwEc0q4), and now I am posting it here, and tagging @ProfFan, per the recommendation from the Google group.

## Description

There seems to be an intermittent segfault in the destructor of `PreintegratedCombinedMeasurements`. I suspect it could be related to alignment, but I don't know for sure.

## Steps to reproduce

Here is a minimum working example, but it may not work on your machine, as I will describe later:

### GraphTest.h

```
#pragma once
#include <gtsam/navigation/CombinedImuFactor.h>
class GraphTest
{
public:
    GraphTest();
    boost::shared_ptr<gtsam::PreintegratedCombinedMeasurements> preint;
};
```

### GraphTest.cpp
```
#include <gtsam_fusion/GraphTest.h>

GraphTest::GraphTest()
{
    auto imuParams = gtsam::PreintegratedCombinedMeasurements::Params::MakeSharedD();
    preint = gtsam::make_shared<gtsam::PreintegratedCombinedMeasurements>(imuParams);
}
```

### Unit test (using gtest)
```
#include <gtsam_fusion/GraphTest.h>
#include <gtest/gtest.h>
#include <iostream>

TEST(GraphManagerTest, test1)
{
    GraphTest eh;
    std::cerr << "GraphTest successfully initialized." << std::endl;  // This DOES display during the test
}

int main(int argc, char *argv[])
{
    testing::InitGoogleTest(&argc, argv);
    return RUN_ALL_TESTS();
}
```
When I run the unit test, `GraphTest successfully initialized` does print, so the segfault is definitely after that. Using a debugger, I think I traced the problem to the destructor of the `preint` shared pointer in `GraphTest`, but I don't have enough experience with debugging to be certain of this.

The strange thing about this problem is that is disappears when I move the implementation of `GraphTest`'s constructor into the header file. While I was removing code to get to this minimum example, the segfault would appear and disappear seemingly at random, even when the code I was removing was not at all related to `preint`.

This, combined with [this mention of a segfault in the docs](https://gtsam-jlblanco-docs.readthedocs.io/en/latest/_static/doxygen/html/namespacegtsam.html#a3ec36279b8211b98545e338ac5371b8c), makes me think that it could be an alignment issue. This would fit what I've seen, in that, depending on how the compiler decides to align `preint` based on other members of `GraphTest`, it could cause the segfault to disappear. However, this is the only evidence I have so far.

This problem may be difficult to reproduce on your machine, as your compiler could align things differently.

I have tried this example with all of the following configurations:

- Using `gtsam::make_shared`
- Using `boost::make_shared`
- Using a `std::shared_ptr` and `std::make_shared`
- Removing the shared pointer altogether and changing the type of `preint` to just `PreintegratedCombinedMeasurements`

In all cases, the problem persists.

## Expected behavior

If alignment is, in fact, the issue, then using `gtsam::make_shared` should prevent it.

## Environment

- _OS_: Ubuntu 18.04.4
- _Compiler_: gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
- _Build Environment_: ROS Melodic, Catkin (Although ROS libraries are not used at all in this example)

dellaert, 2020-08-28:
Excellent issue reporting here! @ProfFan is this something you might easily see?

ProfFan, 2020-08-28:
I see a possibility of an alignment issue here, but as I said in the docs, this is very hard to reproduce. I can try setup a Docker container and see if I can reproduce @ItsTimmy 's environment. If I cannot reproduce, I would probably need some remote access.

ProfFan, 2020-08-28:
@ItsTimmy I need your GTSAM build arguments (cmake), and the CMake file you used to build the test.

dellaert, 2020-08-28:
@ProfFan could you first do a sanity-check in the code that we might have forgotten a gtsam::make_shared inside navigation? Could even be multiple places.

ProfFan, 2020-08-28:
@dellaert No there is none...

ProfFan, 2020-08-28:
I cannot reproduce on a fresh Ubuntu 18.04 Docker image with GTSAM in Release mode.

ItsTimmy, 2020-08-28:
@ProfFan I am using ROS and Catkin. [Here is the CMake file](https://gist.github.com/ItsTimmy/e4d96d52ea4115242f8d9594c79536ae). (I have also created a temporary repo with the whole ROS package [here](https://github.com/ItsTimmy/TemporaryGTSAMDemo)) To build it, I used `catkin build gtsam_fusion --cmake-args -DCMAKE_BUILD_TYPE=Release`. To run the test, I used `make -C build/gtsam_fusion/ run_tests` (From the root of my `catkin_ws`).

I can try to get a Docker image together to reproduce this issue tomorrow.

ItsTimmy, 2020-08-28:
Ok, I got the Docker image ready, at [`scottti/gtsam-bug-demo:latest`](https://hub.docker.com/repository/docker/scottti/gtsam-bug-demo).

To demonstrate the segfault, just open a container with an interactive shell and run the script `RUN_THIS_COMMAND.sh` in the root of the container. To switch to the configuration where the implementation is in the header::

- Comment out everything in `/catkin_ws/src/TemporaryGTSAMDemo/gtsam_fusion/src/gtsam_fusion/GraphTest.cpp`
- Uncomment the constructor implementation in `/catkin_ws/src/TemporaryGTSAMDemo/gtsam_fusion/include/gtsam_fusion/GraphTest.cpp`
- `cd /catkin_ws`
- `catkin build`
- `/RUN_THIS_COMMAND.sh`

I tested this container on a Macbook, and it showed the exact same behavior, so hopefully it should work for you. Let me know if you have any questions.

ProfFan, 2020-08-28:
Eigen version issue?

ItsTimmy, 2020-08-28:
What version of Eigen is required? I have the latest version available through `apt` for Ubuntu 18.04, which is `3.3.4-4`

ProfFan, 2020-08-28:
Oh so you are using a system Eigen? If so you need to also build GTSAM with the system Eigen - I think the flag is `GTSAM_USE_SYSTEM_EIGEN`

ItsTimmy, 2020-08-28:
I tried cleaning and rebuilding everything with `-DGTSAM_USE_SYSTEM_EIGEN=ON`, and have the same results.

ProfFan, 2020-08-28:
I currently have little capacity debugging this, but I think these are the general directions:
1. ditch catkin and build GTSAM separately & build a test project separately, and see if you can reproduce
2. if that works and you cannot reproduce (90% I think)
3. remove find_package(EIGEN) from your catkin pkg cmake, build and see
4. If that does not work, start diagnosing the cmake variables modified by catkin

I would also suggest you enable cmake verbose mode and compare the compiler arguments used to compile the `.o` files.

ItsTimmy, 2020-08-31:
I found that there was a different version of Eigen on my machine in a different install location. `FindEigen3.cmake` was finding this wrong version, leading to an issue.

However, after fixing this, I have now verified that GTSAM and my example are both being built with the same system version of Eigen, and I am seeing the same segfault. I have not yet created a separate project outside of Catkin, but I am working on that now. I'll let you know when I've done that. Thanks for the help!

ItsTimmy, 2020-08-31:
I created a separate project outside of Catkin, and the problem is gone. So it's definitely something that Catkin is misconfiguring. I'll close this issue for now, since it's (probably) not in GTSAM.

closed 2020-08-31 by ItsTimmy

dellaert, 2020-08-31:
Thanks @ItsTimmy for your willingness to investigate !

ItsTimmy, 2020-09-02:
After some more tests, I found that this problem was not present when I abandoned Catkin and just made a basic CMake test project, so the problem must be with Catkin or with my configuration of Catkin.

The default CMakeLists for Catkin includes the following line for each target:

    add_dependencies(${PROJECT_NAME}_lib ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})

I removed this line, and also removed `${catkin_LIBRARIES}` from `target_link_libraries(...)`. This solved the problem.

I don't know why this solved the problem, so I would be interested to hear if anyone has any ideas. If I find out more, I will add it here in case anyone else ever faces a similar issue.

ProfFan, 2020-09-03:
@ItsTimmy Thanks for the precious info :)

I think it is worthwhile to `message()` these variables and paste them here so we can try to figure out what is the problem. Possibly something related to weak symbols, but let's see.

ItsTimmy, 2020-09-03:
So I have not had time to 100% confirm that those changes to my CMakeLists are exactly what solved the problem, so I don't know if it would be worthwhile to spend too much time going through this information. But in case something seems immediately useful to you, here it is:

```
${${PROJECT_NAME}_EXPORTED_TARGETS}:

gtsam_fusion_generate_messages_cpp
gtsam_fusion_generate_messages_eus
gtsam_fusion_generate_messages_lisp
gtsam_fusion_generate_messages_nodejs
gtsam_fusion_generate_messages_py
```

```
${catkin_EXPORTED_TARGETS}:

_catkin_empty_exported_target
roscpp_generate_messages_cpp
roscpp_generate_messages_eus
roscpp_generate_messages_lisp
roscpp_generate_messages_nodejs
roscpp_generate_messages_py
rosgraph_msgs_generate_messages_cpp
rosgraph_msgs_generate_messages_eus
rosgraph_msgs_generate_messages_lisp
rosgraph_msgs_generate_messages_nodejs
rosgraph_msgs_generate_messages_py
std_msgs_generate_messages_cpp
std_msgs_generate_messages_eus
std_msgs_generate_messages_lisp
std_msgs_generate_messages_nodejs
std_msgs_generate_messages_py
std_msgs_generate_messages_cpp
std_msgs_generate_messages_eus
std_msgs_generate_messages_lisp
std_msgs_generate_messages_nodejs
std_msgs_generate_messages_py
std_msgs_generate_messages_cpp
std_msgs_generate_messages_eus
std_msgs_generate_messages_lisp
std_msgs_generate_messages_nodejs
std_msgs_generate_messages_py
sensor_msgs_generate_messages_cpp
sensor_msgs_generate_messages_eus
sensor_msgs_generate_messages_lisp
sensor_msgs_generate_messages_nodejs
sensor_msgs_generate_messages_py
geometry_msgs_generate_messages_cpp
geometry_msgs_generate_messages_eus
geometry_msgs_generate_messages_lisp
geometry_msgs_generate_messages_nodejs
geometry_msgs_generate_messages_py
std_msgs_generate_messages_cpp
std_msgs_generate_messages_eus
std_msgs_generate_messages_lisp
std_msgs_generate_messages_nodejs
std_msgs_generate_messages_py
std_msgs_generate_messages_cpp
std_msgs_generate_messages_eus
std_msgs_generate_messages_lisp
std_msgs_generate_messages_nodejs
std_msgs_generate_messages_py
nav_msgs_generate_messages_cpp
nav_msgs_generate_messages_eus
nav_msgs_generate_messages_lisp
nav_msgs_generate_messages_nodejs
nav_msgs_generate_messages_py
geometry_msgs_generate_messages_cpp
geometry_msgs_generate_messages_eus
geometry_msgs_generate_messages_lisp
geometry_msgs_generate_messages_nodejs
geometry_msgs_generate_messages_py
std_msgs_generate_messages_cpp
std_msgs_generate_messages_eus
std_msgs_generate_messages_lisp
std_msgs_generate_messages_nodejs
std_msgs_generate_messages_py
std_msgs_generate_messages_cpp
std_msgs_generate_messages_eus
std_msgs_generate_messages_lisp
std_msgs_generate_messages_nodejs
std_msgs_generate_messages_py
actionlib_msgs_generate_messages_cpp
actionlib_msgs_generate_messages_eus
actionlib_msgs_generate_messages_lisp
actionlib_msgs_generate_messages_nodejs
actionlib_msgs_generate_messages_py
std_msgs_generate_messages_cpp
std_msgs_generate_messages_eus
std_msgs_generate_messages_lisp
std_msgs_generate_messages_nodejs
std_msgs_generate_messages_py
```

It looks to be all message generation.

And here's `${catkin_LIBRARIES}`, which is probably more relevant:

```
/opt/ros/melodic/lib/libroscpp.so
/usr/lib/x86_64-linux-gnu/libboost_filesystem.so
/opt/ros/melodic/lib/librosconsole.so
/opt/ros/melodic/lib/librosconsole_log4cxx.so
/opt/ros/melodic/lib/librosconsole_backend_interface.so
/usr/lib/x86_64-linux-gnu/liblog4cxx.so
/usr/lib/x86_64-linux-gnu/libboost_regex.so
/opt/ros/melodic/lib/libxmlrpcpp.so
/opt/ros/melodic/lib/libroscpp_serialization.so
/opt/ros/melodic/lib/librostime.so
/opt/ros/melodic/lib/libcpp_common.so
/usr/lib/x86_64-linux-gnu/libboost_system.so
/usr/lib/x86_64-linux-gnu/libboost_thread.so
/usr/lib/x86_64-linux-gnu/libboost_chrono.so
/usr/lib/x86_64-linux-gnu/libboost_date_time.so
/usr/lib/x86_64-linux-gnu/libboost_atomic.so
/usr/lib/x86_64-linux-gnu/libpthread.so
/usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4
```

I did find that I need to link `${catkin_LIBRARIES}` to either my library, or my executable that uses my library (Of course, since `${catkin_LIBRARIES}` includes the ROS functionality I am using). Unfortunately, I have run out of time to look more in depth at this issue, so I can't do any experiments right now to try and isolate which of those libraries is causing the problem. Although if I did, I would start with boost, since the segfaults I was seeing seemed to all be related to `boost::shared_ptr`.