borglab/gtsam #500: FindExampleDataFile in python wrapper does not work

dellaert, 2020-08-30:
## Description

Seems like the path used in gtsam::findExampleDataFile is the one from the python wheel build.

```
ValueError: gtsam::findExampleDataFile could not find a matching file in
/Users/runner/work/gtsam-manylinux-build/gtsam-manylinux-build/gtsam/examples/Data
```

## Steps to reproduce

```python
import gtsam
_ = gtsam.findExampleDataFile("toyExample.g2o")
```

## Expected behavior

Should look in directory where python wheel was installed

## Environment

MacOS, python 3.8

## Additional information

Could be hard to solve

varunagrawal, 2020-08-31:
I'll take care of this. It's some simple setup.py foo.

ProfFan, 2020-08-31:
Should probably be replaced by a Python version

PR #501 (2020-08-31): Example Datasets with Python Package
Updated the `setup.py` to include the example data files and datasets. Resolves #500.

Note that we now install the data to a common "gtsam_example_data" directory, since we need CMake to know the path of the directory when building the `.so` file. This is the only way I've found to break this circular dependency (setup.py depends on .so and .so depends on where setup.py installs the data), so any better suggestions or recommendations are welcome!
closed 2020-08-31 by varunagrawal