borglab/gtsam #508: Bump minimum Boost version

varunagrawal, 2020-09-03:
## Description

Currently, the minimum Boost version required is 1.43. The version required by the tests and Actions CI is 1.69.0. 
Moreover, some tests won't pass with Boost 1.43 due to bugs that have since been fixed in the subsequent versions.

It would be nice to have consistency to avoid unnecessary pain for users.

## Steps to reproduce

1. Install Boost 1.43
2. Compile GTSAM and run tests, sit back and watch things break.

## Expected behavior

All tests should pass and Boost should not be an issue.

## Environment

N/A

## Additional information

ProfFan, 2020-09-03:
Yes I second @chrisbeall. Many ROS systems are still on 16.04, so we should at least support that.

varunagrawal, 2020-09-03:
@chrisbeall @ProfFan okay awesome! Now that we have a consensus, I can create the PR. :slightly_smiling_face:

PR #509 (2020-09-06): Update minimum Boost version required
This PR fixes #508.
closed 2020-09-06 by varunagrawal