borglab/gtsam #515: cmake error

Mr-Xia-6, 2020-09-10:
Please help me,thank you.

99% tests passed, 1 tests failed out of 242

Total Test time (real) =   3.74 sec

The following tests FAILED:
	114 - testSmartProjectionFactor (Failed)
Errors while running CTest
CMakeFiles/check.dir/build.make:57: recipe for target 'CMakeFiles/check' failed
make[3]: *** [CMakeFiles/check] Error 8
CMakeFiles/Makefile2:947: recipe for target 'CMakeFiles/check.dir/all' failed
make[2]: *** [CMakeFiles/check.dir/all] Error 2
CMakeFiles/Makefile2:954: recipe for target 'CMakeFiles/check.dir/rule' failed
make[1]: *** [CMakeFiles/check.dir/rule] Error 2
Makefile:234: recipe for target 'check' failed

dellaert, 2020-09-10:
Please read guidelines when posting an issue

dellaert, 2020-09-10:
BTW you can use GTSAM even if all unit tests do not pass. This is probably a specific class-compiler interaction issue that will not affect anything else. It would be great if you filed a proper issue with platform, compiler, etc and anything else you can find so we can fix it.

closed 2020-09-10 by dellaert