borglab/gtsam #517: Compute Normalized Innovation Error of a Factor

mmmfarrell, 2020-09-10:
I'm sorry if this is not the right forum for this type of question. If there is a better place to ask this question, can you please point me to it.

I am working with a `NonlinearFactorGraph` and want to do my own outlier rejection before adding factors to the graph. Suppose we follow the PoseSLAM example from [the GTSAM tutorials](https://gtsam.org/tutorials/intro.html), but before we add the loop closure constraint we want to check if the constraint is an obvious outlier. I am aware that I can construct a `BetweenFactor` with `BetweenFactor<Pose2>(5, 2, Pose2(2, 0, M_PI_2), model)` and then call `.error()`, `.unwhitenedError()`, or `whitenedError()`. However, none of these functions appear to use the uncertainty in the graph; they only use the noise model from the measurement. I am looking for something that takes into account the uncertainty in the overall system. 

What I am looking for is something similar to the [normalized innovation error squared (NIS)](https://arxiv.org/pdf/1807.08855.pdf) for a Kalman filter. To compute NIS, you only need the residual (given by `unwhitenedError()`) and the innovation covariance (`S = H * P * H^T + R`). The `BetweenFactor` contains the necessary information to provide `H` and `R`, while `P` can come from the overall graph. Is there some function that I can use to compute this?

Is `Marginals.jointMarginalCovariance()` related at all? At first I thought so, but now I am convincing myself that it is not.

mmmfarrell, 2020-09-11:
Here's a rough code snippet that implements the NIS computation described above. Is there a cleaner way to do this?

```
  // Construct and optimize graph
  Values result = LevenbergMarquardtOptimizer(graph, initial).optimize();  
  // Outlier rejection of new loop closure
  auto fac = BetweenFactor<Pose2>(1, 4, Pose2(5.7, 0, 0), model); // loop closure between pose 1 & pose 4
  std::vector<Matrix> Hs;
  Hs.resize(2);
  Vector residual = fac.unwhitenedError(result, Hs);
  Matrix collected_H = gtsam::collect(2, Hs[0], Hs[1]);
  std::vector<Key> keys = {2, 4};
  Marginals marginals(graph, result);
  JointMarginal joint_marg = marginals.jointMarginalCovariance(keys);
  Matrix S = collected_H * joint_marg.fullMatrix() * collected_H.transpose(); // TODO add meas covariance R here
  Matrix Sinv = gtsam::inverse(S);
  auto NIS = residual.transpose() * Sinv * residual;
```

ProfFan, 2020-09-11:
@varunagrawal May have some insights?

varunagrawal, 2020-09-11:
Here's my understanding from reading the linked paper: `H` is the linear measurement model which in this case will come from the `BetweenFactor` definition, `R` is the measurement covariance which you should be providing to the `BetweenFactor` as the noise model, and all you need is `P` which is the marginal covariance of the optimized estimate.

You can get `P` by using `Mariginals(graph, result).marginalCovariance(key)` where `key` is the `Values` key for the particular pose estimate whose covariance you wish to compute.

Since you require `S^(-1)` , you might want to rework the equation to give you the inverse formula directly and use the `marginalInformation` function instead.

varunagrawal, 2020-09-11:
Also for future reference, asking in the Google Group might give you a faster response. :slightly_smiling_face:

mmmfarrell, 2020-09-14:
Awesome, thank you for the response. Yes, using `marginalInformation` makes sense, good idea. 

For future reference, I will point out that if the measurement depends on multiple estimated variables, then `Marginals(graph, result).jointMarginalInformation(keys)` should be used instead of `.marginalInformation(key)` to obtain the off-diagonal terms of the martrix. This is the case for `BetweenFactor`, which depends on two poses.

closed 2020-09-14 by mmmfarrell