borglab/gtsam #519: Matlab wrapper uses python 3

mfinean, 2020-09-11:
Problem:
[ 56%] Generating wrap/gtsam/gtsam_wrapper.cpp
Ignoring classes: ['gtsam::Point2', 'gtsam::Point3']
Traceback (most recent call last):
  File "/path/gtsam/wrap/matlab_wrapper.py", line 1770, in <module>
    cc_content = wrapper.wrap()
  File "/path/gtsam/wrap/matlab_wrapper.py", line 1657, in wrap
    self.wrap_namespace(self.module)
  File "/path/gtsam/wrap/matlab_wrapper.py", line 1096, in wrap_namespace
    self.wrap_namespace(element, namespaces)
  File "/path/gtsam/wrap/matlab_wrapper.py", line 1101, in wrap_namespace
    class_text = self.wrap_instantiated_class(element, "".join(namespace.full_namespaces()))
  File "/path/gtsam/wrap/matlab_wrapper.py", line 1037, in wrap_instantiated_class
    ).splitlines()) + '\n'
  File "/path/gtsam/wrap/matlab_wrapper.py", line 698, in wrap_class_constructors
    methods_wrap = textwrap.indent(textwrap.dedent("""\

To generate the Matlab wrapper it requires python 3.  I fixed by changing "Default" to my python 3 version.

https://github.com/borglab/gtsam/blob/de75367e5e8f49f468474a75e9fd4bdb15fc856e/CMakeLists.txt#L104

ProfFan, 2020-09-11:
Ahh. Should be compatible with Python 2.7... #TODO

varunagrawal, 2020-09-11:
I disagree that it should be compatible with Python 2.x.

Python 2 has [officially reached EOL](https://www.python.org/doc/sunset-python-2/), so I recommend @mfinean upgrade his Python to 3.6+.

varunagrawal, 2020-09-11:
Actually disregard my previous comment. This issue doesn't seem to have anything to do with Python 2.
@mfinean it seems that you haven't pasted the full stack trace and error message. Can you please do so?

mfinean, 2020-09-11:
My apologies that it cut off! First error encountered was:

------------------------------------------------------------------------------------------------------

[ 56%] Generating wrap/gtsam/gtsam_wrapper.cpp
  File "/home/mark/installs/gtsam/wrap/matlab_wrapper.py", line 87
    print(message, file=sys.stderr)
                       ^
SyntaxError: invalid syntax
gtsam/CMakeFiles/gtsam_matlab_wrapper.dir/build.make:76: recipe for target 'gtsam/wrap/gtsam/gtsam_wrapper.cpp' failed
make[2]: *** [gtsam/wrap/gtsam/gtsam_wrapper.cpp] Error 1
CMakeFiles/Makefile2:1188: recipe for target 'gtsam/CMakeFiles/gtsam_matlab_wrapper.dir/all' failed
make[1]: *** [gtsam/CMakeFiles/gtsam_matlab_wrapper.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

------------------------------------------------------------------------------------------------------

This should have been my warning but I used 
`from __future__ import print_function
`
to bypass that. The next error was:

------------------------------------------------------------------------------------------------------

[ 58%] Building CXX object examples/CMakeFiles/Pose3SLAMExampleExpressions_BearingRangeWithTransform.dir/Pose3SLAMExampleExpressions_BearingRangeWithTransform.cpp.o
Ignoring classes: ['gtsam::Point2', 'gtsam::Point3']
Traceback (most recent call last):
  File "/home/mark/installs/gtsam/wrap/matlab_wrapper.py", line 1769, in <module>
    cc_content = wrapper.wrap()
  File "/home/mark/installs/gtsam/wrap/matlab_wrapper.py", line 1656, in wrap
    self.wrap_namespace(self.module)
  File "/home/mark/installs/gtsam/wrap/matlab_wrapper.py", line 1095, in wrap_namespace
    self.wrap_namespace(element, namespaces)
  File "/home/mark/installs/gtsam/wrap/matlab_wrapper.py", line 1100, in wrap_namespace
    class_text = self.wrap_instantiated_class(element, "".join(namespace.full_namespaces()))
  File "/home/mark/installs/gtsam/wrap/matlab_wrapper.py", line 1036, in wrap_instantiated_class
    ).splitlines()) + '\n'
  File "/home/mark/installs/gtsam/wrap/matlab_wrapper.py", line 697, in wrap_class_constructors
    methods_wrap = textwrap.indent(textwrap.dedent("""\
AttributeError: 'module' object has no attribute 'indent'
gtsam/CMakeFiles/gtsam_matlab_wrapper.dir/build.make:76: recipe for target 'gtsam/wrap/gtsam/gtsam_wrapper.cpp' failed
make[2]: *** [gtsam/wrap/gtsam/gtsam_wrapper.cpp] Error 1
CMakeFiles/Makefile2:1188: recipe for target 'gtsam/CMakeFiles/gtsam_matlab_wrapper.dir/all' failed
make[1]: *** [gtsam/CMakeFiles/gtsam_matlab_wrapper.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

------------------------------------------------------------------------------------------------------

I wasn't suggesting for it be made Python 2 compatible but rather that there could be a catch for this to ensure Python 3 is used. In my case, the default was still Python 2.

varunagrawal, 2020-09-11:
Ah thanks a lot! So you're right, in that `Default` picks up Python 2 on your system (I am guessing you are using Ubuntu?). Setting the CMake variable `GTSAM_PYTHON_VERSION` (either via command line or `ccmake`) to `3.6` or whatever version of python you are using, should fix this issue as you have already discovered.

I guess the issue really is that we need to somehow enforce Python 3+ in the wrapper. @ProfFan

varunagrawal, 2020-09-18:
@mfinean #526 uses CMake to find Python3 exclusively. This should solve this issue.

Can you checkout that PR and try it on your local machine to confirm?

varunagrawal, 2020-09-22:
@mfinean PR #526 has been merged into `develop`. Can you please confirm it solves this issue for you?
If not, please feel free to reopen the issue.

PR #526 (2020-09-22): Automatically find Python3 version
Use CMake to identify best Python3 version if `Default` requested.
Note that the CMake code looks specifically for Python3 as we shouldn't be supporting the EOL'd Python2.

This also helps get rid of a bunch of checks for the `Default` value.

Fixes #519 and #522.
closed 2020-09-22 by varunagrawal