borglab/gtsam #521: Is there any reason to code  "-traits<T>::Local(x, prior_);" instead of  "traits<T>::Local(prior_, x);" in PriorFactor.h?

solution3000, 2020-09-13:
PriorFactor.h    
----------------------------------------------------------------------------
/** vector of errors */
    Vector evaluateError(const T& x, boost::optional<Matrix&> H = boost::none) const override {
      if (H) (*H) = Matrix::Identity(traits<T>::GetDimension(x),traits<T>::GetDimension(x));
      // manifold equivalent of z-x -> Local(x,z)
      // TODO(ASL) Add Jacobians.
      return -traits<T>::Local(x, prior_);
    }

Doese the minus sign '-' matter?  very confused!

varunagrawal, 2020-09-13:
From my understanding, I don't believe there is a difference. However, by doing `Local(x, prior_)` we get the negative of `Local(prior_, x)`, and hence the minus sign there. I believe the order matters, but @dellaert should be able to better explain.

dellaert, 2020-09-14:
Yes, it matters, in that it needs to match the derivative. Note that for vector spaces `local(x, prior_)` means 'prior_-x`, and hence the negative is `x -  local_`. with derivative the identity matrix, in the unknown `x`.

Note, maybe GTSAM user group is more appropriate venue, as this is a question rather than a reported issue.

closed 2020-09-14 by dellaert