borglab/gtsam #523: "matlab/gtsam_examples/IMUKittiExampleGPS"  fails to run!

solution3000, 2020-09-14:
[image]

https://github.com/jlblancoc/2020-ual-factor-graphs-course is the matlab toolbox!

varunagrawal, 2021-01-03:
@jlblancoc can you please look at this? Thanks!

ProfFan, 2021-01-03:
Can this be reproduced under Linux? Seems to be an RTTI issue.

mattking-smith, 2021-12-22:
Is this still an open issue? 

I have the current wrap library + GTSAM toolbox on in a Ubuntu 20.04 environment with MATLAB 2021a and I have no issues with this example. 

@solution3000 Can you try wrapping the current GTSAM environment into MATLAB and verify it works on your machine?

varunagrawal, 2021-12-22:
I think this can be marked as resolved. @solution3000 if you still have issues, you can just reopen this issue and provide more details. :slightly_smiling_face:

closed 2021-12-22 by varunagrawal