borglab/gtsam #544: gtsam_unstable/slam/tests/testSerialization.cpp is broken (and not running currently) remove or fix?

asa, 2020-09-29:
Forgive me if I am not able to answer this question elsewhere. What is the philosophy behind working on or retaining code in gtsam_unstable?  For example this test is commented out because it fails, it has headers that are missing / moved etc.  I have run across a few other places in unstable where headers are missing ie: gtsam_unstable/partition/tests/testNestedDissection.cpp:16:#include "SubmapPlanarSLAM.h".   Is the goal to leave unstable as is, or would you welcome some cleanup prs in a few of these cases?  Some of these issues are deeper and would likely call for more thorough culling? 

At the end of the day they are labeled as unstable, so caveat emptor I suppose.

dellaert, 2020-09-29:
Some of the code in unstable is rarely used, but other parts should really be migrated to gtsam. I’d not spend too much time cleaning up for the sake of cleanup, but if you are *using* something and fix bugs/file issues, we would gladly welcome that in addition to having a signal on what code is actually used :-)

dellaert, 2020-09-29:
Closing as answered

closed 2020-09-29 by dellaert