borglab/gtsam #548: VC 2016 compile bug?

dellaert, 2020-10-01:
## Description

It seems VC 2016 fails to complete check after @Alexma3312 ’s PR was merged:
```

Event.cpp
1114
D:\a\gtsam\gtsam\gtsam/geometry/Point3.h(69): error C2491: 'gtsam::mean': definition of dllimport function not allowed (compiling source file D:\a\gtsam\gtsam\gtsam_unstable\geometry\BearingS2.cpp) [D:\a\gtsam\gtsam\build\gtsam_unstable\gtsam_unstable.vcxproj]
1115
D:\a\gtsam\gtsam\gtsam/geometry/Point3.h(69): error C2491: 'gtsam::mean': definition of dllimport function not allowed (compiling source file D:\a\gtsam\gtsam\gtsam_unstable\geometry\Event.cpp) [D:\a\gtsam\gtsam\build\gtsam_unstable\gtsam_unstable.vcxproj]
1116
  Pose3Upright.cpp
```

Alexma3312, 2020-10-01:
https://docs.microsoft.com/en-us/cpp/error-messages/compiler-errors-1/compiler-error-c2491?view=vs-2019

dellaert, 2020-10-01:
I think the issue is the name conflict. I am fixing in a branch now...

ProfFan, 2020-10-02:
Should this be closed?

dellaert, 2020-10-02:
Fixed in #549

closed 2020-10-02 by dellaert