borglab/gtsam #550: Debugging an indeterminant linear exception

aspears1935, 2020-10-01:
## Feature

We have encountered the exception "Indeterminant linear system detected while working near variable XX" pretty consistently across various sensors and designs, especially with larger graphs. It doesn't seem like there is an easy way to debug what is causing this so additional guidance or methods to drill down into the exception would be very helpful in our design and validation process.

## Motivation

This would help our debug and validation process in our designs.

## Pitch

Any notes/guidance on how most users handle and resolve exceptions like this would help us and others in these designs. We encounter this exception even when running some of the example code for longer periods. We have also encountered this exception after about 30mins 10Hz IMU factors with zero accel/gyro values are added to a graph with a strong prior. We also encounter this when adding smart factors from ORBSLAM2 feature tracks. 

## Additional context
Exception: 
Indeterminant linear system detected while working near variable XX

Thrown when a linear system is ill-posed. The most common cause for this
error is having underconstrained variables. Mathematically, the system is
underdetermined. See the GTSAM Doxygen documentation at
http://borg.cc.gatech.edu/ on gtsam::IndeterminantLinearSystemException for
more information.

dellaert, 2020-10-01:
I would love to give better guidance. @gchenfc and I have been thinking about detecting exactly *which* variable is under-determined, and providing better guidance. 

The second point is more about observability over time, and I think is hard to solve in general. I've encountered this in several situations, and it can occur, say, in IMU situations when traveling on a straight line for a while, then hitting a curve. One hardcore solution is to never marginalize and hence use iSAm for all times :-)

duyanwei, 2020-10-22:
@aspears1935 

Some debugging suggestions from my experience:

1. Catch the exception
2. Get the original factor graph and values
3. Linearize the graph to a gaussian factor graph
4. Check its jacobian or hessian matrix, usually the singular values or determinant, it might give you a hint to tune the noise model or find outliers. 

You can even try to optimize the linearized gaussian graph with `gaussian_graph->optimizeDensely()`, hopefully it'll solve the graph without exception. 

The above steps become less efficient when the graph grows bigger, I'm also working to locate the exact variable or measurement that causes the exception.

aryaman-patel, 2023-06-09:
> @aspears1935
>
> Some debugging suggestions from my experience:
>
>     1. Catch the exception
>
>     2. Get the original factor graph and values
>
>     3. Linearize the graph to a gaussian factor graph [...]

Hi! Is it possible for you to sketch a minimal example of how one can do that?