borglab/gtsam #552: Can't build matlab_wrapper

huanhexiao, 2020-10-02:
When compiling gtsam,it stops at "[ 76%] Building CXX object gtsam/CMakeFiles/gtsam_matlab_wrapper.dir/wrap/gtsam/gtsam_wrapper.cpp.o
", the memory has been increasing until overflow and killing by the compiler.

My OS is Ubuntu16.04.

Any help?

jlblancoc, 2020-10-02:
General ideas: replace -g with -g1 if symbols are enabled, or reduce
optimizations to -O1 (in that file), or to split the file in smaller ones
(this only helps sometimes)

fangxu622, 2020-10-15:
@jlblancoc @varunagrawal  hi ~~when build matlab wrapper . it happens some error like this

`
[ 76%] Building CXX object examples/CMakeFiles/Pose2SLAMExampleExpressions.dir/Pose2SLAMExampleExpressions.cpp.o
In file included from /usr/include/boost/smart_ptr/make_shared.hpp:14:0,
                 from /usr/include/boost/make_shared.hpp:14,
                 from /media/fangxu/Segate3T/PHD-Project/gtsam/wrap/matlab.h:40,
                 from /media/fangxu/Segate3T/PHD-Project/gtsam/build/gtsam/wrap/gtsam/gtsam_wrapper.cpp:1:
/usr/include/boost/smart_ptr/make_shared_object.hpp: In instantiation of ‘typename boost::detail::sp_if_not_array<T>::type boost::make_shared(Args&& ...) [with T = std::vector<long unsigned int, tbb::tbb_allocator<long unsigned int> >; Args = {std::vector<long unsigned int, std::allocator<long unsigned int> >}; typename boost::detail::sp_if_not_array<T>::type = boost::shared_ptr<std::vector<long unsigned int, tbb::tbb_allocator<long unsigned int> > >]’:
/media/fangxu/Segate3T/PHD-Project/gtsam/build/gtsam/wrap/gtsam/gtsam_wrapper.cpp:22749:87:   required from here
/usr/include/boost/smart_ptr/make_shared_object.hpp:256:5: error: no matching function for call to ‘std::vector<long unsigned int, tbb::tbb_allocator<long unsigned int> >::vector(std::vector<long unsigned int>)’
     ::new( pv ) T( boost::detail::sp_forward<Args>( args )... );
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/7/vector:64:0,
                 from /usr/include/boost/format.hpp:17,
                 from /media/fangxu/Segate3T/PHD-Project/gtsam/gtsam/base/Matrix.h:31,
                 from /media/fangxu/Segate3T/PHD-Project/gtsam/wrap/matlab.h:26,
                 from /media/fangxu/Segate3T/PHD-Project/gtsam/build/gts
`

fangxu622, 2020-10-17:
have you solved it? @huanhexiao

ProfFan, 2020-10-17:
Since currently no one here is actively using the MATLAB wrapper, you can try use the Python wrapper instead.

fangxu622, 2020-10-18:
> Since currently no one here is actively using the MATLAB wrapper, you can try use the Python wrapper instead.

yeah, python wrapper is ok,  build the matlab wrapper because it has more examples~~

huanhexiao, 2020-10-18:
> have you solved it? @huanhexiao

@fangxu622 
yeah,  it seems like a matter of memory size. When building matlab_wrapper, i close any other applications, and this needs a long time

varunagrawal, 2020-11-04:
I've seen this issue with the python wrapper too unfortunately. This is going to require quite a bit of leg work to figure out. :slightly_frowning_face:

varunagrawal, 2020-11-10:
@fangxu622 can you try compiling with the `GTSAM_WITH_TBB` set to `OFF`?

E.g. `cmake -DGTSAM_WITH_TBB=OFF .. && make -j4`

varunagrawal, 2020-11-16:
@fangxu622 #603 should fix your issue. If you pull the latest `develop` branch, you should be able to compile the Matlab wrapper.

@huanhexiao any updates? Please try with the latest `develop` branch.

varunagrawal, 2020-12-01:
Closing this due to lack of responses. Please feel free to reopen if your issue has not been resolved.

closed 2020-12-01 by varunagrawal