borglab/gtsam #559: Unknown CMake command "gtsamAddExamplesGlob"

gregsensem, 2020-10-08:
I encountered this error : Unknown CMake command "gtsamAddExamplesGlob" when I try to cmake and build examples. Does any one have similar issue? thanks!

CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at CMakeLists.txt:5 (gtsamAddExamplesGlob):
  Unknown CMake command "gtsamAddExamplesGlob".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.16)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/home/peter/Downloads/gtsam-develop/examples/build/CMakeFiles/CMakeOutput.log".

ProfFan, 2020-10-08:
Could be related to #557 ?

Ping @jlblancoc

gregsensem, 2020-10-08:
> Could be related to #557 ?
> 
> Ping @jlblancoc

gregsensem, 2020-10-08:
> Could be related to #557 ?
> 
> Ping @jlblancoc

I think it is not, reverted to previous version the issue is still there. thanks!

closed 2020-10-08 by gregsensem

jlblancoc, 2020-10-09:
@gregsensem Examples are not intended to be used directly by instructing CMake to use their directories as a SOURCE_DIR, I think that's what you are trying to do, right? Build examples by compiling gtsam from its root directory instead.

To bootstrap your own example / program, use as a template this directory instead: `GTSAM/cmake/example_cmake_find_gtsam/`

dellaert, 2020-10-09:
Running an example should just be possible using 'make Blahexample.run` in the build directory.

gregsensem, 2020-10-09:
> @gregsensem Examples are not intended to be used directly by instructing CMake to use their directories as a SOURCE_DIR, I think that's what you are trying to do, right? Build examples by compiling [...]

Thanks Jose! Exactly like you said, I made a mistake to try to build examples from example's directory. Thank you and have a nice weeked! @jlblancoc

gregsensem, 2020-10-09:
> Running an example should just be possible using 'make Blahexample.run` in the build directory.

Thanks Professor! I am reading your "Factor Graphs for Robot Perception" and testing with GTSAM library at the moment.  It is really nice material to learn Factor Graph and really handy library as I am trying to apply pose graph optimization in my visual SLAM project. Thanks for your team's effort for bringing this great work to us. Have a nice weekend! @dellaert