borglab/gtsam #561: Incremental Smoothing with Plane Factor occasionally fails

mcamurri, 2020-10-10:
Hi all,
I'm experiencing some problems with isam2 optimization with plane factors.
Occasionally, the system throws an `gtsam::IndeterminantLinearSystemException` when two planes are present. But if I comment one of them, the system converges.
## Description

Let's consider the following minimal example (or at least as minimal as I could get to reproduce the error).

The code instantiates a prior pose, two planes, and the priors for the planes.

All the values used are perfect: the measurements and the priors have no errors from the ground truth, so the system should converge immediately to the solution. However, with two planes the system throws a `gtsam::IndeterminantLinearSystemException`. If I comment one of the two planes, no exception is thrown.

```

// GTSAM
#include <gtsam/base/numericalDerivative.h>
#include <gtsam/inference/Symbol.h>
#include <gtsam/nonlinear/ISAM2.h>
#include <gtsam/nonlinear/LevenbergMarquardtOptimizer.h>
#include <gtsam/nonlinear/NonlinearFactorGraph.h>
#include <gtsam/nonlinear/Values.h>
#include <gtsam/slam/OrientedPlane3Factor.h>
#include <gtsam/slam/PriorFactor.h>
#include <gtsam/slam/SmartProjectionPoseFactor.h>
#include <gtsam_unstable/nonlinear/IncrementalFixedLagSmoother.h>

using namespace gtsam;

// Create unique keys for each state.
using symbol_shorthand::X; //< Pose3 (x,y,z,r,p,y)
using symbol_shorthand::P; //< Planes

int main(void){
 // Typedefs
  using Plane       = OrientedPlane3;
  using PlaneFactor = gtsam::OrientedPlane3Factor;

  // Setup graph & object
  FastMap<char, Vector> thresholds;
  thresholds['x'] = (Vector(6) << 0.05, 0.05, 0.05, 0.03, 0.03, 0.03).finished();
  thresholds['p'] = Vector3{0.03, 0.03, 0.1};

  ISAM2Params isam2params;
  isam2params.setRelinearizeThreshold(thresholds);
  isam2params.setRelinearizeSkip(3);
  isam2params.findUnusedFactorSlots = true;
  double lag = 1;
  gtsam::IncrementalFixedLagSmoother smoother(lag, isam2params);

  NonlinearFactorGraph graph;
  Values initialEstimate;
  std::map<gtsam::Key, double> timestampMap;

  // Initial values
  Plane p849(0.211098835, 0.214292752, 0.95368543, 26.4269514);
  Plane p897(0.301901811, 0.151751467, 0.941183717, 33.4388229);

  Pose3 x1527(Rot3(0.799903913, -0.564527097,  0.203624376, 0.552537226,   0.82520195,  0.117236322, -0.234214312, 0.0187322547,  0.972004505), Vector3{-91.7500013,-0.47569366,-4.61067606});

  // Setup priors
  initialEstimate.insert(P(849) , p849 );
  initialEstimate.insert(P(897) , p897 );
  initialEstimate.insert(X(1527), x1527);

  // Setup prior factors
  Pose3 x1527_prior(Rot3(0.799903913, -0.564527097,  0.203624376, 0.552537226,   0.82520195,  0.117236322, -0.234214312, 0.0187322547,  0.972004505), Vector3{-91.7500013, -0.47569366, -4.61067606});
  auto x1527_noise = gtsam::noiseModel::Isotropic::Sigma(6, 0.01);

  graph.emplace_shared<PriorFactor<Pose3>>(X(1527), x1527_prior, x1527_noise);

  Plane p849_prior(0.211098835, 0.214292752, 0.95368543, 26.4269514);
  Plane p897_prior(0.301901811, 0.151751467, 0.941183717, 33.4388229);

  auto p849_noise  = gtsam::noiseModel::Diagonal::Sigmas(Vector3{0.785398163, 0.785398163, 5});
  auto p897_noise  = gtsam::noiseModel::Diagonal::Sigmas(Vector3{0.785398163, 0.785398163, 5});

  graph.emplace_shared<PriorFactor<Plane>>(P(849), p849_prior, p849_noise);
  graph.emplace_shared<PriorFactor<Plane>>(P(897), p897_prior, p897_noise);

  Plane p849_meas = Plane(0.0638967294, 0.0755284627, 0.995094297, 2.55956073);
  Plane p897_meas = Plane(0.104902077, -0.0275756528, 0.994100165, 1.32765088);

  // Setup other factors

  const auto x1527_p897_noise = noiseModel::Isotropic::Sigma(3, 0.0322889);

  graph.emplace_shared<PlaneFactor>(p897_meas.planeCoefficients(), x1527_p897_noise, X(1527), P(897));

  const auto x1527_p849_noise = noiseModel::Isotropic::Sigma(3, 0.0451801);

  // OFFENDING LINE BELOW:
  //graph.emplace_shared<PlaneFactor>(p849_meas.planeCoefficients(), x1527_p849_noise, X(1527), P(849));

  // Setup timestamps
  // Update the timestamps of all referenced nodes
  for (const auto & factor: graph) {
    auto keys = factor->keys();
    for (const Key & key: keys) {
      if (timestampMap.count(key) == 0) {
        timestampMap[key] = 156485.0213; // seconds
      }
    }
  }

  // Optimize
  try {
    smoother.update(graph, initialEstimate, timestampMap);
    std::cerr << "ALL OK" << std::endl;
  } catch(gtsam::IndeterminantLinearSystemException e){
    std::cerr << "CAPTURED THE EXCEPTION: " << e.nearbyVariable() << std::endl;
  }

  smoother.getDelta().print();
  smoother.calculateEstimate().print();
}
```

## Steps to reproduce

1. Uncomment the offending line
2. Compile and launch the small main file 

## Expected behavior

The system should converge to the same values of the priors

## Environment
Ubuntu 18.04 equipped with GTSAM 4.0.3 stock.
Please note though that the problem persists even when using the branch that fixes the derivatives by @dwisth [here](https://github.com/borglab/gtsam/pull/362). 

## Additional information

- The two planes being added are not degenerate.
- Changing the thresholds or other parameters for iSAM does not help
- The system converges when using QR decomposition instead of Cholesky (but we want to use Cholesky)
- The exception seems to be thrown from `gtsam/nonlinear/ISAM2Clique.cpp`
- The issue persist when randomly changing the plane coefficients

dellaert, 2020-10-11:
This is a great example! Before o look at it in detail: does the optimization work if you just do gauss-Newton optimization rather than iSAM?

mcamurri, 2020-10-12:
> This is a great example! Before o look at it in detail: does the optimization work if you just do gauss-Newton optimization rather than iSAM?

Yes, exact same behavior with Gauss-Newton. Here is the listing:
```

// GTSAM
#include <gtsam/base/numericalDerivative.h>
#include <gtsam/inference/Symbol.h>
#include <gtsam/nonlinear/ISAM2.h>
#include <gtsam/nonlinear/LevenbergMarquardtOptimizer.h>
#include <gtsam/nonlinear/NonlinearFactorGraph.h>
#include <gtsam/nonlinear/Values.h>
#include <gtsam/slam/OrientedPlane3Factor.h>
#include <gtsam/slam/PriorFactor.h>
#include <gtsam/slam/SmartProjectionPoseFactor.h>
#include <gtsam_unstable/nonlinear/IncrementalFixedLagSmoother.h>
#include <gtsam/nonlinear/GaussNewtonOptimizer.h>

using namespace gtsam;

// Create unique keys for each state.
using symbol_shorthand::X; //< Pose3 (x,y,z,r,p,y)
using symbol_shorthand::P; //< Planes

int main(void){
 // Typedefs
  using Plane       = OrientedPlane3;
  using PlaneFactor = gtsam::OrientedPlane3Factor;

  // Setup graph & object
  FastMap<char, Vector> thresholds;
  thresholds['x'] = (Vector(6) << 0.05, 0.05, 0.05, 0.03, 0.03, 0.03).finished();
  thresholds['p'] = Vector3{0.03, 0.03, 0.1};

  NonlinearFactorGraph graph;
  Values initialEstimate;
  std::map<gtsam::Key, double> timestampMap;

  // Initial values
  Plane p849(0.211098835, 0.214292752, 0.95368543, 26.4269514);
  Plane p897(0.301901811, 0.151751467, 0.941183717, 33.4388229);

  Pose3 x1527(Rot3(0.799903913, -0.564527097,  0.203624376, 0.552537226,   0.82520195,  0.117236322, -0.234214312, 0.0187322547,  0.972004505), Vector3{-91.7500013,-0.47569366,-2.2});

  // Setup priors
  initialEstimate.insert(P(849) , p849 );
  initialEstimate.insert(P(897) , p897 );
  initialEstimate.insert(X(1527), x1527);

  // Setup prior factors
  Pose3 x1527_prior(Rot3(0.799903913, -0.564527097,  0.203624376, 0.552537226,   0.82520195,  0.117236322, -0.234214312, 0.0187322547,  0.972004505), Vector3{-91.7500013, -0.47569366, -2.2});
  auto x1527_noise = gtsam::noiseModel::Isotropic::Sigma(6, 0.01);

  graph.emplace_shared<PriorFactor<Pose3>>(X(1527), x1527_prior, x1527_noise);

  Plane p849_prior(0.211098835, 0.214292752, 0.95368543, 26.4269514);
  Plane p897_prior(0.301901811, 0.151751467, 0.941183717, 33.4388229);

  auto p849_noise  = gtsam::noiseModel::Diagonal::Sigmas(Vector3{0.785398163, 0.785398163, 5});
  auto p897_noise  = gtsam::noiseModel::Diagonal::Sigmas(Vector3{0.785398163, 0.785398163, 5});

  graph.emplace_shared<PriorFactor<Plane>>(P(849), p849_prior, p849_noise);
  graph.emplace_shared<PriorFactor<Plane>>(P(897), p897_prior, p897_noise);

  Plane p849_meas = Plane(0.0638967294, 0.0755284627, 0.995094297, 2.55956073);
  Plane p897_meas = Plane(0.104902077, -0.0275756528, 0.994100165, 1.32765088);

  // Setup other factors

  const auto x1527_p897_noise = noiseModel::Isotropic::Sigma(3, 0.0322889);

  graph.emplace_shared<PlaneFactor>(p897_meas.planeCoefficients(), x1527_p897_noise, X(1527), P(897));

  const auto x1527_p849_noise = noiseModel::Isotropic::Sigma(3, 0.0451801);

  // OFFENDING LINE BELOW:
  graph.emplace_shared<PlaneFactor>(p849_meas.planeCoefficients(), x1527_p849_noise, X(1527), P(849));

  // Setup timestamps
  // Update the timestamps of all referenced nodes
  for (const auto & factor: graph) {
    auto keys = factor->keys();
    for (const Key & key: keys) {
      if (timestampMap.count(key) == 0) {
        timestampMap[key] = 156485.0213; // seconds
      }
    }
  }

  gtsam::GaussNewtonParams params;
  params.setVerbosity("TERMINATION");  // show info about stopping conditions
  gtsam::GaussNewtonOptimizer optimizer(graph, initialEstimate, params);

  // Optimize
  try {
    Values result = optimizer.optimize();
    std::cout << "Optimization complete" << std::endl;

    std::cout << "initial error=" << graph.error(initialEstimate) << std::endl;
    std::cout << "final error=" << graph.error(result) << std::endl;

    std::cerr << "ALL OK" << std::endl;
  } catch(gtsam::IndeterminantLinearSystemException e){
    std::cerr << "CAPTURED THE EXCEPTION: " << e.nearbyVariable() << std::endl;
  }

}
```

dellaert, 2020-10-12:
Marco, I added your test in #564 
There is definitely an issue, but it's weirder than I even imagined. Take a look. Could be a subtle memory issue.
Running out of time but could you try a valgrind on the example?

mcamurri, 2020-10-12:
Here is the output of the second listing (Gauss-Newton) with the offending line through the command:
`valgrind --tool=memcheck --leak-check=yes --show-reachable=yes --num-callers=20 --track-fds=yes -v ./indet_system`

Output:
```
==24449== Memcheck, a memory error detector
==24449== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==24449== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==24449== Command: ./indet_system
==24449== 
--24449-- Valgrind options:
--24449--    --tool=memcheck
--24449--    --leak-check=yes
--24449--    --show-reachable=yes
--24449--    --num-callers=20
--24449--    --track-fds=yes
--24449--    -v
--24449-- Contents of /proc/version:
--24449--   Linux version 5.4.0-47-generic (buildd@lgw01-amd64-038) (gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)) #51~18.04.1-Ubuntu SMP Sat Sep 5 14:35:50 UTC 2020
--24449-- 
--24449-- Arch and hwcaps: AMD64, LittleEndian, amd64-cx16-lzcnt-rdtscp-sse3-avx-avx2-bmi
--24449-- Page sizes: currently 4096, max supported 4096
--24449-- Valgrind library directory: /usr/lib/valgrind
--24449-- Reading syms from /home/mcamurri/git/drs_ros_packages/vilens/vilens/build/devel/lib/vilens/indet_system
--24449-- Reading syms from /lib/x86_64-linux-gnu/ld-2.27.so
--24449--   Considering /lib/x86_64-linux-gnu/ld-2.27.so ..
--24449--   .. CRC mismatch (computed ac9397f6 wanted d0d82632)
--24449--   Considering /usr/lib/debug/lib/x86_64-linux-gnu/ld-2.27.so ..
--24449--   .. CRC is valid
--24449-- Reading syms from /usr/lib/valgrind/memcheck-amd64-linux
--24449--   Considering /usr/lib/valgrind/memcheck-amd64-linux ..
--24449--   .. CRC mismatch (computed 41ddb025 wanted 9972f546)
--24449--    object doesn't have a symbol table
--24449--    object doesn't have a dynamic symbol table
--24449-- Scheduler: using generic scheduler lock implementation.
--24449-- Reading suppressions file: /usr/lib/valgrind/default.supp
==24449== embedded gdbserver: reading from /tmp/vgdb-pipe-from-vgdb-to-24449-by-mcamurri-on-???
==24449== embedded gdbserver: writing to   /tmp/vgdb-pipe-to-vgdb-from-24449-by-mcamurri-on-???
==24449== embedded gdbserver: shared mem   /tmp/vgdb-pipe-shared-mem-vgdb-24449-by-mcamurri-on-???
==24449== 
==24449== TO CONTROL THIS PROCESS USING vgdb (which you probably
==24449== don't want to do, unless you know exactly what you're doing,
==24449== or are doing some strange experiment):
==24449==   /usr/lib/valgrind/../../bin/vgdb --pid=24449 ...command...
==24449== 
==24449== TO DEBUG THIS PROCESS USING GDB: start GDB like this
==24449==   /path/to/gdb ./indet_system
==24449== and then give GDB the following command
==24449==   target remote | /usr/lib/valgrind/../../bin/vgdb --pid=24449
==24449== --pid is optional if only one valgrind process is running
==24449== 
--24449-- REDIR: 0x401f320 (ld-linux-x86-64.so.2:strlen) redirected to 0x580608c1 (???)
--24449-- REDIR: 0x401f100 (ld-linux-x86-64.so.2:index) redirected to 0x580608db (???)
--24449-- Reading syms from /usr/lib/valgrind/vgpreload_core-amd64-linux.so
--24449--   Considering /usr/lib/valgrind/vgpreload_core-amd64-linux.so ..
--24449--   .. CRC mismatch (computed 50df1b30 wanted 4800a4cf)
--24449--    object doesn't have a symbol table
--24449-- Reading syms from /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so
--24449--   Considering /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so ..
--24449--   .. CRC mismatch (computed f893b962 wanted 95ee359e)
--24449--    object doesn't have a symbol table
==24449== WARNING: new redirection conflicts with existing -- ignoring it
--24449--     old: 0x0401f320 (strlen              ) R-> (0000.0) 0x580608c1 ???
--24449--     new: 0x0401f320 (strlen              ) R-> (2007.0) 0x04c32db0 strlen
--24449-- REDIR: 0x401d390 (ld-linux-x86-64.so.2:strcmp) redirected to 0x4c33ee0 (strcmp)
--24449-- REDIR: 0x401f860 (ld-linux-x86-64.so.2:mempcpy) redirected to 0x4c374f0 (mempcpy)
--24449-- Reading syms from /usr/local/lib/libgtsam.so.4.0.3
--24449-- Reading syms from /usr/lib/x86_64-linux-gnu/libtbb.so.2
--24449--    object doesn't have a symbol table
--24449-- Reading syms from /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25
[... 437 lines cut ...]
==24841==    at 0x4C3089F: operator new[](unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==24841==    by 0x54D1752: ??? (in /usr/lib/x86_64-linux-gnu/libtbb.so.2)
==24841==    by 0x54D1A6F: ??? (in /usr/lib/x86_64-linux-gnu/libtbb.so.2)
==24841==    by 0x54CEAFB: ??? (in /usr/lib/x86_64-linux-gnu/libtbb.so.2)
==24841==    by 0x54CD992: ??? (in /usr/lib/x86_64-linux-gnu/libtbb.so.2)
==24841==    by 0x54CBD60: tbb::internal::get_initial_auto_partitioner_divisor() (in /usr/lib/x86_64-linux-gnu/libtbb.so.2)
==24841==    by 0x5190207: gtsam::NonlinearFactorGraph::linearize(gtsam::Values const&) const (in /usr/local/lib/libgtsam.so.4.0.3)
==24841==    by 0x5150AD7: gtsam::GaussNewtonOptimizer::iterate() (in /usr/local/lib/libgtsam.so.4.0.3)
==24841==    by 0x51945DE: gtsam::NonlinearOptimizer::defaultOptimize() (in /usr/local/lib/libgtsam.so.4.0.3)
==24841==    by 0x1158B6: main (in /home/mcamurri/git/drs_ros_packages/vilens/vilens/build/devel/lib/vilens/indet_system)
==24841== 
==24841== LEAK SUMMARY:
==24841==    definitely lost: 0 bytes in 0 blocks
==24841==    indirectly lost: 0 bytes in 0 blocks
==24841==      possibly lost: 608 bytes in 2 blocks
==24841==    still reachable: 7,880 bytes in 6 blocks
[... cut ...]
==24841==                         newarray           : 6,168 bytes in 3 blocks
==24841==         suppressed: 0 bytes in 0 blocks
==24841== 
==24841== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
==24841== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```

Note that I wasn't able to reproduce the success by commenting out both the plane priors. I will try later on with the test you have made. This was done on my small main but it shouldn't make much difference.

dellaert, 2020-10-12:
Hmmm. I'm not an expert on debugging memory errors but don't see crazy things here at first glance. Still, the fact that you can't get error by commenting out prior again points to non-determinism. I'll take another look after teaching but please pore over the code as well?

My bet is on some subtle bug in OrientedPlane3 or OrientedPlane3Factor, and non-determinism points to either uninitialized memory or worse, overwriting some nearby memory. I did use Debug, however, and I think that adds bounds checking in Eigen - but nothing triggered.

ProfFan, 2020-10-12:
TBB is famous for generating false positives with Valgrind. Could you try reproducing without TBB? That would make the stacktrace a bit cleaner.

mcamurri, 2020-10-12:
@ProfFan you are right. I've recompiled gtsam without TBB and run the test @dellaert made. The stack is clear of errors:
```
==5045== Memcheck, a memory error detector
==5045== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==5045== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==5045== Command: ./testOrientedPlane3Factor
==5045== 
--5045-- Valgrind options:
--5045--    --tool=memcheck
--5045--    --leak-check=yes
--5045--    --show-reachable=yes
--5045--    --num-callers=20
--5045--    --track-fds=yes
--5045--    -v
--5045-- Contents of /proc/version:
--5045--   Linux version 5.4.0-47-generic (buildd@lgw01-amd64-038) (gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)) #51~18.04.1-Ubuntu SMP Sat Sep 5 14:35:50 UTC 2020
--5045-- 
--5045-- Arch and hwcaps: AMD64, LittleEndian, amd64-cx16-lzcnt-rdtscp-sse3-avx-avx2-bmi
--5045-- Page sizes: currently 4096, max supported 4096
--5045-- Valgrind library directory: /usr/lib/valgrind
--5045-- Reading syms from /home/mcamurri/git/libraries/gtsam/build/gtsam/slam/tests/testOrientedPlane3Factor
--5045-- Reading syms from /lib/x86_64-linux-gnu/ld-2.27.so
--5045--   Considering /lib/x86_64-linux-gnu/ld-2.27.so ..
--5045--   .. CRC mismatch (computed ac9397f6 wanted d0d82632)
--5045--   Considering /usr/lib/debug/lib/x86_64-linux-gnu/ld-2.27.so ..
--5045--   .. CRC is valid
--5045-- Reading syms from /usr/lib/valgrind/memcheck-amd64-linux
--5045--   Considering /usr/lib/valgrind/memcheck-amd64-linux ..
--5045--   .. CRC mismatch (computed 41ddb025 wanted 9972f546)
--5045--    object doesn't have a symbol table
--5045--    object doesn't have a dynamic symbol table
--5045-- Scheduler: using generic scheduler lock implementation.
--5045-- Reading suppressions file: /usr/lib/valgrind/default.supp
==5045== embedded gdbserver: reading from /tmp/vgdb-pipe-from-vgdb-to-5045-by-mcamurri-on-???
==5045== embedded gdbserver: writing to   /tmp/vgdb-pipe-to-vgdb-from-5045-by-mcamurri-on-???
==5045== embedded gdbserver: shared mem   /tmp/vgdb-pipe-shared-mem-vgdb-5045-by-mcamurri-on-???
==5045== 
==5045== TO CONTROL THIS PROCESS USING vgdb (which you probably
==5045== don't want to do, unless you know exactly what you're doing,
==5045== or are doing some strange experiment):
==5045==   /usr/lib/valgrind/../../bin/vgdb --pid=5045 ...command...
==5045== 
==5045== TO DEBUG THIS PROCESS USING GDB: start GDB like this
==5045==   /path/to/gdb ./testOrientedPlane3Factor
==5045== and then give GDB the following command
==5045==   target remote | /usr/lib/valgrind/../../bin/vgdb --pid=5045
==5045== --pid is optional if only one valgrind process is running
==5045== 
--5045-- REDIR: 0x401f320 (ld-linux-x86-64.so.2:strlen) redirected to 0x580608c1 (???)
--5045-- REDIR: 0x401f100 (ld-linux-x86-64.so.2:index) redirected to 0x580608db (???)
--5045-- Reading syms from /usr/lib/valgrind/vgpreload_core-amd64-linux.so
--5045--   Considering /usr/lib/valgrind/vgpreload_core-amd64-linux.so ..
--5045--   .. CRC mismatch (computed 50df1b30 wanted 4800a4cf)
--5045--    object doesn't have a symbol table
--5045-- Reading syms from /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so
--5045--   Considering /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so ..
--5045--   .. CRC mismatch (computed f893b962 wanted 95ee359e)
--5045--    object doesn't have a symbol table
==5045== WARNING: new redirection conflicts with existing -- ignoring it
--5045--     old: 0x0401f320 (strlen              ) R-> (0000.0) 0x580608c1 ???
--5045--     new: 0x0401f320 (strlen              ) R-> (2007.0) 0x04c32db0 strlen
--5045-- REDIR: 0x401d390 (ld-linux-x86-64.so.2:strcmp) redirected to 0x4c33ee0 (strcmp)
--5045-- REDIR: 0x401f860 (ld-linux-x86-64.so.2:mempcpy) redirected to 0x4c374f0 (mempcpy)
--5045-- Reading syms from /home/mcamurri/git/libraries/gtsam/build/gtsam/libgtsam.so.4.1.0
--5045-- Reading syms from /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25
--5045--    object doesn't have a symbol table
--5045-- Reading syms from /lib/x86_64-linux-gnu/libgcc_s.so.1
--5045--    object doesn't have a symbol table
--5045-- Reading syms from /lib/x86_64-linux-gnu/libc-2.27.so
--5045--   Considering /lib/x86_64-linux-gnu/libc-2.27.so ..
[... 112 lines cut ...]
==5045== Searching for pointers to 1 not-freed blocks
==5045== Checked 335,480 bytes
==5045== 
==5045== 1,552 bytes in 1 blocks are still reachable in loss record 1 of 1
==5045==    at 0x4C30B2F: operator new[](unsigned long, std::nothrow_t const&) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==5045==    by 0x518DCEE: boost::pool<boost::default_user_allocator_new_delete>::malloc_need_resize() (in /home/mcamurri/git/libraries/gtsam/build/gtsam/libgtsam.so.4.1.0)
==5045==    by 0x518C8A8: gtsam::Values::tryInsert(unsigned long, gtsam::Value const&) (in /home/mcamurri/git/libraries/gtsam/build/gtsam/libgtsam.so.4.1.0)
==5045==    by 0x518CABC: gtsam::Values::insert(unsigned long, gtsam::Value const&) (in /home/mcamurri/git/libraries/gtsam/build/gtsam/libgtsam.so.4.1.0)
==5045==    by 0x123137: void gtsam::Values::insert<gtsam::Pose3>(unsigned long, gtsam::Pose3 const&) (in /home/mcamurri/git/libraries/gtsam/build/gtsam/slam/tests/testOrientedPlane3Factor)
==5045==    by 0x11E539: OrientedPlane3Factorlm_translation_errorTest::run(TestResult&) (in /home/mcamurri/git/libraries/gtsam/build/gtsam/slam/tests/testOrientedPlane3Factor)
[... cut ...]
==5045== 
==5045== LEAK SUMMARY:
==5045==    definitely lost: 0 bytes in 0 blocks
==5045==    indirectly lost: 0 bytes in 0 blocks
==5045==      possibly lost: 0 bytes in 0 blocks
==5045==    still reachable: 1,552 bytes in 1 blocks
==5045==         suppressed: 0 bytes in 0 blocks
==5045== 
==5045== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
==5045== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

If I recompile with TBB, then I get 3 errors (which I spare you).

ProfFan, 2020-10-12:
Eigen's Debug mode run time bound checks are not helpful here because the code path is completely different under Release. I would suggest an uninitialized variable (which is the nastiest since both valgrind and ASAN cannot detect). I have work on my hand so I can't investigate so this is a pointer for anyone who's working on this.

dellaert, 2020-10-12:
It fails under Debug, too, though... Maybe that's a hint?

jlblancoc, 2020-10-13:
Running the new test in a debug build (u18.04, g++7) with valgrind gives no suspicious memory access. Neither as is in git, nor uncommenting the p2_prior factor. In that latter case the test failed, of course, but it looks more like a logic or math problem than an uninitialized/corrupted memory issue.

jlblancoc, 2020-10-13:
The augmented information matrix on p1-x0 has **large** (but not absurdly large) negative eigenvalues, btw...

```
eig(I)
ans =

  -2.2134e+01
  -1.3863e+01
  -1.5577e+00
  -6.6390e-13
  -5.6631e-15
   2.1241e-16
   6.6629e-16
   2.2134e+01
   3.5339e+01
   1.9767e+03
```

dellaert, 2020-10-13:
Hmmm. Marco, I propose you print out the singular values of the entire information matrix: it should be positive definite. If not, we should figure out why. Don’t use the augmented part if you use the method @jlblancoc refers to, just the top-left block, omitting the last row and column.

dwisth, 2020-10-18:
Hi Frank,

Thanks again for the feedback.

I added the following code to your example to calculate the singular values (I didn't have permission to push to the branch):
```
Matrix info = graph.linearizeToHessianFactor(initialEstimate)->information();
Eigen::JacobiSVD<Matrix> svd(info, Eigen::ComputeThinU);
std::cout << "Singular Values:\n" << svd.singularValues().transpose() << std::endl;
std::cout << "Matrix U:\n" << svd.matrixU() << std::endl;
```

I also added the `GaussNewtonParams` to the optimizer as you mentioned [here](https://github.com/borglab/gtsam/pull/564#discussion_r503949213).

Please let me know if this isn't the best way to calculate this an I can run it again.

Without the priors on the planes the output is:
```
errorThreshold: 1.34864e-26 <? 0
absoluteDecrease: -6.93331028744e-27 <? 1e-05
relativeDecrease: -1.05802556953 <? 1e-05
iterations: 2 >? 100
Singular Values:
  7227856.57593   3900046.66513    11554.603727   11447.8952301   10004.1893614           10000     9999.992414   9997.42007424   876.885178384   463.629436563  0.117115818303 0.0584303566143
Matrix U:
 3.05749155202e-08  0.000203902807992     0.043944479922  -0.00133561731752   0.00327181364483 -8.13856900573e-14 -1.86582853059e-05 -3.16663230232e-06     0.106971160401     0.993284210451  1.48677042265e-05 -1.61163266599e-06
 0.000150569143325     0.999874039535 -1.04508237136e-05  7.61870498253e-06  1.96616620193e-05 -5.28274821756e-13 -0.000639944543154   -0.0112168372323 -2.19274190229e-05 -0.000202567354402   0.00101435534489   -0.0111617400842
 1.68764240573e-06    0.0112070304694 -1.54012737984e-05 -0.000478990846787  2.14200557276e-05 -2.35713441824e-15 -6.83693024657e-06 -0.000125797549625 -1.13078314566e-05  1.85379398803e-06   -0.0904992302243     0.995833342313
    0.998750564567 -0.000279062238723  -0.00425225577754  6.80738691313e-05  0.000173737506859  5.12382147205e-13  0.000621406264083   -0.0115045141176    0.0468019236689  -0.00485248849205   -0.0114597193946     -0.00104093244
  -0.0472386502513  1.31983933644e-05   -0.0899197346123   0.00123405396733   0.00322719142037  7.02600019382e-14 -4.84575065675e-05   0.00054336790131       0.9895172693    -0.102596268418  0.000540122053905  5.96455652937e-05
   0.0115189560064 -3.21851829849e-06 -1.36086058062e-05 -0.000964913738444 -2.21415804757e-05  5.60199194881e-15  6.83580815522e-06  -0.00013260883655 -5.94154734593e-07 -1.53174371527e-05     0.995829503712    0.0904984175442
  7.7307419102e-05  7.43572915294e-05     0.813574265572      0.56837703658    0.0920197314187 -1.66783091651e-12 -0.000641683245406   0.00681276961955    0.0684733214383   -0.0429068723044  0.000533004251426  0.000333304664153
 0.000107986943198  0.000100831065696    -0.568317890614     0.820638180575   0.00663674256352  9.05842138274e-13 -0.000200109211003   0.00924311059846   -0.0494189235086    0.0315470359615  0.000746807763752  0.000453070406121
-5.16349440162e-06 -1.24301092223e-05   -0.0712522635495   -0.0580178575669     0.995706232728 -4.60585274124e-12  -0.00573296628998   0.00104517090576  -0.00956546816723  0.000824554774567  -2.9987268781e-05 -5.30283529548e-05
  0.00121014453295  0.000717594935947 -1.79246945802e-05  -0.00104230353744  -0.00227821376876     0.925362353338    -0.369407708441    0.0850541215878  7.54385827748e-08  7.23604838035e-08 -1.15573562131e-06 -4.27268452632e-07
 -0.00031795517522  0.000848714700754 -5.85658021976e-06 -7.56605307881e-05    0.0053213985507     0.368867026621     0.929190529936    0.0226479865482  8.05147599797e-08 -2.73422722084e-08  4.01664135017e-07 -4.24897565619e-07
    0.011468542568    0.0111775125185 -0.000214457843987   -0.0113527533003  -0.00166198509197   -0.0874164269234    0.0104300936341     0.995922398508  1.13831276344e-06  6.50610106653e-07 -1.05393644222e-05  -6.3158476992e-06
```

With a prior only on p1:
```
converged
errorThreshold: 0.0630133707 <? 0
absoluteDecrease: 1.08871245352e-13 <? 1e-05
relativeDecrease: 1.72774832075e-12 <? 1e-05
iterations: 3 >? 100
Singular Values:
 7227856.57593  3900048.28587  11554.6068581   11447.895233  10004.1893787          10000  9999.99241466  9997.42027821  876.903800868  465.228801703 0.118107841678 0.097636936477
Matrix U:
  3.0574930447e-08  0.000203902808013    0.0439508961498  -0.00133577499349   0.00327236228026 -9.17320066784e-14 -1.86613148709e-05  -3.1665143594e-06     0.107386723055     0.993239082816  1.48655755096e-05  1.07834713144e-06
 0.000150569216665     0.999874039638 -1.04521337561e-05  7.61875257792e-06  1.96621800382e-05 -8.80834660302e-13 -0.000639995035471   -0.0112168298853  -2.2012160763e-05 -0.000202558084258   0.00300614452676    -0.010797052687
 1.68764285858e-06    0.0112070259271 -1.54008947176e-05  -0.00047899254574  2.14201111227e-05 -6.11941717528e-15 -6.83636036435e-06 -0.000125777584024 -1.13077011445e-05  1.85637452258e-06    -0.268203721303     0.963296901437
    0.998750564567 -0.000279062258366   -0.0042522533701  6.80700019074e-05  0.000173738994954  8.60528572836e-13  0.000621354215108   -0.0115045169203    0.0467998922282  -0.00487204416798   -0.0110853912383  -0.00308590464483
  -0.0472386502513   1.3198394294e-05   -0.0899196835801   0.00123397216971   0.00322720977533  4.31728603128e-14 -4.84550602443e-05  0.000543368296285     0.989474319248    -0.103009714795  0.000520576887842  0.000155857944773
[... cut ...]
   0.0012101442612  0.000717594637112 -1.79219972615e-05  -0.00104230360176   -0.0022782060513     0.925362353198    -0.369407708743    0.0850541220102  7.54642914084e-08  7.23822983471e-08 -1.15782410858e-06 -4.28713338982e-07
-0.000317955103855  0.000848714347112  -5.8563302783e-06 -7.56605503577e-05   0.00532135215213     0.368867026973     0.929190530089    0.0226479854579  8.05407923734e-08 -2.73520097114e-08  4.02370554863e-07 -4.26347525692e-07
   0.0114685399925    0.0111775078629 -0.000214428236363   -0.0113527540638  -0.00166209918326   -0.0874164269195    0.0104300954127     0.995922398378  1.13869475423e-06  6.50799939747e-07  -1.0558491634e-05 -6.33726113327e-06
```

dellaert, 2020-10-19:
OK, so this is interesting, and rather poorly conditioned (both very
high and low singular values). Can you hypothesize why there is such a

discrepancy ? Why are the lowest directions so poorly constrained? What

variables do they correspond to?

—
<https://github.com/borglab/gtsam/issues/561#issuecomment-711238374>, or

auth/ACQHGSINWP4HBLA4P4KLDHTSLMEOXANCNFSM4SLGTCZQ>.

dwisth, 2021-04-05:
Just realised this issue was still open - it was resolved in #680.

closed 2021-04-05 by dwisth