borglab/gtsam #568: LinearFactorContainer rekey()

pshea76, 2020-10-15:
Hi all, 

It appears that the LinearFactorContainer has a broken rekey member function (inherited from NonlinearFactor).  

## Description

Upon construction, the LinearFactorContainer sets its keys attribute (inherited from NonlinearFactor) and stores a shared pointer to the linear factor:
https://github.com/borglab/gtsam/blob/develop/gtsam/nonlinear/LinearContainerFactor.cpp#L31

Then when NonlinearFactor's rekey function is called, the NonlinearFactor keys attribute is modified without also rekeying the stored factor.

[image]

## Steps to reproduce

1. Create a LinearFactorContainer
2. Call its rekey() member function

## Expected behavior

The LinearFactorContainer should rekey its stored factor.

## Environment

C++, Ubuntu 20.04, stock gtsam 4.1 commit: 76a30a208a967634bcd33b530e7e74a5cd6aa7f3

varunagrawal, 2020-11-11:
Hi @pshea76. Can you please provide a minimal reproducible example, possibly a unit test, that I can use to figure this out?

I see you have added a screenshot, so the code snippet that generated that screenshot would also work.

varunagrawal, 2020-12-02:
@pshea76 gentle reminder. This seems to be a valid issue so I want to figure it out asap.

varunagrawal, 2021-04-16:
@pshea76 reminder!

pshea76, 2021-04-20:
Hey Varun,

Sorry for the delay, forgot about this.  The few lines of code below should be enough to recreate the issue.  The issue would be easy to resolve if NonlinearFactor's rekey method was virtual.  Any good reason why it isn't?

Best,
Pat
'

    #include <gtsam/nonlinear/LinearContainerFactor.h>
    #include <gtsam/slam/BetweenFactor.h>
    #include <gtsam/geometry/Point3.h>
    #include <gtsam/inference/Symbol.h>

    #include <glog/logging.h>
    
    // Make an example factor
    auto nonlinear_factor = boost::make_shared<gtsam::BetweenFactor<gtsam::Point3>>(    gtsam::Symbol('x',0),   
                                                                                        gtsam::Symbol('l',0), 
                                                                                        gtsam::Point3(), 
                                                                                        gtsam::noiseModel::Isotropic::Sigma(3,1));
    
    // Linearize and create an LCF
    gtsam::Values linearization_pt;
    linearization_pt.insert(gtsam::Symbol('x',0), gtsam::Point3());
    linearization_pt.insert(gtsam::Symbol('l',0), gtsam::Point3());

    LinearContainerFactor lcf_factor = LinearContainerFactor(nonlinear_factor->linearize(linearization_pt), linearization_pt);

    // Define a key mapping
    std::map<gtsam::Key, gtsam::Key> key_map;
    key_map[gtsam::Symbol('x',0)] = gtsam::Symbol('x',4);
    key_map[gtsam::Symbol('l',0)] = gtsam::Symbol('l',4);

    // Rekey (Calls NonlinearFactor::rekey() which should probably be overriden)
    auto lcf_factor_rekeyed = lcf_factor.rekey(key_map);

    // Print for fun
    lcf_factor.print("lcf_factor");
    lcf_factor_rekeyed->print("lcf_factor_rekeyed");

    // Cast back to LCF ptr
    LinearContainerFactor::shared_ptr lcf_factor_rekey_ptr = boost::dynamic_pointer_cast<LinearContainerFactor>(lcf_factor_rekeyed);
    CHECK(lcf_factor_rekey_ptr);

    // Note the difference here:
    gtsam::PrintKeyVector(lcf_factor_rekeyed->keys());
    gtsam::PrintKeyVector(lcf_factor_rekey_ptr->factor()->keys());

    // For each key
    for(uint i =0; i<lcf_factor_rekeyed->keys().size(); i++){
        // If the nonlinear factor's key is different from the underlying gaussian factor, log error
        if(lcf_factor_rekeyed->keys().at(i) != lcf_factor_rekey_ptr->factor()->keys().at(i) ){
            LOG(ERROR) << "Rekey on LCF is still broken.";
        }
    }

    // For extra fun lets try linearizing this LCF
    gtsam::Values linearization_pt_rekeyed;
    for(auto key_val : linearization_pt){
        linearization_pt_rekeyed.insert(key_map.at(key_val.key), key_val.value);
    }

    lcf_factor_rekeyed->linearize(linearization_pt_rekeyed)->print("relin");    // This fails with the error below

    /** The above throws the error
     * terminate called after throwing an instance of 'gtsam::DynamicValuesMismatched'
     *      what():  The Values 'this' and the argument passed to Values::localCoordinates have mismatched keys and values
     * */
     
'

varunagrawal, 2021-05-30:
@pshea76 awesome, that really helped! I figured out the issue and have opened PR #777 to fix it. Thanks for all your help.

PR #777 (2021-05-31): Fix rekey of LinearContainerFactor
As per #568, `LinearContainerFactor`'s rekey method fails because the `rekey` method doesn't update the keys for the Gaussian Factor and Linearization point properties.

This PR fixes #568 by overriding `rekey` and doing the necessary updates.
closed 2021-05-31 by varunagrawal