borglab/gtsam #574: Building MATLAB toolbox throws Python version "Default"

Cryptum169, 2020-10-26:
## Description
When building Matlab toolbox, throws 
```
CMake Error at cmake/GtsamMatlabWrap.cmake:218 (find_package):
  find_package called with invalid argument "Default"
Call Stack (most recent call first):
  cmake/GtsamMatlabWrap.cmake:68 (wrap_library_internal)
  gtsam/CMakeLists.txt:220 (wrap_and_install_library)
```

## Steps to reproduce

```
mkdir -p build && cd build
cmake -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON .. 
```

## Expected behavior
Shouldn't be throwing error.

## Environment
- Ubuntu 20.04
- gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04) 
- cmake version 3.16.3
- `python -V` outputs Python 3.8.5
## Additional information
None

ProfFan, 2020-10-26:
Could be related to the recent cmake split? I think @varunagrawal has already fixed this issue.

Cryptum169, 2020-10-26:
My bad for not doing enough search, this seems to be quite similar to #522 . I tried building with Python and it works all fine with the following.
```
cmake -DGTSAM_BUILD_PYTHON=ON ..
```
Going through generated cmake file it seems that `GTSAM_PYTHON_VERSION` is initially assigned "Default" in `cmake/HandleGeneralOptions.cmake`, and is then assigned a proper Interpreter version in `cmake/HandlePython.cmake`, but `HandlePython` needs flag `GTSAM_BUILD_PYTHON` to be on,  which would be off when building MATLAB. 

Perhaps it would be nice to add a second flag that's OR of `GTSAM_BUILD_PYTHON` and `GTSAM_INSTALL_MATLAB_TOOLBOX` ? Since it seems both MATLAB toolbox and Python requires Python.

varunagrawal, 2020-11-04:
Ahh that's a good point @Cryptum169, since the wrapper needs Python either way. I'll create the PR later today.

closed 2020-11-10 by dellaert

varunagrawal, 2020-11-10:
@Cryptum169 the latest version of `develop` fixes this issue. Please feel free to reopen this in case it still doesn't work for you.

(Note, you make have to clear your build directory and run CMake from scratch)

PR #584 (2020-11-10): Fix python discovery for MATLAB wrapper
This PR fixes #574