borglab/gtsam #575: tbb Allocator with boost

hz658832, 2020-10-26:
## Description

While compling gtsam (>4.0.x) with:
1. python and pybind11
2. matlab wrapper 
3. tbb
4. boost (1.71)
compiler throws out error "no matching function ..." by gtsam objects which are allocated with tbb instead of stl 

## Steps to reproduce
Just setup with the description.

## Expected behavior
Fix bug in CMakeList, perhaps set(preferred_allocator STL)?

## Environment
OS: Ubuntu 20.04
Matlab R2020b
Python 3.8
libboost 1.71

## Question
Was it because of the version of boost?

varunagrawal, 2020-11-11:
Hi @hz658832. Can you please try with GTSAM 4.1 and see if that works for you?

TBB is seeing a lot of changes and deprecations right now and we're having a hard time keeping up with them.

tmcg0, 2020-11-13:
@hz658832, what version of TBB were you using here?

varunagrawal, 2020-11-16:
This has been fixed by #603. Please pull the latest `develop` branch and try again.

Closing.

varunagrawal, 2020-11-16:
Feel free to re-open in case your issue has not been resolved.

closed 2020-11-16 by varunagrawal