borglab/gtsam #578: How to self-inlcude gtsam ?

YangSiri, 2020-11-03:
Aim: Use specific gtsam in a single project instead of installing it into system;

Description:  I git clone the gtsam code in my _**project/thirdparty**_ folder where i do the build. Instead I manually use the **_make DESTDIR="prjoect/thirdparty/gtsam_installed" install_**. So there will be two folders in _**thirdparty: gtsam; gtsam_installed**_. In my project's **_CMakeList.txt_**, instead of _**find_package(GTSAM required)**_, i set _**include_directories("${PROJECT_SOURCE_DIR}/thirdparty/gtsam_install/usr/local/include")
link_directories("${PROJECT_SOURCE_DIR}/thirdparty/gtsam_install/usr/local/lib")**_. And the cpp file build is fine, but when i run it, SIG problem occured, something like in [https://github.com/borglab/gtsam/issues/75](url). 

Question: How to exactly use gtsam in own project? How to set CMakeList.txt?

YangSiri, 2020-11-09:
Recently I'm using GTSAM with ISAM2 to incremental solve the graph optimizaiton. That is, I build a local smaller graph every time and add it to the whole ISAM graph by _**isam2->update(graph, values)**_. But it got stuck right at this line. It's like it is doing the update but the result never come out. 

Question: What may cause this kind of situation?

Looking forward to your help ;>

jlblancoc, 2020-11-09:
After calling update(), you must call iSAM2->calculateEstimate()
<https://gtsam-jlblanco-docs.readthedocs.io/en/latest/_static/doxygen/html/classgtsam_1_1ISAM2.html#ae12fc71728266a75fa3893f40531f597>
to actually get the estimated values.

JL

YangSiri, 2020-11-09:
Yes, @jlblancoc 

The problem is the program got stuck in the update step and it won't go forword to the next step which may be  _**iSAM2->calculateEstimate()**_

YangSiri, 2020-11-09:
So far it turn out to be some problem with Ceres.

varunagrawal, 2020-11-12:
@YangSiri this seems to be something that you should ask on the Google Group to get a faster response. Closing since this is not a code issue.

closed 2020-11-12 by varunagrawal