borglab/gtsam #581: Add GTSAM citation

varunagrawal, 2020-11-04:
## Feature

Generate a bibtex citation for use in academic publishing.

## Motivation

We use GTSAM for research, and Github repositories get cited all the time these days.

## Pitch

The process is really simple:

1. Link this repo with Zenodo (instructions [here](https://docs.github.com/en/free-pro-team@latest/github/creating-cloning-and-archiving-repositories/referencing-and-citing-content)). This will create DOIs for each new release. Only @dellaert can do this since it requires admin privileges.
2. Create a bibtex citation in the README as per [this tutorial](https://github.com/AASJournals/Tutorials/blob/master/Repositories/CitingRepositories.md).

## Alternatives

NA

## Additional context

NA

dellaert, 2020-11-04:
Feel free to start a draft about GTSAM 4

varunagrawal, 2020-11-07:
If you create the DOI, I can add the badge to the README as well as autogenerate the BibTex via a `curl` command.

varunagrawal, 2020-11-18:
An attempt at humor in this repo. Seemed apt for this issue.

[image]

dellaert, 2020-11-19:
You're gonna have to me walk through what needs to happen next time we meet

varunagrawal, 2021-12-22:
@dellaert we can now finish this. We should do it in our next meeting.

closed 2022-08-09 by varunagrawal