borglab/gtsam #598: pre-4.1 build issue in MATLAB wrapper: possible Boost/TBB incompatibility?

tmcg0, 2020-11-12:
I'm having build issues when building the MATLAB wrapper in the current pre-4.1 develop which I don't have on the 4.0.3 release. The problem occurs only when building with TBB on.

(This may be a system compatibility issue that I need to fix on my personal machine, but I'm documenting it here just in case others have this issue. Perhaps my build flags are misconfigured?)

## Description

In current develop, the following error is thrown when attempting to build the MATLAB wrapper. CMake errors when building file `gtsam/CMakeFiles/gtsam_matlab_wrapper.dir/wrap/gtsam/gtsam_wrapper.cpp.o`
 
`In file included from /usr/include/boost/smart_ptr/make_shared.hpp:14,`
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;`from /usr/include/boost/make_shared.hpp:14,`
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;`from ~/gtsam/wrap/matlab.h:40,`
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;`from ~/gtsam/build/gtsam/wrap/gtsam/gtsam_wrapper.cpp:1:`
`/usr/include/boost/smart_ptr/make_shared_object.hpp: In instantiation of ‘typename boost::detail::sp_if_not_array<T>::type boost::make_shared(Args&& ...) [with T = std::vector<long unsigned int, tbb::tbb_allocator<long unsigned int> >; Args = {std::vector<long unsigned int, std::allocator<long unsigned int> >}; typename boost::detail::sp_if_not_array<T>::type = boost::shared_ptr<std::vector<long unsigned int, tbb::tbb_allocator<long unsigned int> > >]’:`
`~/gtsam/build/gtsam/wrap/gtsam/gtsam_wrapper.cpp:23087:87:   <-- required from here`
**ERROR:** `/usr/include/boost/smart_ptr/make_shared_object.hpp:256:5: error: no matching function for call to ‘std::vector<long unsigned int, tbb::tbb_allocator<long unsigned int> >::vector(std::vector<long unsigned int>)’
  256 |     ::new( pv ) T( boost::detail::sp_forward<Args>( args )... );`

## Steps to reproduce

1. pull [latest GTSAM](https://github.com/borglab/gtsam/commit/d8d0cec0da49c3bee1185a61b180e6638ec81d85) from develop/
2. build with default except `GTSAM_INSTALL_MATLAB_TOOLBOX=ON` (important that `GTSAM_WITH_TBB=ON`)

## Expected behavior

For this to build with no issues. I have no problems with 4.0.3 release.

## Environment

- Ubuntu 20.04.1 LTS
- Boost 1.67
- gcc 9.3.0
- `python = 2.7.18`, `python3 = 3.8.5`
- MATLAB R2020a
- TBB 2020.1-2 amd64

## Additional information

To be clear, when building `GTSAM_WITH_TBB=OFF` this problem doesn't exist. I'm assuming this is an incompatibility between my versions of Boost and TBB, but more investigation is needed.

varunagrawal, 2020-11-13:
It might be something that TBB deprecated which we haven't been on top of due lack of time and support. To be sure, compilation works with the exact same TBB version for 4.0.3?

varunagrawal, 2020-11-13:
This also might be the same issue as #575

tmcg0, 2020-11-13:
> It might be something that TBB deprecated which we haven't been on top of due lack of time and support. To be sure, compilation works with the exact same TBB version for 4.0.3?

Yes, compilation works with the exact same TBB version (2020.1-2) for 4.0.3.

That's the only caveat that makes me think this _might not_ just be a TBB deprecated functionality problem. Something changed between 4.0.3 and 4.1.

varunagrawal, 2020-11-16:
Okay I am able to reproduce this with TBB 4.4 and Matlab 2018. I'll take a closer look at it and get it resolved soon.

Thanks for reporting @tmcg0!

tmcg0, 2020-11-16:
Let me know if you want any help tracking it down. I'm probably not nearly as useful but I can help a little. Maybe something for GTSAM to consider is breaking out TBB versions (and maybe Boost versions?) into the build matrix for CI if it isn't there already.

varunagrawal, 2020-11-16:
Any and all help is appreciated. :slightly_smiling_face:

varunagrawal, 2020-11-16:
I discovered and fixed the issue. As always, it was something trivial.

That being said, we currently don't have a CI in place for Matlab, hence this escaped into the codebase. If you have any suggestions, we'd be all ears. The only fix I have right now is to use a [Docker container](https://github.com/mathworks-ref-arch/matlab-dockerfile) with Github Actions, so if you'd like you can take a stab at it.

tmcg0, 2020-11-16:
So glad it was something trivial! I was really worried it was a full scale TBB version incompatibility.

MATLAB CI would be really useful. I may or may not be the person to pull that feature, but would it be helpful to track MATLAB CI development as an issue with a milestone? Just something to organize contributors' communication on it.

PR #603 (2020-11-16): Use KeyVector for proper wrapping with TBB
This fixes #598. All I had to do was replace `vector<Key>` with `KeyVector` since the latter supports allocator overloading (which is needed for TBB).

@tmcg0
closed 2020-11-16 by varunagrawal