borglab/gtsam #599: Some questions about the FactorGraph class

hunkyu, 2020-11-14:
## Feature

It is found that FactorGraph only has the function of adding factor graphs. How can I remove or modify the added factor node constraints?

## Motivation

Remove or modify the added factor node constraints

## Pitch

I see that the FactorGraph class has remove and erase, but I don't know how to use it yet

## Alternatives

graph_ptr->add(BetweenFactor<Pose3>(gtsam::symbol('x',start_id), gtsam::symbol('x',end_id),pose_from_in.between(pose_to_in), odometryNoise));
graph_ptr->remove(...)

dellaert, 2020-11-14:
Our philosophy is to be functional. Constructors and "add" methods built things that remain constant. All other methods are const. A factor graph is just a vector so building a new graph is O(n). 

We deviate from it a bit in iSAM, for performance reasons.

hunkyu, 2020-11-14:
`  /// @}
  /// @name Modifying Factor Graphs (imperative, discouraged)
  /// @{

  /** non-const STL-style begin() */
  iterator begin() { return factors_.begin(); }

  /** non-const STL-style end() */
  iterator end() { return factors_.end(); }

  /** Directly resize the number of factors in the graph. If the new size is
   * less than the original, factors at the end will be removed.  If the new
   * size is larger than the original, null factors will be appended.
   */
  void resize(size_t size) { factors_.resize(size); }

  /** delete factor without re-arranging indexes by inserting a NULL pointer
   */
  void remove(size_t i) { factors_[i].reset(); }

  /** replace a factor by index */
  void replace(size_t index, sharedFactor factor) { at(index) = factor; }

  /** Erase factor and rearrange other factors to take up the empty space */
  iterator erase(iterator item) { return factors_.erase(item); }

  /** Erase factors and rearrange other factors to take up the empty space */
  iterator erase(iterator first, iterator last) {
    return factors_.erase(first, last);
  }`

Yes, it does! I still want to ask, but I found that call remove(start_id), remove(end_id), update() after execution, and add new add(BetweenFactor(gtsam::symbol('x', start_id)), gtsam::symbol( 'x', end_id)....) A segfault occurred

varunagrawal, 2020-11-16:
@hunkyu can you please share a minimal reproducible example (MRE)? We can help debug with some more context.

varunagrawal, 2020-12-02:
Closing due to lack of response. @hunkyu please reopen if you have any new information.

closed 2020-12-02 by varunagrawal