borglab/gtsam #604: Matlab wrapper not recognizing C++ files

danielmcrook, 2020-11-18:
## Description

After installing GTSAM and running the units of `cmake check`, I added the Matlab folder to paths and tried to run `test_gtsam.m` which returned the error: 
```
Unrecognized function or variable 'Cal3Unified'.

Error in testCal3Unified (line 4)
K = Cal3Unified;

Error in test_gtsam (line 5)
testCal3Unified
```

I went through the [installation steps](https://gtsam.org/build/) independently on MacOS and Ubuntu, so I probably have some fundamental misunderstanding given the same error's emergence on both machines.  Any help?

gchenfc, 2020-11-18:
When you say "still found the Matlab toolbox to be Disabled" do you mean there's a cmake flag that makes you think it's not building the matlab toolbox or do you just mean that `test_gtsam.m` is not working?  Also, I think 
```
cmake GTSAM_INSTALL_MATLAB_TOOLBOX=ON ..
```
should actually be (note the "-D")
```
cmake -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON ..
```
not sure if that was just a typo or not.

Also, which folder are you adding to your matlab path?  It should be something like `<install_path>/gtsam_toolbox/`, where `gtsam_toolbox` includes the following folders:

* `+gtsam`
* `gtsam_tests`
* `gtsam_wrapper.mexmaci64`
* `gtsam_examples`
* `gtsam_unstable_wrapper.mexmaci64` (optional)
* `unstable_examples` (optional)

To know where the install path is, it is printed out in the `cmake ..` command, for example on my computer:
```
-- Installing Matlab Toolbox to /Users/Gerry/GIT_REPOS/gtsam/build_4_1rc/install/gtsam_toolbox
```
And it is also printed out towards the end of the `make install` command, for example:
```
-- Installing: /Users/Gerry/GIT_REPOS/gtsam/build_4_1rc/install/gtsam_toolbox/gtsam_examples/Data/victoria_park.txt
```
If you can't find the folder, then we know it's something in the build process.  If you are using the right folder, then something deeper may be wrong.

If you're still having issues, could you specify your platform (ie Mac, Linux, Windows), which branch you're on (ie develop, master, tagged version), and maybe the output of `cmake ..` ?

danielmcrook, 2020-11-19:
Thanks for the response.  To your final note I'm on Ubuntu and I cloned the master branch.  

I ran the following commands once in the root folder:
```
mkdir build
cd build
cmake ..
cmake -DCMAKE_BUILD_TYPE=debug ..
cmake -DGTSAM_TOOLBOX_INSTALL_PATH:PATH=$HOME/toolbox ..
cmake -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON ..
make check
make install
```

I am adding the folder ` /home/daniel/gtsam/matlab ` to my path. I don't see any folder labeled toolbox, however, this folder does have the contents you listed.

mikesheffler, 2020-11-19:
Hi, @danielmcrook , 

> I cloned the master branch.

This may have been a typo, but you probably want to clone `develop` instead (it is generally treated as `master` on this project, so there is sometimes confusion).

>  however, this folder does have the contents you listed.

You should be in business, or close to it.

Are you sure that your GTSAM installation is functional besides MATLAB? I see that you ran `make install` and not `sudo make install`. I think the default install prefix is `/usr/local`, and `install` shouldn't work without `sudo` for that, so where do your `libgtsam****.so` files (the **** part is a little different for different build profiles) end up? I'm not sure what sort of errors you would see on the MATLAB end if you are calling the wrapper correctly (through `test_gtsam.m`), but the GTSAM 'so's (shared objects) are missing.

The only other thing I can think of is that there there might be a problem with your cmake string. Can you pass all of the variables at once? Like:

```cmake -DCMAKE_BUILD_TYPE=debug -DGTSAM_TOOLBOX_INSTALL_PATH:PATH=$HOME/toolbox  -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON ..```

You could also try using `cmake-gui` (you might have to install it; it's in apt) to have an easy snapshot of what all of the variables read at the same time. If that goes well, I believe `$HOME/toolbox` should be the location where the folders @gchenfc mentioned should show up.

Good luck! :)

gchenfc, 2020-11-19:
As @mikesheffler mentioned, `$HOME/toolbox` is where I would expect the toolbox to be... which I presume would be `/home/daniel/toolbox` in your case, rather than `/home/daniel/gtsam/matlab`

Another possible point of confusion is that the [matlab folder in the root directory of the github repo](https://github.com/borglab/gtsam/tree/develop/matlab) also has many of the same folder names, but is actually _not_ the full installation.  Rather, it's the "source" folder with some tests, examples, and utility functions that later get copied into the "real" installation folder.  To check, the "correct" toolbox folder should have subfolders `+imuBias`, `+noiseModel`, etc inside the `+gtsam` folder:

```
toolbox
├── +gtsam
│   ├── +imuBias
│   ├── +noiseModel
│   │   └── +mEstimator
│   ├── +symbol_shorthand
│   └── +utilities
├── gtsam_examples
│   └── Data
├── gtsam_tests
└── unstable_examples
    └── +imuSimulator
        └── +lib
```

In contrast, the matlab "source" folder (NOT the full matlab installation) doesn't have many of these folders and has significantly fewer files:
```
matlab
├── +gtsam
├── gtsam_examples
├── gtsam_tests
└── unstable_examples
    └── +imuSimulator
        └── +lib
```

varunagrawal, 2020-11-19:
Okay so what is up with this: `-DGTSAM_TOOLBOX_INSTALL_PATH:PATH=$HOME/toolbox ..`? Why is the cmake variable `GTSAM_TOOLBOX_INSTALL_PATH:PATH` rather than `GTSAM_TOOLBOX_INSTALL_PATH`?

I would recommend doing a standard install from the `develop` branch first: 

```sh
cmake -DCMAKE_BUILD_TYPE=debug -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON ..
make -j4
sudo make install
```

and try again.

There's some extra steps you have to take when using custom install locations for GTSAM which are documented [here](https://github.com/borglab/gtsam/blob/653c69fccdcbcd86042915b1eab40f9c53c2d0bc/matlab/README.md).

gchenfc, 2020-11-19:
@varunagrawal I think the `PATH` is the "type" specifier for cmake command line args (see [cmake command line args](https://cmake.org/cmake/help/latest/manual/cmake.1.html#options)) and is acceptable, though optional.

I think this was based on the [INSTALL.md](https://github.com/borglab/gtsam/blob/develop/INSTALL.md) instructions.  Ironically, according to git blame you were supposedly the author 2 years ago 😂  (but this was just because you renamed the file - that line is actually ~9 years old).

Nonetheless, I agree - @danielmcrook you probably don't need a custom install path...  Just make sure to read the cmake output for the line that tells you where the matlab toolbox will be installed to.

danielmcrook, 2020-11-19:
It looks like passing all the `cmake` arguments at once (thanks @mikesheffler ) prompted the toolbox to appear where @gchenfc mentioned it should.  After adding the toolbox folder to my path in Matlab I've been able to run things smoothly, thanks for your help all!  And yes I was heavily leaning on the .md instructions for the ordering and arguments to set things up.  Thanks folks!

closed 2020-11-19 by danielmcrook