borglab/gtsam #606: Error-prone assignable lvalue returned by Values.at<T>()

jlblancoc, 2020-11-19:
## Description

Possible bug in `operator =` of `gtsam::Value` 
I'll write a small test code to reproduce later, but just wanted to leave it reported here as a reminder, since this drove me crazy for a long time.

`Values::at<T>(key)` returns a const ref, but at least in the case of "T=gtsam::Vector", it's **assignable** without any warning or error (!).

## Steps to reproduce

Better example will be provided, this is just a snippet:

```cpp
Values v;
v.insert(...)
v.at<gtsam::Vector>(key) = gtsam::Vector();
// Correct usage would be: v.update(...);
```

## Expected behavior

IMO, the "=" above should not compile, since at() returns `const Value&`. At least, it was confusing to me and didn't give any clue about the value not being actually updating, which should be done via `update()`.

## Environment

gtsam develop branch. Ubuntu 20.04, default g++.

varunagrawal, 2020-11-19:
I think I see the problem here: the templated version of `at` **does not** return a const ref. This is why you're seeing this issue.

Here are the definitions in the header file:

```
template<typename ValueType>
ValueType at(Key j) const;

const Value& at(Key j) const;
```

It seems this was a design choice for convenience. @dellaert thoughts?

dellaert, 2020-11-19:
It should be const! PR?

varunagrawal, 2020-11-19:
On it.

jlblancoc, 2020-11-19:
Thanks Varun!
Sure that looks to be the problem.
Returning a "const T" (no ref) should fix it.

dellaert, 2020-11-19:
Const T will copy, though? Not const ref?

varunagrawal, 2020-11-19:
Const ref is not recommended for Eigen since it gives the dangling pointer error. [StackOverflow](https://stackoverflow.com/a/62919971/1236990)

Another option is to use `Eigen::Ref` but that breaks the generality of `Values`. My expectation is that Eigen's expression templates will make the return efficient by eliminating the temporary variable in the `at` function.

jlblancoc, 2020-11-20:
> Const T will copy, though? Not const ref?

@dellaert Sure, but I saw the original code returned by value for some reason, hence proposing just returning a const value.

I have dig into *why* the templated at<T> returns by value and found the reason: check out Values-inl.h -> internal::handle. 
Returning by copy is done, apparently, with the sole purpose of providing a transparent way for users to request `values.at<T>()` with `T` being Eigen matrices of different types of those actually stored, and still doing the conversion transparently, e.g. Vector -> Vector31, or the other way around, etc. 

Of course, that comes at the cost of always making a copy of either, the whole Eigen::MatrixBase data (for fixed-size) or just of the thin wrapper around the heap memory for dynamic size ones. 

In short: it's not as efficient as it could be, to allow users using the "incorrect" type while accessing `values.at<T>`.

Was that intentional?

If you want the higher-level `template <typename ValueType> const ValueType at(Key j) const;` to return a const ref, we'll have to edit those internal functions to raise an exception when trying to access using the incorrect type. So, this is a backwards-incompatible change, but OTOH will be slightly more efficient.

IMO, let's don't merge #608 until a decision is taken about this.

jlblancoc, 2020-11-20:
PS: A possible alternative might be adding a new `atRef<T>()` which doesn't make any conversion and returns a const ref, and documenting in the .h docs that the existing `at<T>()` performs copies as needed to convert between Eigen types.

jlblancoc, 2020-11-20:
> Const ref is not recommended for Eigen since it gives the dangling pointer error. [StackOverflow](https://stackoverflow.com/a/62919971/1236990)

Sorry I didn't read your linked page until now: of course, we can't just return a reference to the local (on the stack) Eigen matrix, that's why I recommended either returning by value (as now) or throwing an exception if `at<>()` is to always return a const ref and the actual stored data is not identical, since there is no legal way to return a const ref... if it's not referring **to** the actual stored object itself.

dellaert, 2020-11-30:
Sorry about delay in getting back to this. *Is there a compelling reason to change anything at this time?* #608 seems to fix the issue you point here, and indeed, the "at" method is a bit funky to deal with the Eigen types - as I recall vaguely. This might not be as efficient as we want, but I'm not sure it's a bottleneck anywhere.

PR #608 (2020-11-30): Mark Values::at return type as const
Fixes #606
closed 2020-11-30 by varunagrawal