borglab/gtsam #618: Python nonlinearfactorgraph api save via graphviz

Harsharma2308, 2020-11-29:
## Feature
`graph = gtsam.NonlinearFactorGraph()`
`graph.save_graph(format='graphViz')`

## Motivation
Hi visualizing the graph is very important especially if optimization is done incrementally. GraphViz seems very useful for this purpose and I was wondering if python nonlinearfactorgraph could be saved in graphviz format.
## Pitch

For my usecase where I am working on tracking algorithm on a given image/lidar sequence(Kitti), I want to be able to do the following- 

1. Add factors to my graph 
2. Call isam.optimize() at each step when a factor is added.
3. At the end when the sequence finishes visualize the entire graph created.(Also in the middle for debugging purposes)

## Alternatives

varunagrawal, 2020-11-30:
Thanks for the suggestion. Would you be able to create a PR for this?

ProfFan, 2020-11-30:
I remember we have a `dot` output?

dellaert, 2020-11-30:
@Harsharma2308 we actually already have this, right? Please confirm by checking out code that produces .dot file and re-open this issue if that is not what you meant :-)

closed 2020-11-30 by dellaert

Harsharma2308, 2020-12-04:
Sorry could you explain the exact function call? `graph.saveGraph` in Python doesn't exist right now.

varunagrawal, 2020-12-05:
It should be `isam2.saveGraph`.