borglab/gtsam #620: Cannot build with GTSAM4.0.2

LiShuaixin, 2020-11-30:
## Description
Hi, I tried to compiled my own project using `GTSAM 4.0.2`, but failed. The program is compatible with `GTSAM 4.0.0 alpha` which is also the version I've been used. 

It seems like the problem is caused by the difference of eigen version used in the project. Therefore, to ensure the consistency of version of `Eigen`, I install `Eigen 3.3.4` from the source, and recompile all libs (`PCL 1.8` and `GTSAM 4.0.2`) which are dependent on the `Eigen`. Below is the screenshot of cmake values when compiled my project.
[image]
[image]
The confused errors output:

- `/usr/local/include/eigen3/Eigen/src/plugins/CommonCwiseBinaryOps.h:50:1: error: no type named ‘type’ in ‘struct Eigen::internal::promote_scalar_arg<double, gtsam::SO<3>, false>’
`
- `XXXX.hpp:137:32: error: no match for ‘operator*’ (operand types are ‘const SO3 {aka const gtsam::SO<3>}’ and ‘const double’)
`

I also disabled the `march=native` and recompiled gtsam and my project, the problem is not be solved yet.

Could you please help me to figure out why this happend?

varunagrawal, 2020-11-30:
You should use the Eigen version bundled in the project under `3rdparty`.

LiShuaixin, 2020-11-30:
@varunagrawal Thanks for your reply. If I use the `Eigen` version bundled in the `GTSAM`, it could be different from the one used in my system. If that, which version of `Eigen` should I used in my project?

varunagrawal, 2020-11-30:
Definitely try with the version provided by GTSAM. We test against that so it should work no problem.

LiShuaixin, 2020-12-01:
I tried to recompile the `GTSAM` with its bundled `Eigen 3.3.7`, but it makes the difference eigen version problem occurs. Then I changed to use `Eigen 3.3.7` as the system eigen, and recompiled everything, the problem solved. However, I found the current version of `GTSAM` may conflict with some bundled lib of `ROS`. For instance, adding `tf_conversion` in the `catkin package`  (in CMakeLists `find_package(catkin REQUIRED COMPONENTS  tf_conversion`), causes eigen conflict error `/usr/local/include/gtsam/base/Vector.h:73:1: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
`

I think it is caused by the difference of eigen version used in `ROS kinetic` and the one I compiled `GTSAM`.

varunagrawal, 2020-12-27:
That's exactly the issue. We do provide a cmake flag to use the system (e.g. ROS provided) eigen library so you may want to use that.

varunagrawal, 2021-01-02:
@LiShuaixin please reopen this issue if you still have a problem.

closed 2021-01-02 by varunagrawal