borglab/gtsam #627: Problem with IMU preintegration during impacts

raabuchanan, 2020-12-02:
There may be an issue with IMU preintegration for impacts and impulses. Bellow is a link to simulated data from an IMU in a robot's foot as it takes a step. The figure shows the raw acceleration and rotation velocities in the top two plots. The bottom plot is the result of continuous IMU preintegration and the Z velocity is shown in cyan. The velocity should return to zero as the foot makes contact but instead stays negative.

[image]

[Step Dataset](https://drive.google.com/drive/folders/1sFaGuOueWxV4IAz65uJ9IrYTLNa67zja?usp=sharing) 
rosbag and csvs are provided. In the rosbag, the relevant topics are:

- /anymal/RH_FLATFOOT_IMU (IMU signal)
- /kins_ros/propagated_state (IMU preintegration)
  - custom message definition is provided in link

tmcg0, 2020-12-02:
Forgive me if this is a naïve response: but it doesn't look like you have any position reference like a GPS. If you're merely doing pure IMU gyro/accelerometer dead reckoning, velocity drift due to sensor noise is expected, right?

dellaert, 2020-12-02:
Thanks @raabuchanan ! Tim, you are spot on, but it would not drift this far in this short amount of time. We actually asked Russell to create this issue to debug the IMU factor in the face of impacts. 

@raabuchanan I assume you will also create a PR that has the unit test in it?

raabuchanan, 2020-12-02:
@dellaert Yes the unit test is coming

@tmcg0 In addition to Frank's point, this is simulated data without noise. Other kinds of motions without impacts have virtually no drift over several minutes.

tmcg0, 2020-12-02:
@raabuchanan Great point, I totally missed that this was simulated data.

raabuchanan, 2020-12-08:
See PR #633

closed 2021-03-08 by varunagrawal