borglab/gtsam #628: Build Error on Tiger Lake Processor

bjack205, 2020-12-02:
## Description

I get a build error on my computer, which has an 11th-gen Tiger Lake CPU. It looks like the error is related to my cpu architecture.

`make build` output
```
build|develop ⇒ make install
Scanning dependencies of target CppUnitLite
[  0%] Building CXX object CppUnitLite/CMakeFiles/CppUnitLite.dir/Test.cpp.o
[  2%] Building CXX object CppUnitLite/CMakeFiles/CppUnitLite.dir/TestRegistry.cpp.o
[  2%] Building CXX object CppUnitLite/CMakeFiles/CppUnitLite.dir/TestResult.cpp.o
[  2%] Linking CXX static library libCppUnitLite.a
[  2%] Built target CppUnitLite
[ 20%] Built target metis-gtsam
[ 20%] Building C object gtsam/CMakeFiles/gtsam.dir/3rdparty/CCOLAMD/Source/ccolamd.c.o
cc1: error: bad value (‘tigerlake’) for ‘-march=’ switch
cc1: note: valid arguments to ‘-march=’ switch are: nocona core2 nehalem corei7 westmere sandybridge corei7-avx ivybridge core-avx-i haswell core-avx2 broadwell skylake skylake-avx512 cannonlake icelake-client icelake-server cascadelake bonnell atom silvermont slm goldmont goldmont-plus tremont knl knm x86-64 eden-x2 nano nano-1000 nano-2000 nano-3000 nano-x2 eden-x4 nano-x4 k8 k8-sse3 opteron opteron-sse3 athlon64 athlon64-sse3 athlon-fx amdfam10 barcelona bdver1 bdver2 bdver3 bdver4 znver1 znver2 btver1 btver2 native
cc1: error: bad value (‘tigerlake’) for ‘-mtune=’ switch
cc1: note: valid arguments to ‘-mtune=’ switch are: nocona core2 nehalem corei7 westmere sandybridge corei7-avx ivybridge core-avx-i haswell core-avx2 broadwell skylake skylake-avx512 cannonlake icelake-client icelake-server cascadelake bonnell atom silvermont slm goldmont goldmont-plus tremont knl knm intel x86-64 eden-x2 nano nano-1000 nano-2000 nano-3000 nano-x2 eden-x4 nano-x4 k8 k8-sse3 opteron opteron-sse3 athlon64 athlon64-sse3 athlon-fx amdfam10 barcelona bdver1 bdver2 bdver3 bdver4 znver1 znver2 btver1 btver2 generic native
make[2]: *** [gtsam/CMakeFiles/gtsam.dir/build.make:63: gtsam/CMakeFiles/gtsam.dir/3rdparty/CCOLAMD/Source/ccolamd.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3112: gtsam/CMakeFiles/gtsam.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
```

`make check` output
```
build|develop ⇒ make install
Scanning dependencies of target CppUnitLite
[  0%] Building CXX object CppUnitLite/CMakeFiles/CppUnitLite.dir/Test.cpp.o
[  2%] Building CXX object CppUnitLite/CMakeFiles/CppUnitLite.dir/TestRegistry.cpp.o
[  2%] Building CXX object CppUnitLite/CMakeFiles/CppUnitLite.dir/TestResult.cpp.o
[  2%] Linking CXX static library libCppUnitLite.a
[  2%] Built target CppUnitLite
[ 20%] Built target metis-gtsam
[ 20%] Building C object gtsam/CMakeFiles/gtsam.dir/3rdparty/CCOLAMD/Source/ccolamd.c.o
cc1: error: bad value (‘tigerlake’) for ‘-march=’ switch
cc1: note: valid arguments to ‘-march=’ switch are: nocona core2 nehalem corei7 westmere sandybridge corei7-avx ivybridge core-avx-i haswell core-avx2 broadwell skylake skylake-avx512 cannonlake icelake-client icelake-server cascadelake bonnell atom silvermont slm goldmont goldmont-plus tremont knl knm x86-64 eden-x2 nano nano-1000 nano-2000 nano-3000 nano-x2 eden-x4 nano-x4 k8 k8-sse3 opteron opteron-sse3 athlon64 athlon64-sse3 athlon-fx amdfam10 barcelona bdver1 bdver2 bdver3 bdver4 znver1 znver2 btver1 btver2 native
cc1: error: bad value (‘tigerlake’) for ‘-mtune=’ switch
cc1: note: valid arguments to ‘-mtune=’ switch are: nocona core2 nehalem corei7 westmere sandybridge corei7-avx ivybridge core-avx-i haswell core-avx2 broadwell skylake skylake-avx512 cannonlake icelake-client icelake-server cascadelake bonnell atom silvermont slm goldmont goldmont-plus tremont knl knm intel x86-64 eden-x2 nano nano-1000 nano-2000 nano-3000 nano-x2 eden-x4 nano-x4 k8 k8-sse3 opteron opteron-sse3 athlon64 athlon64-sse3 athlon-fx amdfam10 barcelona bdver1 bdver2 bdver3 bdver4 znver1 znver2 btver1 btver2 generic native
make[2]: *** [gtsam/CMakeFiles/gtsam.dir/build.make:63: gtsam/CMakeFiles/gtsam.dir/3rdparty/CCOLAMD/Source/ccolamd.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3112: gtsam/CMakeFiles/gtsam.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
```

## Steps to reproduce

1. Clone the repo
2. Create the `mkdir` folder and `cd` into it
3. Run `make ..`
4. Run `make install`

## Expected behavior

I'd expect to be able to build the repo :) Or at the very least tell me explicitly that my CPU hardware isn't supported yet.

## Environment
OS: Ubuntu 20.04 LTS
Processor: Intel i7-1165G7
Graphics: Mesa Intel Xe Graphics (TGL GT2)
Cmake version: 3.16.3
Boost version: 1.71
GCC version: 9.3.0

## Additional information

varunagrawal, 2020-12-02:
This is a problem with CCOLAMD, not GTSAM. I guess we'll have to check if they have updated their code to work on the new TigerLake processors.

varunagrawal, 2020-12-02:
Actually, this looks like a GCC problem. You will need to use GCC 10 as per the docs: https://gcc.gnu.org/gcc-10/changes.html

closed 2020-12-02 by varunagrawal