borglab/gtsam #634: error: no type named 'library_version_type' in namespace 'boost::serialization'

DYFeng, 2020-12-09:
## Description

Compile on macOS failed.

## Steps to reproduce

1.  git checkout develop
2. cmake .. && make

```
In file included from /Users/dyfeng/Projects/gtsam/gtsam/inference/BayesTree.cpp:21:
In file included from /Users/dyfeng/Projects/gtsam/gtsam/inference/BayesTree.h:22:
In file included from /Users/dyfeng/Projects/gtsam/gtsam/inference/Key.h:20:
In file included from /Users/dyfeng/Projects/gtsam/gtsam/base/FastList.h:25:
/usr/local/include/boost/serialization/list.hpp:53:33: error: no type named 'library_version_type' in namespace 'boost::serialization'; did you mean
      'item_version_type'?
    const boost::serialization::library_version_type library_version(
          ~~~~~~~~~~~~~~~~~~~~~~^
```

## Expected behavior

## Environment

OS: macOS
Apple clang version 12.0.0 (clang-1200.0.32.2)
Target: x86_64-apple-darwin20.1.0
boost: 1.74.0

## Additional information

varunagrawal, 2020-12-11:
So this is actually an issue with Boost 1.74. I encountered this issue today morning and found this: https://github.com/boostorg/serialization/issues/217#issuecomment-737284894 

@DYFeng can you please downgrade to Boost 1.73?

closed 2020-12-12 by DYFeng

wolfv, 2021-03-11:
Adding 

```
#include <boost/serialization/version.hpp>
```

to LossFunctions.h and SubGraphBuilder.h fixes this issue.

Yuanzhuangz, 2021-11-01:
> Adding
> 
> ```
> #include <boost/serialization/version.hpp>
> ```
> 
> to LossFunctions.h and SubGraphBuilder.h fixes this issue.

why can't i find LossFunction.h,where is it?

wwtinwhu, 2024-01-27:
> Adding
> 
> ```
> #include <boost/serialization/version.hpp>
> ```
> 
> to LossFunctions.h and SubGraphBuilder.h fixes this issue.

Hello, after I adding it, this issue still not solved.  Any advice? Thanks! @DYFeng @wolfv @varunagrawal 
### Environment
OS: ubuntu22.04
Compiler: gcc 11.4
boost: 1.74
gtsam version: 4.0.3

wwtinwhu, 2024-01-28:
> Hello, after I adding it, this issue still not solved. Any advice? Thanks! @DYFeng @wolfv @varunagrawal
>
> ### Environment
> OS: ubuntu22.04 Compiler: gcc 11.4 boost: 1.74 gtsam version: 4.0.3

Solution: Hug with latest GTSAM4.2.0! @dellaert