borglab/gtsam #637: Ask for Matlab version(MEX file '/usr/local/gtsam_toolbox/gtsam_wrapper.mexa64' invalid , undefined symbol "some detailed functions")

daidluvkb, 2020-12-18:
Hello,
I am trying to build gpmp2, it required gtsam>=4.0.
I have successfully compiled the two library with ubuntu18.04, gcc-7.5(lower-version gcc has compile errs).But when executing the matlab scripts, it reports message like"MEX file '/usr/local/gtsam_toolbox/gtsam_wrapper.mexa64' invalid , undefined symbol "some detailed functions"". I have read the readme like[this](https://github.com/borglab/gtsam/blob/develop/matlab/README.md).And I have followed the steps, but it didn't work. The error I have met seems like some thing wrong with compiler, and the executing process can not find the function in shared lib.
I'm using matlab r2018b\gcc-7.5\ubuntu18.04. Is there any version problem in my choice?
Thanks in advance!

varunagrawal, 2020-12-23:
When you say Matlab scripts, do you mean the ones in GPMP2? It seems like this is an issue with GPMP2 showing its age and not being updated *sigh.

daidluvkb, 2020-12-24:
> When you say Matlab scripts, do you mean the ones in GPMP2? It seems like this is an issue with GPMP2 showing its age and not being updated *sigh.

Yes,  actually I changed a lower matlab version(R2014b) to complie mexfile,the err I mentioned above disappeared, but when I'm running the gpmp matlab scripts, there are always unstable errors like ' Segmentation violation detected, This error was detected while a MEX-file was running....' 
'Unstable' means the codes caused this error, can be execuate correctly sometime, but at most times matlab reported error info and crashed.  
I dont know what to do, maybe learn from the cpp files directly (it gonna be harder/(ㄒoㄒ)/~~)

ProfFan, 2020-12-24:
I would suggest the following:

1. You can post more detailed reproduction instructions, for example, detailed commands and logs;
2. The message you mentioned is very strange. Actually GTSAM has been tested with MATLAB 2020, so I would say that is almost definitely not GTSAM issue.
3. Maybe you could work with the GPMP2 author to update the dependency to latest GTSAM 4.1?

varunagrawal, 2020-12-24:
Yeah this is not a MATLAB version issue. It definitely seems like an issue with the GPMP2 code not being updated.

The best and fastest way around this would be to convert the Matlab scripts to python.

daidluvkb, 2020-12-24:
> I would suggest the following:
>
>     1. You can post more detailed reproduction instructions, for example, detailed commands and logs;
>
>     2. The message you mentioned is very strange. Actually [...]
Here is the error info printed by matlab. 
> ------------------------------------------------------------------------
>        Segmentation violation detected at Thu Dec 24 14:44:11 2020
> ------------------------------------------------------------------------
> 
> Configuration:
>   Crash Decoding     : Disabled
>   Current Visual     : None
>   Default Encoding   : UTF-8
>   GNU C Library      : 2.27 stable
>   MATLAB Architecture: glnxa64
>   MATLAB Root        : /usr/local/MATLAB/R2014b
>   MATLAB Version     : 8.4.0.150421 (R2014b)
>   Operating System   : Linux 4.15.0-128-generic #131-Ubuntu SMP Wed Dec 9 06:57:35 UTC 2020 x86_64
>   Processor ID       : x86 Family 6 Model 60 Stepping 3, GenuineIntel
>   Software OpenGL    : 0
>   Virtual Machine    : Java 1.7.0_11-b21 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
>   Window System      : No active display
> 
> Fault Count: 1
> 
> 
> Abnormal termination:
> Segmentation violation
> 
> Register State (from fault):
>   RAX = 00007f2dc5162fa0  RBX = 00007f2c80d53df0
>   RCX = 0000000000000000  RDX = 0000000000000000
>   RSP = 00007f2dc5162b40  RBP = 00007f2dc5163230
>   RSI = 00007f2c80d53540  RDI = 00007f2dc5162af8
> 
>    R8 = 00007f2dc5162fe0   R9 = 00007f2dc5162ae0
>   R10 = 00007f2dc5163250  R11 = 00007f2dc5162af8
>   R12 = 0000000000000007  R13 = 0000000000000000
>   R14 = 00007f2d37f35020  R15 = 00007f2dc5163320
> 
>   RIP = 00007f2c9af4e4cc  EFL = 0000000000010207
> 
>    CS = 0033   FS = 0000   GS = 0000
> 
> Stack Trace (from fault):
> [  0] 0x00007f2c9af4e4cc                       /usr/local/lib/libgpmp2.so.0+00525516 _ZNK5gpmp23Arm17forwardKinematicsERKN5Eigen6MatrixIdLin1ELi1ELi0ELin1ELi1EEEN5boost8optionalIS5_EERSt6vectorIN5gtsam5Pose3ESaISB_EENS7_IRS9_INS2_IdLi3ELi1ELi0ELi3ELi1EEESaISF_EEEENS7_IRS9_INS2_IdLin1ELin1ELi0ELin1ELin1EEESaISK_EEEESO_SO_+00001036
> [  1] 0x00007f2c9b2d0798      /usr/local/gtsam_toolbox/gpmp2_wrapper.mexa64+00972696 _ZNK5gpmp217ForwardKinematicsIN5Eigen6MatrixIdLin1ELi1ELi0ELin1ELi1EEES3_E25forwardKinematicsPositionERKS3_+00000216
> [  2] 0x00007f2c9b28ffb7      /usr/local/gtsam_toolbox/gpmp2_wrapper.mexa64+00708535 _Z37gpmp2Arm_forwardKinematicsPosition_35iPP11mxArray_tagiPPKS_+00000183
> [  3] 0x00007f2c9b29cf83      /usr/local/gtsam_toolbox/gpmp2_wrapper.mexa64+00761731 mexFunction+00004627
> [  4] 0x00007f2dd0220c0a     /usr/local/MATLAB/R2014b/bin/glnxa64/libmex.so+00150538 mexRunMexFile+00000090
> [  5] 0x00007f2dd021d5c4     /usr/local/MATLAB/R2014b/bin/glnxa64/libmex.so+00136644
> [  6] 0x00007f2dd021e414     /usr/local/MATLAB/R2014b/bin/glnxa64/libmex.so+00140308
> [  7] 0x00007f2dcf555329 /usr/local/MATLAB/R2014b/bin/glnxa64/libmwm_dispatcher.so+00791337 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000697
> [  8] 0x00007f2dce42fb70 /usr/local/MATLAB/R2014b/bin/glnxa64/libmwm_interpreter.so+04053872
> [  9] 0x00007f2dce3dfe02 /usr/local/MATLAB/R2014b/bin/glnxa64/libmwm_interpreter.so+03726850
> [ 10] 0x00007f2dce3e2022 /usr/local/MATLAB/R2014b/bin/glnxa64/libmwm_interpreter.so+03735586
> [ 11] 0x00007f2dce3e7f87 /usr/local/MATLAB/R2014b/bin/glnxa64/libmwm_interpreter.so+03760007
> [ 12] 0x00007f2dce3e36ff /usr/local/MATLAB/R2014b/bin/glnxa64/libmwm_interpreter.so+03741439
> [ 13] 0x00007f2dce3e4334 /usr/local/MATLAB/R2014b/bin/glnxa64/libmwm_interpreter.so+03744564
> [ 14] 0x00007f2dce45952e /usr/local/MATLAB/R2014b/bin/glnxa64/libmwm_interpreter.so+04224302
> [ 15] 0x00007f2dcf555329 /usr/local/MATLAB/R2014b/bin/glnxa64/libmwm_dispatcher.so+00791337 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000697
> [ 16] 0x00007f2dcf10dca8  /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcos.so+02448552
> [ 17] 0x00007f2dcf0a57e2  /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcos.so+02021346
> [ 18] 0x00007f2dcf0a71ce  /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcos.so+02027982
> [ 19] 0x00007f2dcf0acb70  /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcos.so+02050928
> [ 20] 0x00007f2dcf0a858d  /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcos.so+02033037
> [ 21] 0x00007f2dcf1914d0  /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcos.so+02987216
> [ 22] 0x00007f2dcf500e34 /usr/local/MATLAB/R2014b/bin/glnxa64/libmwm_dispatcher.so+00446004 _ZN13Mfh_MATLAB_fn11dispatch_fhEiPP11mxArray_tagiS2_+00000244
> [ 23] 0x00007f2dcf18edd1  /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcos.so+02977233
[... 52 lines cut ...]
> [ 76] 0x00007f2ddb45917c /usr/local/MATLAB/R2014b/bin/glnxa64/libmwservices.so+02523516
> [ 77] 0x00007f2ddb45507f /usr/local/MATLAB/R2014b/bin/glnxa64/libmwservices.so+02506879
> [ 78] 0x00007f2ddb45a4b5 /usr/local/MATLAB/R2014b/bin/glnxa64/libmwservices.so+02528437
> [ 79] 0x00007f2ddb45a8e7 /usr/local/MATLAB/R2014b/bin/glnxa64/libmwservices.so+02529511
> [ 80] 0x00007f2ddb45afc0 /usr/local/MATLAB/R2014b/bin/glnxa64/libmwservices.so+02531264 _Z25svWS_ProcessPendingEventsiib+00000080
> [ 81] 0x00007f2ddac97248   /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcr.so+00635464
> [ 82] 0x00007f2ddac97564   /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcr.so+00636260
> [ 83] 0x00007f2ddac83cdd   /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcr.so+00556253
> [ 84] 0x00007f2dd9a4e6db              /lib/x86_64-linux-gnu/libpthread.so.0+00030427
[... cut ...]
> for errors. Please consult the External Interfaces Guide for information
> on debugging MEX-files.
> 
> If this problem is reproducible, please submit a Service Request via:
>     http://www.mathworks.com/support/contact_us/
> 
> A technical support engineer might contact you with further information.
> 
> Thank you for your help.** This crash report has been saved to disk as /home/jimeng/matlab_crash_dump.9347-1 **

It does seem like something wrong with gpmp2. I will ask this issue in gpmp2, thanks a lot!

ProfFan, 2020-12-24:
If you demangle the names you will see that this is indeed a problem in GPMP2:

```
> Stack Trace (from fault):
[  0] 0x00007f2c9af4e4cc                       /usr/local/lib/libgpmp2.so.0 00525516 gpmp2::Arm::forwardKinematics(Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, boost::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> const&>, std::vector<gtsam::Pose3, std::allocator<gtsam::Pose3> >&, boost::optional<std::vector<Eigen::Matrix<double, 3, 1, 0, 3, 1>, std::allocator<Eigen::Matrix<double, 3, 1, 0, 3, 1> > >&>, boost::optional<std::vector<Eigen::Matrix<double, -1, -1, 0, -1, -1>, std::allocator<Eigen::Matrix<double, -1, -1, 0, -1, -1> > >&>, boost::optional<std::vector<Eigen::Matrix<double, -1, -1, 0, -1, -1>, std::allocator<Eigen::Matrix<double, -1, -1, 0, -1, -1> > >&>, boost::optional<std::vector<Eigen::Matrix<double, -1, -1, 0, -1, -1>, std::allocator<Eigen::Matrix<double, -1, -1, 0, -1, -1> > >&>) const 00001036
[  1] 0x00007f2c9b2d0798      /usr/local/gtsam_toolbox/gpmp2_wrapper.mexa64 00972696 gpmp2::ForwardKinematics<Eigen::Matrix<double, -1, 1, 0, -1, 1>, Eigen::Matrix<double, -1, 1, 0, -1, 1> >::forwardKinematicsPosition(Eigen::Matrix<double, -1, 1, 0, -1, 1> const&) const 00000216
[  2] 0x00007f2c9b28ffb7      /usr/local/gtsam_toolbox/gpmp2_wrapper.mexa64 00708535 gpmp2Arm_forwardKinematicsPosition_35(int, mxArray_tag**, int, mxArray_tag const**) 00000183
[  3] 0x00007f2c9b29cf83      /usr/local/gtsam_toolbox/gpmp2_wrapper.mexa64 00761731 mexFunction 00004627
[  4] 0x00007f2dd0220c0a     /usr/local/MATLAB/R2014b/bin/glnxa64/libmex.so 00150538 mexRunMexFile 00000090
```

closed 2020-12-24 by ProfFan