borglab/gtsam #638: tbb::interface5::internal::concurrent_unordered_base<>::internal_copy()

hunkyu, 2020-12-18:
## Description

## Steps to reproduce

*** Aborted at 1608279752 (unix time) try "date -d @1608279752" if you are using GNU date ***
PC: @                0x0 (unknown)
0  775 737
1  776 738
2  778 740
3  779 741
4  780 742
5  781 743
6  782 744
7  784 743
8  785 743
9  786 741
10  788 740
11  790 738
12  791 737
13  792 736
14  793 734
15  795 732
16  796 730
17  797 728
18  799 726
19  800 725
20  801 723
21  802 722
22  803 721
23  804 719
24  805 717
25  806 716
26  807 714
27  809 712
28  811 709
29  812 707
30  813 705
31  815 702
32  817 700
33  818 699
34  819 698
35  820 698
36  822 695
37  823 693
*** SIGSEGV (@0x0) received by PID 509 (TID 0x7f898c3010) from PID 0; stack trace: ***
38  824 691
39  826 689
40  828 687
41  829 685
42  831 682
43  832 681
44  833 680
45  835 677
46  836 674
47  837 673
48  838 672
49  839 671
50  840 669
51  841 668
52  842 667
53  843 665
    @       0x7f9bdd86c0 ([vdso]+0x6bf)
    @       0x7f97b7d874 tbb::interface5::internal::concurrent_unordered_base<>::internal_copy()
    @       0x7f97b9a094 gtsam::ISAM2::updateDelta()
    @       0x7f97b9fe2c gtsam::ISAM2::update()
    @       0x7f97b9e144 gtsam::ISAM2::update()

## Expected behavior

In order to improve the real-time performance as the size of the node increases, I manually reset the isam, and then continue to add the error of the node, it is not a segfault at once, and it took a long time to prompt the error

## Environment

gtsam branch  develop
ubuntu 18.04

ProfFan, 2020-12-18:
Could you provide a minimal reproducer?

hunkyu, 2020-12-19:
Sorry, it's my code problem, I didn't do interlocking daozhi

closed 2020-12-19 by hunkyu