borglab/gtsam #639: Python Wrapper Error

dlee640, 2020-12-20:
## Description

**I have ubuntu 18.04, and I have used release 4.1.0 to install gtsam. My C++ version works fine (all the unittests work), but I am more familiar with python, therefore decided to install python wrapper. However, using python 3.6.9, I get an error saying python version of  >= 3.7.x is required. Yet, if I set -DGTSAM_PYTHON_VERSION= to anything past 3.7.x, it throws me an error as well. ** 

## Steps to reproduce my error...

**First attempt: Python 3.6.9 as default on my environment**
1. $ git clone https://github.com/borglab/gtsam.git
2. $ cd gtsam 
3. $ mkdir build
4. $ cd build
5. $ pip install -r <gtsam_folder>/python/requirements.txt   # this is to install pyparsing
6. $ cmake .. -DGTSAM_BUILD_PYTHON=1 -DGTSAM_PYTHON_VERSION=3.6.9
    **Here, it won't let me set python version => 3.7.x. Thus, I used 3.6.9 since the instruction says it has to be >3.6.x...
    This outputs:**
    -- Found Eigen version: 3.3.7
    -- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
    -- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
    -- Found PythonInterp: /usr/bin/python3.6 (found suitable exact version "3.6.9") 
    -- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable exact version "3.6.9") 
    -- pybind11 v2.6.0 dev1
    -- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so
    -- Performing Test HAS_FLTO
    -- Performing Test HAS_FLTO - Success
    -- Building 3rdparty
    -- Looking for execinfo.h
    -- Looking for execinfo.h - found
    -- Looking for getline
    -- Looking for getline - found
    -- checking for thread-local storage - found
    -- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
    -- Building base
    -- Building geometry
    -- Building inference
    -- Building symbolic
    -- Building discrete
    -- Building linear
    -- Building nonlinear
    -- Building sam
    -- Building sfm
    -- Building slam
    -- Building navigation
    -- GTSAM Version: 4.1.0
    -- Install prefix: /usr/local
    -- Building GTSAM - shared: ON
    -- Building base_unstable
    -- Building geometry_unstable
    -- Building linear_unstable
    -- Building discrete_unstable
    -- Building dynamics_unstable
    -- Building nonlinear_unstable
    -- Building slam_unstable
    -- Building partition_unstable
    -- GTSAM_UNSTABLE Version: 4.1.0
    -- Install prefix: /usr/local
    -- Wrote /home/dlee640/gtsam_py/gtsam/build/GTSAMConfig.cmake
    -- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
    -- ===============================================================
    -- ================  Configuration Options  ======================
    --  CMAKE_CXX_COMPILER_ID type                       : GNU
    --  CMAKE_CXX_COMPILER_VERSION                       : 7.5.0
    --  CMake version                                    : 3.10.2
    --  CMake generator                                  : Unix Makefiles
    --  CMake build tool                                 : /usr/bin/make
    -- Build flags                                               
    --  Build Tests                                      : Enabled
    --  Build examples with 'make all'                   : Enabled
    --  Build timing scripts with 'make all'             : Disabled
    --  Build shared GTSAM libraries                     : Enabled
    --  Put build type in library name                   : Enabled
[... 401 lines cut ...]
**If I try to set python to 3.7 (also tried 3.8), it returns following error:**

cmake .. -DGTSAM_BUILD_PYTHON=1 -DGTSAM_PYTHON_VERSION=3.7.9
-- Found Eigen version: 3.3.7
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Found PythonInterp: /usr/bin/python3.7 (found suitable exact version "3.7.9") 
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find PythonLibs: Found unsuitable version "3.6.9", but required
  is exact version "3.7.9" (found /usr/lib/x86_64-linux-gnu/libpython3.6m.so)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:376 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.10/Modules/FindPythonLibs.cmake:262 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  wrap/cmake/PybindWrap.cmake:17 (find_package)
  python/CMakeLists.txt:19 (include)

-- Configuring incomplete, errors occurred!
See also "/home/dlee640/gtsam_py/gtsam/build/CMakeFiles/CMakeOutput.log".
See also "/home/dlee640/gtsam_py/gtsam/build/CMakeFiles/CMakeError.log".

Someone please help. Thanks!

## Expected behavior

Python wrapper was supposed to compile clean without errors.

varunagrawal, 2020-12-23:
1. I don't understand why you are using `sudo make python-install`. `make python-install` (without the sudo) should be sufficient. Nowhere in the instructions do we recommend using sudo when installing the python wrapper.
2. The issue from what I can see, seems to be that you are using numpy 1.20.0rc1 which enforces a requirement for Python 3.7 and greater. I don't understand why you would need 1.20.0 when 1.19 is available on PyPI and works great (unless you don't know how it was installed in the first place). This might be related to your use of `sudo` in the previous point.

dlee640, 2020-12-27:
Dear Varun,

Thank you for your response. I closed the issue regarding python wrapper documentation, since
```
cmake ..
make -j4
make python-install
```
all worked successfully. 

However, upon running unittests for python, I get following error:
```
======================================================================
ERROR: test_logging_optimizer (unittest.loader._FailedTest)
----------------------------------------------------------------------
ImportError: Failed to import test module: test_logging_optimizer
Traceback (most recent call last):
  File "/usr/lib/python3.6/unittest/loader.py", line 428, in _find_test_path
    module = self._get_module_from_name(name)
  File "/usr/lib/python3.6/unittest/loader.py", line 369, in _get_module_from_name
    __import__(name)
  File "/home/dlee640/gtsam/python/gtsam/tests/test_logging_optimizer.py", line 16, in <module>
    import gtsam
  File "/home/dlee640/.local/lib/python3.6/site-packages/gtsam/__init__.py", line 1, in <module>
    from .gtsam import *
ImportError: /home/dlee640/.local/lib/python3.6/site-packages/gtsam/gtsam.so: invalid ELF header
```
I followed instructions on the github,

```
cd <GTSAM_SOURCE_DIRECTORY>/python/gtsam/tests
python -m unittest discover
```

Is there something I am missing from the python installation phase?  Thank you.

varunagrawal, 2020-12-27:
What is the output when you run `sudo python -V`?

dlee640, 2020-12-27:
python 3.6.9

varunagrawal, 2020-12-27:
Looks like something went wrong during your build, most likely due to all the python version switching or something else.

Can you rebuild the entire thing again from scratch following the process that works for you?

varunagrawal, 2020-12-27:
Also, I believe you'll have a faster response by asking this question on the Google group.

dlee640, 2020-12-29:
Varun, thank you for your help. After following the instructions about the HandlePython file and making corrections, my unittests for python binding passed.

Only concern was that during
$ make -j4,
I needed to attempt it 3 times in order for it to compile.

varunagrawal, 2020-12-29:
That's strange. I don't understand why you would need to run make multiple times.

What errors do you get in the initial runs?

dlee640, 2020-12-29:
[image]
This is the screenshot I took when the error occured. As you can see, when I attempted another $ make -j4, it ended up compiling successfully.

varunagrawal, 2020-12-29:
Hmm it seems that you're potentially running out of RAM. I recommend turning off building of examples via ccmake. The wrapper is a real memory hog and we've not figured out the best way to tackle it thus far.

closed 2021-01-05 by varunagrawal