borglab/gtsam #640: Help needed for Matlab/Python Wrapper Instructions

dlee640, 2020-12-21:
## Feature

I am a student who wants to get familiarized with GTSAM python / matlab functionalities.

## Motivation

I have followed the instructions on readme.md and was able to build & run OdometryExample on terminal. My system is Ubuntu 18.04, and I have used release 4.1 on borglab/gtsam page. However, I am not sure how to build/install python and matlab. I tried doing both, but I have messed up my system couple times, and would want to clarify the steps before reattempting.

## Pitch

PLEASE document step-by-step procedure for setting up matlab/python wrappers.

If I already build the gtsam, do I need to rebuild the entire gtsam with new cmake options for allowing matlab and python wrappers? I am confused if instructions for python <https://github.com/borglab/gtsam/tree/release/4.1.0/python> or matlab <https://github.com/borglab/gtsam/tree/release/4.1.0/matlab> needs to be executed within the gtsam/python or gtsam/matlab folder.. meaning, do I have to make a new 'build' folder within each one of them to build them on top of the existing c++ gtsam? or do I need to only build once for all c++/matlab/python to work with gtsam? 

Currently, step-by-step instructions on readme.md only works for c++ version of gtsam. Python and matlab wrapper instructions are unclear.

varunagrawal, 2020-12-23:
Hi @dlee640 
Does [this document](https://github.com/borglab/gtsam/tree/develop/python) answer all your questions?

We always recommend using the latest version of GTSAM, in this case 4.1.0.

varunagrawal, 2020-12-23:
Here are the answers to your questions:

1. If you have already built GTSAM via CMake, when you enable the flags for the wrapper, `CMake` and `Make` are smart enough to compile only the required parts. Thus, they won't re-compile GTSAM from scratch, but you need to run `cmake ..` with the correct flags enabled so that CMake knows what to do.
2. All CMake/Make instructions are to be executed from the same build directory as for C++. This is because the whole compiling and wrapping process is controlled via CMake to make it as streamlined as possible. You do not need to have multiple build directories.
3. If you are using a release, always the latest version since it has various improvements and bugfixes. In this case, it would be 4.1. The best would be to use the `develop` branch if you're using git and the repository directly.
4. The instructions in the links specified give you the instructions for what to type in the terminal. It should be as simple as:
    ```sh
    cmake -DGTSAM_BUILD_PYTHON=ON -DGTSAM_PYTHON_VERSION=Default ..
    make -j4
    make python-install
    ```
5. We support all versions of Python 3.5+. Matlab is a little more tricky since it is closed source and we can't test the latest versions automatically, but upto version 2019b have worked without issues and unless Mathworks changes their Mex compiler significantly (which is highly unlikely), it should work for modern versions of Matlab as well. We, however, recommend using Python as Python (and Swift) will be the supported languages moving forward.
6. If you are encountering issues, please add the error messages to the issue as well. You can paste the errors as a comment below and we can help you debug the problem.

dlee640, 2020-12-27:
Dear Varun,

Thank you very much for your help. I needed sudo for make python-install step, but everything worked.

closed 2020-12-27 by dlee640