borglab/gtsam #648: findExampleDataFiles broken on macOS

ProfFan, 2021-01-01:
## Description

`findExampleDataFile` should work on macOS but it does not.

## Steps to reproduce

1. `pip3 install --user --pre gtsam`
2. try find any data file.

## Expected behavior

Should return the path of data file.

## Environment

macOS Big Sur, Python 3.9.1

## Additional information

Upon inspection the data files were installed to 
```
/Users/proffan/Library/Python/3.9/gtsam/Data/sphere2500.txt
```
not 
```
/Users/proffan/Library/Python/3.9/lib/python/site-packages/gtsam/Data/sphere2500.txt
```

I think this behavior is wrong as this is the "root" of the directory hierarchy.

Furthermore, a non-user install trys to write to `/usr/local/gtsam`, again not correct.

I may be wrong here as I have touched the wheels myself before uploading, though. Need someone to test a from-source install and the wheel to be sure.

ayushbaid, 2021-01-02:
Just FYI: this is an issue on linux too

varunagrawal, 2021-01-02:
This is fixed by #609 :slightly_smiling_face:

PR #649 (2021-01-02): Add dataset directly to package
This PR fixes #648.
closed 2021-01-05 by varunagrawal