borglab/gtsam #658: Logging and Code Coverage

varunagrawal, 2021-01-05:
## Feature

Request to add in:

- [ ] Logging via [glog](https://github.com/google/glog#cmake).
- [ ] Unit test code coverage via either [gcov](https://gcc.gnu.org/onlinedocs/gcc/Gcov.html) or [CodeCoverage](https://github.com/bilke/cmake-modules/blob/master/CodeCoverage.cmake).

## Motivation

Logging will help us print exceptions, warnings and other useful information while at the same time providing a switch which will not impact performance.

Code coverage will ensure we are unit test driven in our development.

## Pitch

This feature will require quite a bit of CMake competency, but it should be straightforward enough. Any user should then be able to add various logging statements throughout the codebase (or via a simple include such as `<gtsam/base/logging.h>`.

The code coverage module will be invoked via a Github Action that will allow us to place a Markdown badge on the README.

## Alternatives

## Additional context

jlblancoc, 2021-02-13:
Have you considered presenting GTSAM to Google Summer of Code to have extra hands helping in this kind of topics? I could give you a hand as a mentor in case you decide to participate.

dellaert, 2021-02-15:
@jlblancoc I have zero cycles :-) If you want to do this, I fully support it! You are a GTSAM primary by now :-)

jlblancoc, 2021-02-16:
Hmm... I could help you with the registration process, etc. since I've been through it in past years. 
However, there is a key page that needs to be ready, by this Friday (Feb 19th): a wiki with the "list of project ideas". 
If you, @varunagrawal , and the other developers want to fill this wiki page that I created based on an older one from our project: 

https://github.com/borglab/gtsam/wiki/Ideas-page-for-GTSAM-Google-Summer-of-Code-2021

Please, confirm whether we can have by the next Friday: 
- a list of potential mentors.
- a list of proposed projects, 

Best,

jlblancoc, 2021-02-16:
PS: project ideas may be based on some pending issues / PRs, or brand new ideas, etc.

dellaert, 2021-02-17:
OK, let me take a look and give a think. Do you have a list of projects you want to mentor?

varunagrawal, 2021-02-19:
I have a bunch of project suggestions. Things like getting the QP solver working, better examples from existing datasets, wrapper features and some old issues that we've basically put on the backburner. I'll update the Wiki page during my lunch break today.

dellaert, 2021-02-19:
It's a bit too late now. Focus on your A->B

varunagrawal, 2021-02-19:
Less work for me, plus I made a breakthrough on the URDF issue I was having so the A->B is progressing quickly.

jlblancoc, 2021-02-19:
Varun: my advice is, try to write down those project ideas in the wiki, in
your spare time, so they will help next year's proposal. But don't let this
distract you from whatever is your key goal now! :-)

Best

varunagrawal, 2021-02-19:
You're right @jlblancoc. I just saw your comments on this issue today morning, so replied in case this was time sensitive. 😊