borglab/gtsam #661: Building with TBB on macOS for "debug" build types

NikolausDemmel, 2021-01-11:
## Description

On macOS (10.15 Catalina) -- with current TBB installed via homebrew -- the GTSAM cmake only picks it up for `Release` build type, but fails to do so for `RelWithDebInfo` and `Debug`. This is with the latest `develop` branch.

## Steps to reproduce

`Release` is ok:

```
cmake .. -DCMAKE_BUILD_TYPE=Release

[...]

-- Found TBB: /usr/local/include (found suitable version "2020.3", minimum required is "4.4") found components: tbb tbbmalloc
```

`Debug` or `RelWithDebInfo` fails:

```
cmake .. -DCMAKE_BUILD_TYPE=Debug

[...]

-- Could NOT find TBB (missing: tbb tbbmalloc) (found suitable version "2020.3", minimum required is "4.4")
```

## Workaround

It seems that the cmake find module wants to find dedicated debug libraries, but they don't seem to exist on macOS, at least not with the homebrew install:

```
$ ls  /usr/local/opt/tbb/lib/
cmake                    libtbb.a                 libtbb.dylib             libtbbmalloc.a           libtbbmalloc.dylib       libtbbmalloc_proxy.dylib python3.9
```

Looks like there is a flag in the find module to turn this off:

```
cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -DTBB_USE_DEBUG_BUILD=OFF
```

Maybe on macOS this should be the default? (I haven't tried other installation methods for TBB next to homebrew.)

## Environment

* macOS Catalina Version 10.15.7 (19H114)
* cmake version 3.18.0
* Homebrew 2.7.2-76-gd75e9c9
* TBB 2020_U3

varunagrawal, 2021-01-12:
Hmm this seems to be a very specific case (though I may be wrong). I am not sure if having GTSAM enforce requiring external libraries to follow the same build type makes any sense. We should be able to build a Debug version of GTSAM while using a Release version of TBB (at least from a performance perspective).

NikolausDemmel, 2021-01-12:
> Hmm this seems to be a very specific case (though I may be wrong). 

Maybe. But I think it is the default (easiest) way to get TBB on macOS these days, so IMHO it's the default setup to compile GTSAM on macOS.

> I am not sure if having GTSAM enforce requiring external libraries to follow the same build type makes any sense.

I don't really care about what libraries it uses (on Linux, where it seems it has both release and debug libraries shipped by TBB), but I agree this is a bit strange in the TBB find module, to enforce this. I'd rather expect it to choose release libraries unless you pass an extra flag saying you want debug libraries for TBB. But I'm not sure if some people would be disturbed if you would change the default on all platforms (i.e. always use release libs for TBB unless a specific flag is passed to cmake).

But besides that, the main point of this issue here is that it would be nice if it would pick up TBB on macOS without requiring the additional cmake flag described in the workaround section above.

(But it's also not very high priority, given that the workaround is quite easy if you know it.)

dellaert, 2021-01-12:
> We should be able to build a Debug version of GTSAM while using a Release version of TBB (at least from a performance perspective).

Agreed!

closed 2021-01-29 by jlblancoc

NikolausDemmel, 2021-01-29:
Awesome, thanks! I verified that now `mkdir build; cd build; cmake ..; make` works as expected and picks up TBB if installed (via homebrew) on macOS.

PR #683 (2021-01-29): Update CMake Variables and TBB build type
1. Update non-user-editable CMake variables to be `INTERNAL`.
2. TBB should be in Release mode by default unless `TBB_USE_DEBUG_BUILD=ON` is specified. This fixes #661.