borglab/gtsam #663: Python wheel CI builds are broken after gtwrap changes

ProfFan, 2021-01-12:
The borglab/gtsam-manylinux-build CI builds are now broken because of the recent changes made to the way `gtwrap` operates.

```
Scanning dependencies of target pybind_wrap_gtsam_unstable
[  0%] Generating gtsam_unstable.cpp
Traceback (most recent call last):
  File "/Users/runner/work/gtsam-manylinux-build/gtsam-manylinux-build/gtsam/wrap/scripts/pybind_wrap.py", line 11, in <module>
    import gtwrap.interface_parser as parser
ModuleNotFoundError: No module named 'gtwrap'
make[2]: *** [python/gtsam_unstable.cpp] Error 1
```

@varunagrawal I think in the PR you said that there will be no behavior changes? Or we need to install `gtwrap` for building GTSAM now? Could you look at the build logs [here](https://github.com/borglab/gtsam-manylinux-build/runs/1684460491?check_suite_focus=true) and help me debug the build? Thanks a lot :)

varunagrawal, 2021-01-12:
Huh that's strange. Sure thing I'll help debug :)

varunagrawal, 2021-01-12:
So what's happening is that `gtwrap` is being installed to Python3.9 rather than 3.8 and that's why it is "missing". Let me do some CMake fixes.

varunagrawal, 2021-01-12:
@ProfFan I am transferring this issue to `gtsam-manylinux-build` since it is an issue there.