borglab/gtsam #667: Can isam perform accelerated processing?

hunkyu, 2021-01-14:
## Feature
Whether the Bayesian tree adjusts the node optimization position so that the optimization can be accelerated

## Motivation

I want to do a stress test to see the node size and the efficiency of isam optimization

## Pitch

When the number of factor nodes is greater than 5000, it contains three factors (betweent facotr gps factor prior factor), and isam update() takes 100ms. Is this normal?

## Alternatives

Whether the Bayesian tree adjusts the node optimization position so that the optimization can be accelerated

## Additional context
[image]

dellaert, 2021-01-14:
There are several mechanisms in iSAM that are governed by parameters. for `calculateEstimate` the wildfire threshold especially might matter to you. I suggest you read the paper thoroughly and examine what exactly makes this slow, and then come up with a concrete proposal if you think it is not yet implemented and will make a real difference.

hunkyu, 2021-01-15:
@dellaert  The iSAM parameter I use is like this   :
gtsam::ISAM2Params parameters;
        parameters.relinearizeThreshold = 0.01;
        parameters.relinearizeSkip = 1;
        gtsam::ISAM2DoglegParams dogleg_param;
        dogleg_param.setVerbose(false);
        parameters.setOptimizationParams(dogleg_param);

dellaert, 2021-01-18:
I suggest you read the paper thoroughly and examine what exactly makes this slow, and then come up with a concrete proposal if you think it is not yet implemented and will make a real difference.

varunagrawal, 2021-03-22:
@hunkyu closing this for now. Please reopen if you'd like to explore this further.

closed 2021-03-25 by hunkyu