borglab/gtsam #694: Example of solving NLS with robust kernels in Python

heethesh, 2021-02-11:
Can I use the Python wrapper of GTSAM to solve simple non-linear least square problems with robust kernels using LM or Gauss-Newton methods with known/manually computed Jacobians and error functions? Is there an example or something similar to help get started?

dellaert, 2021-02-11:
Please also ask in GTSAM user group, but yes: https://github1s.com/borglab/gtsam/blob/HEAD/gtsam/linear/tests/testNoiseModel.cpp

closed 2021-02-11 by dellaert

brunoeducsantos, 2021-06-28:
I tried to add a costumize Jacobian in Python as follows
```
graph.add(gtsam.CustomFactor(icl_map_model, [unknowns[-1]], partial(map_fac.error_map, np.array([gtsam.Pose2(x_final,y_final, phi_final)]),jacobians=[jacobian_map])))
```

where 

```
def error_map(measurement: np.ndarray, this: gtsam.CustomFactor, values, jacobians: Optional[List[np.ndarray]]):
    """Map Factor error function
    :param measurement: ICL constraint, to be filled with `partial`
    :param this: gtsam.CustomFactor handle
    :param values: gtsam.Values
    :param jacobians: Optional list of Jacobians
    :return: the error
    """
    key = this.keys()[0]
    estimate = values.atVector(key)
    t_r= measurement[0].translation()
    print(jacobians)
    t_icl= estimate[:2]
    R= measurement[0].rotation()
    e_t= R.inverse().matrix()@(t_r-t_icl)
    e_phi= estimate[2]- R.theta()
    return np.array([e_t[0],e_t[1],e_phi])
```
And I got the following error ,

` error_map() got multiple values for argument 'jacobians' `

Isn't it possible to have costumize Jacobians ?
Thanks,
Bruno

dellaert, 2021-06-28:
@ProfFan

ProfFan, 2021-06-28:
@brunoeducsantos Note this is not the correct way to use `CustomFactor`.

Look at the example:

```
def error_gps(measurement: np.ndarray, this: gtsam.CustomFactor, values, jacobians: Optional[List[np.ndarray]]):
    """GPS Factor error function
    :param measurement: GPS measurement, to be filled with `partial`
    :param this: gtsam.CustomFactor handle
    :param values: gtsam.Values
    :param jacobians: Optional list of Jacobians
    :return: the unwhitened error
    """
    key = this.keys()[0]
    estimate = values.atVector(key)
    error = estimate - measurement
    if jacobians is not None:
        jacobians[0] = I

    return error
```

`jacobians` are references to the GTSAM C++ list of Jacobians, just as in the C++ factor. They are not to be filled with `partial`, but to be filled by the error function.

brunoeducsantos, 2021-06-29:
Thanks a lot @ProfFan  ! It works well now . 
Thanks for your help!

halajun, 2022-05-19:
> @brunoeducsantos Note this is not the correct way to use `CustomFactor`.
>
> Look at the example:
>
> ```
> def error_gps(measurement: np.ndarray, this: gtsam.CustomFactor, values, jacobians: [...]

Hi @ProfFan, may I ask you how to initialize the jacobians in error function? I tried to do it inside my custom error function, as below:

```
def error_kp_ref(measurement: np.ndarray,
             T_cs: gtsam.Pose3,
             T_cp: gtsam.Pose3,
             this: gtsam.CustomFactor,
             values: gtsam.Values,
             jacobians: Optional[List[np.ndarray]]):
    """Keypoint Factor error function
    :param measurement/T_cs/T_cp: Keypoint measurement, sensor and ping offsets, to be filled with `partial`
    :param this: gtsam.CustomFactor handle
    :param values: gtsam.Values
    :param jacobians: Optional list of Jacobians
    :return: the unwhitened error
    """

    key_1 = this.keys()[0]
    kp_3d = values.atPoint3(key_1)
    S_meas = T_cs.transformTo( T_cp.transformTo(kp_3d) )
    estimate = [np.linalg.norm(S_meas), S_meas[0]]

    jacobians = [np.zeros((2,3))]

    error = estimate - measurement
    if jacobians is not None:
        J_s_kp = T_cs.rotation().inverse()*T_cp.rotation().inverse()
        jacobians[0] = [S_meas@J_s_kp.matrix()/np.linalg.norm(S_meas),[1,0,0]@J_s_kp.matrix()]

    return error
```
but comes with error saying, "A JacobianFactor was attempted to be constructed with a matrix block of
inconsistent dimension.  The JacobianFactor has 2 rows (i.e. length of
error vector) but the provided matrix block has 0 rows." , which I guess is because the jacobian is initialize in the c++ function as NONE.