borglab/gtsam #697: Pinhole camera factor for localization only

jlblancoc, 2021-02-13:
## Feature

Having a factor where the pose of the camera is the only variable, with fixed landmarks and calibration.

## Motivation

Vision-based localization / tracking / etc.

## Additional context

I have already written such a factor by modifying the existing `GenericProjectionFactor`... would it be useful to integrate it into GTSAM? If yes: any preference on its name?

dellaert, 2021-02-16:
There is also one in unstable, and @lucacarlone is creating another one. We could probably benefit from coordinating :-)

jlblancoc, 2021-02-16:
:+1:

In gtsam_unstable, I found `ProjectionFactorPPP` and `ProjectionFactorPPPC`, which estimate *more* variables (body-camera pose), but landmarks are still an unknown too. I'll let Luca to decide whether his proposal in #696 makes this one #697 unnecessary.

closed 2021-05-27 by dellaert