borglab/gtsam #699: Unable to get Python examples to build

soulslicer, 2021-02-17:
## Description

I followed the instructions on https://github.com/borglab/gtsam/tree/develop/python but I am unable to get the python examples to compile

## Steps to reproduce

1. cmake .. -DGTSAM_BUILD_PYTHON=1 -DGTSAM_PYTHON_VERSION=3.5.2
```
-- GTSAM_SOURCE_ROOT_DIR: [/home/raaj/slam/gtsam]
-- Could NOT find MKL (missing:  MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Found Eigen version: 3.3.7
-- Building 3rdparty
-- checking for thread-local storage - found
-- Could NOT find GeographicLib (missing:  GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building smart
-- Building navigation
-- GTSAM Version: 4.0.3
-- Install prefix: /usr/local
-- Building GTSAM - shared: ON
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.0.3
-- Install prefix: /usr/local
-- Wrote /home/raaj/slam/gtsam/build/GTSAMConfig.cmake
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- ===============================================================
-- ================  Configuration Options  ======================
--   CMAKE_CXX_COMPILER_ID type     : GNU
--   CMAKE_CXX_COMPILER_VERSION     : 5.4.0
--   CMake version                  : 3.5.1
--   CMake generator                : Unix Makefiles
--   CMake build tool               : /usr/bin/make
-- Build flags                                               
--   Build Tests                    : Enabled
--   Build examples with 'make all' : Enabled
--   Build timing scripts with 'make all': Disabled
--   Build shared GTSAM libraries   : Enabled
--   Put build type in library name : Enabled
--   Build libgtsam_unstable        : Enabled
--   Build for native architecture  : Enabled
--   Build type                     : Release
--   C compilation flags            :  -O3 -DNDEBUG
--   C++ compilation flags          :  -Wno-misleading-indentation -O3 -DNDEBUG
--  GTSAM_COMPILE_FEATURES_PUBLIC                    : 
--  GTSAM_COMPILE_OPTIONS_PRIVATE                    : -Wall;-fPIC;$<$<CONFIG:Debug>:-g;-fno-inline>;$<$<CONFIG:Release>:-O3>;$<$<CONFIG:Timing>:-g;-O3>;$<$<CONFIG:Profiling>:-O3>;$<$<CONFIG:RelWithDebInfo>:-g;-O3>;-Wno-unused-local-typedefs
--  GTSAM_COMPILE_OPTIONS_PUBLIC                     : $<$<COMPILE_LANGUAGE:CXX>:-std=c++11>;-march=native
--  GTSAM_COMPILE_DEFINITIONS_PRIVATE                : $<$<CONFIG:Debug>:_DEBUG;EIGEN_INITIALIZE_MATRICES_BY_NAN>;$<$<CONFIG:Release>:NDEBUG>;$<$<CONFIG:Timing>:NDEBUG;ENABLE_TIMING>;$<$<CONFIG:Profiling>:NDEBUG>;$<$<CONFIG:RelWithDebInfo>:NDEBUG>
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC                 : 
--  GTSAM_COMPILE_OPTIONS_PRIVATE_DEBUG              : -g;-fno-inline
--  GTSAM_COMPILE_OPTIONS_PUBLIC_DEBUG               : 
--  GTSAM_COMPILE_DEFINITIONS_PRIVATE_DEBUG          : _DEBUG;EIGEN_INITIALIZE_MATRICES_BY_NAN
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_DEBUG           : 
--  GTSAM_COMPILE_OPTIONS_PRIVATE_RELEASE            : -O3
--  GTSAM_COMPILE_OPTIONS_PUBLIC_RELEASE             : 
--  GTSAM_COMPILE_DEFINITIONS_PRIVATE_RELEASE        : NDEBUG
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_RELEASE         : 
--  GTSAM_COMPILE_OPTIONS_PRIVATE_TIMING             : -g;-O3
--  GTSAM_COMPILE_OPTIONS_PUBLIC_TIMING              : 
--  GTSAM_COMPILE_DEFINITIONS_PRIVATE_TIMING         : NDEBUG;ENABLE_TIMING
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_TIMING          : 
--  GTSAM_COMPILE_OPTIONS_PRIVATE_PROFILING          : -O3
--  GTSAM_COMPILE_OPTIONS_PUBLIC_PROFILING           : 
--  GTSAM_COMPILE_DEFINITIONS_PRIVATE_PROFILING      : NDEBUG
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_PROFILING       : 
--  GTSAM_COMPILE_OPTIONS_PRIVATE_RELWITHDEBINFO     : -g;-O3
--  GTSAM_COMPILE_OPTIONS_PUBLIC_RELWITHDEBINFO      : 
--  GTSAM_COMPILE_DEFINITIONS_PRIVATE_RELWITHDEBINFO : NDEBUG
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_RELWITHDEBINFO  : 
--  GTSAM_COMPILE_OPTIONS_PRIVATE_MINSIZEREL         : 
--  GTSAM_COMPILE_OPTIONS_PUBLIC_MINSIZEREL          : 
--  GTSAM_COMPILE_DEFINITIONS_PRIVATE_MINSIZEREL     : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_MINSIZEREL      : 
--   Use System Eigen                : OFF (Using version: 3.3.7)
--   Use Intel TBB                   : Yes
--   Eigen will use MKL              : MKL not found
--   Eigen will use MKL and OpenMP   : OpenMP found but GTSAM_WITH_EIGEN_MKL is disabled
--   Default allocator               : TBB
--   Cheirality exceptions enabled   : YES
--   Build with ccache               : No
-- Packaging flags                                               
--   CPack Source Generator          : TGZ
--   CPack Generator                 : TGZ
-- GTSAM flags                                               
--   Quaternions as default Rot3     : Disabled
--   Runtime consistency checking    : Disabled
--   Rot3 retract is full ExpMap     : Enabled
--   Pose3 retract is full ExpMap    : Enabled
--   Deprecated in GTSAM 4 allowed   : Enabled
--   Point3 is typedef to Vector3    : Disabled
--   Metis-based Nested Dissection   : Enabled
--   Use tangent-space preintegration: Enabled
--   Build Wrap                      : Enabled
-- MATLAB toolbox flags                                      
--   Install MATLAB toolbox          : Disabled
-- Cython toolbox flags                                      
--   Install Cython toolbox          : Disabled
-- ===============================================================
-- Configuring done
-- Generating done
-- Build files have been written to: /home/raaj/slam/gtsam/build
```
2. make python-install
```
make: *** No rule to make target 'python-install'.  Stop.
```

What am i doing wrong here? I am on Ubuntu 16.04 using system Python 3.5.2

soulslicer, 2021-02-17:
@varunagrawal Would like your help please

varunagrawal, 2021-02-17:
Seems like you're using a very old version of GTSAM? The Cmake output says "CYTHON_TOOLBOX" which we deprecated last summer.

soulslicer, 2021-02-17:
I am using the following branch:

`* (HEAD detached at 4.0.3)`

Should i switch to master?

soulslicer, 2021-02-17:
I pulled master and recompiled and i get the same error:

```
raaj@raaj:~/slam/gtsam/build$ cmake .. -DGTSAM_BUILD_PYTHON=1 -DGTSAM_PYTHON_VERSION=3.5.2
-- GTSAM_SOURCE_ROOT_DIR: [/home/raaj/slam/gtsam]
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   serialization
--   system
--   filesystem
--   thread
--   program_options
--   date_time
--   timer
--   chrono
--   regex
--   atomic
-- GTSAM_BOOST_LIBRARIES: optimized;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;optimized;/usr/lib/x86_64-linux-gnu/libboost_system.so;optimized;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;optimized;/usr/lib/x86_64-linux-gnu/libboost_thread.so;optimized;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;optimized;/usr/lib/x86_64-linux-gnu/libboost_regex.so;debug;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;debug;/usr/lib/x86_64-linux-gnu/libboost_system.so;debug;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;debug;/usr/lib/x86_64-linux-gnu/libboost_thread.so;debug;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;debug;/usr/lib/x86_64-linux-gnu/libboost_regex.so
Ignoring Boost restriction on optional lvalue assignment from rvalues
-- Could NOT find MKL (missing:  MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Found Eigen version: 3.3.7
-- Building 3rdparty
-- checking for thread-local storage - found
-- Could NOT find GeographicLib (missing:  GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building smart
-- Building navigation
-- GTSAM Version: 4.0.2
-- Install prefix: /usr/local
-- Building GTSAM - shared: ON
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.0.2
-- Install prefix: /usr/local
-- Wrote /home/raaj/slam/gtsam/build/GTSAMConfig.cmake
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- ===============================================================
-- ================  Configuration Options  ======================
--   CMAKE_CXX_COMPILER_ID type     : GNU
--   CMAKE_CXX_COMPILER_VERSION     : 5.4.0
--   CMake version                  : 3.5.1
--   CMake generator                : Unix Makefiles
--   CMake build tool               : /usr/bin/make
-- Build flags                                               
--   Build Tests                    : Enabled
--   Build examples with 'make all' : Enabled
--   Build timing scripts with 'make all': Disabled
--   Build shared GTSAM libraries   : Enabled
--   Put build type in library name : Enabled
--   Build libgtsam_unstable        : Enabled
--   Build for native architecture  : Enabled
--   Build type                     : Release
--   C compilation flags            :  -O3 -DNDEBUG
--   C++ compilation flags          :  -O3 -DNDEBUG
--  GTSAM_COMPILE_FEATURES_PUBLIC                    : 
[... 35 lines cut ...]
--   CPack Source Generator         : TGZ
--   CPack Generator                : TGZ
-- GTSAM flags                                               
--   Quaternions as default Rot3     : Disabled
--   Runtime consistency checking    : Disabled
--   Rot3 retract is full ExpMap     : Enabled
--   Pose3 retract is full ExpMap    : Enabled
--   Deprecated in GTSAM 4 allowed   : Enabled
--   Point3 is typedef to Vector3    : Disabled
--   Metis-based Nested Dissection   : Enabled
--   Use tangent-space preintegration: Enabled
--   Build Wrap                     : Enabled
-- MATLAB toolbox flags                                      
--   Install matlab toolbox         : Disabled
-- Cython toolbox flags                                      
--   Install Cython toolbox         : Disabled
-- ===============================================================
-- Configuring done
-- Generating done
-- Build files have been written to: /home/raaj/slam/gtsam/build
```

```
make: *** No rule to make target 'python-install'.  Stop.
```

varunagrawal, 2021-02-17:
You should be using the `develop` branch.

varunagrawal, 2021-02-19:
@soulslicer does this work for you now?

mmattamala, 2021-02-23:
@varunagrawal  I faced a similar issue today while trying to build the python wrappers using the Docker instructions.

I realized that the problem seems to be caused by the last commit in the `wrap` package (8 days ago) and older CMake versions, since the `install` command used [here](https://github.com/borglab/gtsam/blob/bb479c2e387ed840a2aecd7c5a50f9859db08e48/wrap/CMakeLists.txt#L36) is using `TYPE` instead of `DESTINATION`, which was noted in the [previous commit ](https://github.com/borglab/gtsam/commit/bb479c2e387ed840a2aecd7c5a50f9859db08e48#diff-30be3762636c7b587b5dce1127719d72fd987982e243edb7d7cad6ff9076828eL31).

I manually replaced the line and I was able to compile it but I'm not sure if other problems due to the changes in that commit exist.

varunagrawal, 2021-02-23:
@ProfFan this was my comment on the review for your PR. 🙂

Should we revert to DESTINATION? I have no time right now so I can look at this next week.

ProfFan, 2021-02-24:
I think I have made that change but probably it's not synced? I am busy with IROS but can look into this later.

varunagrawal, 2021-03-08:
@ProfFan if you have the changes ready, can you please push them? I'll take care of the rest of this PR.

ProfFan, 2021-03-09:
I don't...

closed 2021-03-14 by varunagrawal