borglab/gtsam #705: Program Crashed using quaternions

klissanzmp, 2021-03-04:
Hi,
I'm trying to use IMU factor to make predictions of robot poses and I want to enable quaternions.
I added `#define GTSAM_USE_QUATERNIONS` in `config.h` under gtsam folder, but my program crashed, displaying the following information:
```
*** stack smashing detected ***: terminated
Aborted (core dumped)
```
The data i'm using is euroc MH_01_esay. The code i'm using is `ImuFactorsExample.cpp`. I modified the code to remove GPS factors. I was wondering if I was using gtsam correctly or is it a bug? Sorry for bothering.
Thank you.

varunagrawal, 2021-03-08:
@klissanzmp can you please share your changes? I am able to run the vanilla program with `#define GTSAM_USE_QUATERNIONS` no problem. If I can reproduce this issue, I can help debug and fix it.

klissanzmp, 2021-03-09:
@varunagrawal Yes. The code is as follows. Basically, i removed GPS factor and read imu measurements from a file as the example did. Thank you. 

```
// GTSAM related includes.
#include <gtsam/inference/Symbol.h>
#include <gtsam/navigation/CombinedImuFactor.h>
#include <gtsam/navigation/ImuFactor.h>
#include <gtsam/nonlinear/LevenbergMarquardtOptimizer.h>
#include <gtsam/nonlinear/NonlinearFactorGraph.h>
#include <gtsam/slam/BetweenFactor.h>
#include <gtsam/slam/dataset.h>

#include <cstring>
#include <fstream>
#include <iostream>

#define INTERVAL 1

#define accel_noise_sigma 0.002
#define gyro_noise_sigma 0.00016968
#define accel_bias_rw_sigma 0.003
#define gyro_bias_rw_sigma 0.000019393

using namespace gtsam;
using namespace std;

using symbol_shorthand::B;  // Bias  (ax,ay,az,gx,gy,gz)
using symbol_shorthand::V;  // Vel   (xdot,ydot,zdot)
using symbol_shorthand::X;  // Pose3 (x,y,z,r,p,y)

boost::shared_ptr<PreintegratedCombinedMeasurements::Params> imuParams() {
  // We use the sensor specs to build the noise model for the IMU factor.
//   double accel_noise_sigma = 0.002;
//   double gyro_noise_sigma = 0.00016968;
//   double accel_bias_rw_sigma = 0.003;
//   double gyro_bias_rw_sigma = 0.000019393;
  Matrix33 measured_acc_cov = I_3x3 * pow(accel_noise_sigma, 2);
  Matrix33 measured_omega_cov = I_3x3 * pow(gyro_noise_sigma, 2);
  Matrix33 integration_error_cov =
      I_3x3 * 1e-8;  // error committed in integrating position from velocities
  Matrix33 bias_acc_cov = I_3x3 * pow(accel_bias_rw_sigma, 2);
  Matrix33 bias_omega_cov = I_3x3 * pow(gyro_bias_rw_sigma, 2);
  Matrix66 bias_acc_omega_int =
      I_6x6 * 1e-5;  // error in the bias used for preintegration

  auto p = PreintegratedCombinedMeasurements::Params::MakeSharedD(9.81);
  // PreintegrationBase params:
  p->accelerometerCovariance =
      measured_acc_cov;  // acc white noise in continuous
  p->integrationCovariance =
      integration_error_cov;  // integration uncertainty continuous
  // should be using 2nd order integration
  // PreintegratedRotation params:
  p->gyroscopeCovariance =
      measured_omega_cov;  // gyro white noise in continuous
  // PreintegrationCombinedMeasurements params:
  p->biasAccCovariance = bias_acc_cov;      // acc bias in continuous
  p->biasOmegaCovariance = bias_omega_cov;  // gyro bias in continuous
  p->biasAccOmegaInt = bias_acc_omega_int;

  return p;
}

int main(int argc, char* argv[]) {
    string imu_file_name(argv[1]), output_filename = "result.txt";

    FILE* fp_out = fopen(output_filename.c_str(), "w+");
    ifstream file(imu_file_name.c_str());
    string value;

[... 82 lines cut ...]
            initial_values.insert(B(correction_count), prev_bias);

            Values result;
            LevenbergMarquardtOptimizer optimizer(*graph, initial_values);
            result = optimizer.optimize();
            // Overwrite the beginning of the preintegration for the next step.
            prev_state = NavState(result.at<Pose3>(X(correction_count)),
                            result.at<Vector3>(V(correction_count)));
            prev_bias = result.at<imuBias::ConstantBias>(B(correction_count));
            // Reset the preintegration object.
            preintegrated->resetIntegrationAndSetBias(prev_bias);

            Vector3 gtsam_position = prev_state.pose().translation();
            Quaternion gtsam_quat = prev_state.pose().rotation().toQuaternion();
            fprintf(fp_out, "%s,%f,%f,%f,%f,%f,%f,%f\n",
                time.c_str(), gtsam_position(0), gtsam_position(1),
                gtsam_position(2), gtsam_quat.x(), gtsam_quat.y(), gtsam_quat.z(),
                gtsam_quat.w());
        }
    }
    fclose(fp_out);
    cout << "Done" << endl;
    return 0;
}
```

ProfFan, 2021-03-09:
wait, we have fprintf in our examples? @klissanzmp Could you run valgrind on your code?

varunagrawal, 2021-03-11:
@klissanzmp I am not able to reproduce the error. I advise you to set `GTSAM_USE_QUATERNIONS` in the CMake options and perhaps try again?

ProfFan, 2021-03-11:
No wonder this will crash, as GTSAM is not compiled to use quaternions. You cannot just change `config.h` and pretend :)

closed 2021-03-11 by ProfFan

NickPerezCarletonUniversity, 2022-01-19:
@ProfFan where and how are you supposed to set this option? I don't see it in 'CMakeLists.txt'