borglab/gtsam #710: How to create symbol keys in a loop?

mc13756, 2021-03-10:
Hello. I'm new to gtsam and I'm trying to use it on a mobile robot application. I create a new odometry factor whenever the robot covers a certain distance. For every new node I want to create a symbol key (but each time with a different number). I don't want to use only integers since I'll use several types on factors in the future and it will be harder for the user to read. Here is a simplified version of my code:

```
#include <gtsam/geometry/Pose2.h>
#include <gtsam/slam/BetweenFactor.h>
#include <gtsam/nonlinear/NonlinearFactorGraph.h>
#include <gtsam/nonlinear/LevenbergMarquardtOptimizer.h>
#include <gtsam/nonlinear/Marginals.h>
#include <gtsam/nonlinear/Values.h>
#include <fstream>
#include <gtsam/inference/Symbol.h>

int prev_pose_num = 1;

BContainer odometry_mesurements;
NonlinearFactorGraph graph;
Values initial;
Values result;

void odometry_callback(...............){

    static Symbol x_prev('x',prev_pose_num);

    if (first_odom_msg){
        Pose2 priorMean(current_odom[0], current_odom[1], current_odom[2]); 
        auto priorNoise = noiseModel::Diagonal::Sigmas(Vector3(0.1, 0.1, 0.1));
        graph.addPrior(x_prev, priorMean, priorNoise);
        initial.insert(x_prev, current_odom_1);
    }
    else if( ............odometry condition.........){
        int next_pose_num = prev_pose_num + 1;
        static Symbol x_prev('x',prev_pose_num);
        static Symbol x_next('x',next_pose_num);
        graph.emplace_shared<BetweenFactor<Pose2> >(x_prev, x_next, odometry, odometryNoise);

        initial.insert(x_next, current_odom_1); // the current odometry is used
        initial.print("\nInitial Estimate:\n");  // print

        optimize_graph(graph, initial);
        // Visualize the results
        visualize_results(result);

        prev_pose_num++;
    }
}
```

When I run my code I get the following error:

```
terminate called after throwing an instance of 'gtsam::ValuesKeyDoesNotExist'
  what():  Attempting to retrieve the key "1", which does not exist in the Values.
Aborted (core dumped)
```
How can I fix it and how can I create a new symbol key for each new node without affecting the previous key?
Thanks in advance.

varunagrawal, 2021-03-10:
It seems like your `first_odom_msg` condition isn't being satisfied, which is why `initial.insert(x_prev, current_odom_1);` is not being called, thus explaining your error. What you are doing is correct otherwise and you can check for existing keys with `initial.exists(keyX)`.

This is also more of a question for the GTSAM Google Group. You would get a lot more help asking over there. For this reason, I am closing this.

closed 2021-03-10 by varunagrawal

mc13756, 2021-03-11:
I found my problem. I was trying to extract double values from the results with a for loop (variable keys as the counter), but this cannot work with the symbol keys. Thanks!