borglab/gtsam #722: crash

AndreV84, 2021-03-22:
any insights?

```
[ERROR] [1616434537.293689297]: /tagslam: 
Indeterminant linear system detected while working near variable
126 (Symbol: 126).
Thrown when a linear system is ill-posed.  The most common cause for this
error is having underconstrained variables.  Mathematically, the system is
underdetermined.  See the GTSAM Doxygen documentation at
http://borg.cc.gatech.edu/ on gtsam::IndeterminantLinearSystemException for
more information.
[tagslam-1] process has died [pid 4665, exit code 255, cmd /home/imp/tagslam_root/devel/lib/tagslam/tagslam_node __name:=tagslam __log:=/home/imp/.ros/log/f7605034-8b1e-11eb-8950-48b02d0785f5/tagslam-1.log].

varunagrawal, 2021-03-24:
Can you please share the code you're using so we can reproduce? It's really hard to figure out what's happening from just this message.

In general, what this means is that your factor graph is under-constrained and/or singular, which leads to the Jacobian being singular and hence the optimization fails. You might be well served to include some priors or additional factors to constrain your problem.

varunagrawal, 2021-04-16:
Closing due to lack of response. If this is a code issue, please feel free to reopen, else asking on the Google Group might be better.

closed 2021-04-16 by varunagrawal