borglab/gtsam #725: Matlab wrapper fails to compile on develop due to header inclusion gtwarp vs wrap

carlmorgan, 2021-03-24:
### Summary
Generated code in target_build/wrap/gtsam/gtsam_wrapper.cpp refers to gtsam/matlab.h and not the wrap/matlab.h location and thus fails to find during compilation.

[ To be researched further - it appears the matlab test suite is also stated failing between 4.0.2 and 4.0.3) - treating as a seperate issue ]

### Environment
Ubuntu Focal, Matlab 2020b Update1 based.  develop branch - 74e4fc392 (19Mar2021)
(Running in CI and local desktop). Not seen on 4.0.3 based builds)

### To reproduce
```cmake
-DGTSAM_USE_SYSTEM_EIGEN:BOOLEAN=FALSE -DGTSAM_WITH_EIGEN_MKL:BOOLEAN=FALSE -DGTSAM_WITH_TBB:BOOLEAN=FALSE -DGTSAM_BUILD_EXAMPLES_ALWAYS:BOOL=FALSE -DCMAKE_BUILD_TYPE:STRING=Release -DGTSAM_BUILD_UNSTABLE:OPTION=OFF -DGTSAM_INSTALL_MATLAB_TOOLBOX:BOOLEAN=true -DMEX_COMMAND:STRING=/usr/local/MATLAB/R2020b/bin/mex
```

```
Scanning dependencies of target gtsam_matlab_wrapper
make[2]: Leaving directory '<HOMEPATH>gtsam/target_build'
make -f matlab/CMakeFiles/gtsam_matlab_wrapper.dir/build.make matlab/CMakeFiles/gtsam_matlab_wrapper.dir/build
make[2]: Entering directory '<HOMEPATH>gtsam/target_build'
[100%] Building CXX object matlab/CMakeFiles/gtsam_matlab_wrapper.dir/__/wrap/gtsam/gtsam_wrapper.cpp.o
cd <HOMEPATH>gtsam/target_build/matlab && /usr/bin/c++  -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SERIALIZATION_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBOOST_TIMER_DYN_LINK -Dgtsam_matlab_wrapper_EXPORTS -I<HOMEPATH>gtsam/gtsam/3rdparty/metis/include -I<HOMEPATH>gtsam/gtsam/3rdparty/metis/libmetis -I<HOMEPATH>gtsam/gtsam/3rdparty/metis/GKlib -I<HOMEPATH>gtsam -I<HOMEPATH>gtsam/target_build -I<HOMEPATH>gtsam/CppUnitLite -I<HOMEPATH>gtsam/gtsam/3rdparty/Eigen -isystem <HOMEPATH>gtsam/gtsam/3rdparty/SuiteSparse_config -isystem <HOMEPATH>gtsam/gtsam/3rdparty/Spectra -isystem <HOMEPATH>gtsam/gtsam/3rdparty/CCOLAMD/Include  -DEIGEN_DONT_VECTORIZE -DEIGEN_DISABLE_UNALIGNED_ARRAY_ASSERT -O3 -DNDEBUG -fPIC      "-I/usr/local/MATLAB/R2020b/extern/include" -DMATLAB_MEX_FILE -DMX_COMPAT_32 -march=native -o CMakeFiles/gtsam_matlab_wrapper.dir/__/wrap/gtsam/gtsam_wrapper.cpp.o -c <HOMEPATH>gtsam/target_build/wrap/gtsam/gtsam_wrapper.cpp
<HOMEPATH>gtsam/target_build/wrap/gtsam/gtsam_wrapper.cpp:1:10: fatal error: gtwrap/matlab.h: No such file or directory
    1 | #include <gtwrap/matlab.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [matlab/CMakeFiles/gtsam_matlab_wrapper.dir/build.make:77: matlab/CMakeFiles/gtsam_matlab_wrapper.dir/__/wrap/gtsam/gtsam_wrapper.cpp.o] Error 1
make[2]: Leaving directory '<HOMEPATH>gtsam/target_build'
make[1]: *** [CMakeFiles/Makefile2:23352: matlab/CMakeFiles/gtsam_matlab_wrapper.dir/all] Error 2
make[1]: Leaving directory '<HOMEPATH>gtsam/target_build'
make: *** [Makefile:163: all] Error 2
```

### Additional testing
I have run / pulled through the latest **update_wrap.sh** versions, no change.

### Simplistic fix / work around
Without understanding the subtleties of the gtwrap / wrap / old / new methods of the GTSam / Matlab wrapping, a change to the generator code, does enable completion of compilation and functional wrapper

```
diff --git a/wrap/gtwrap/matlab_wrapper.py b/wrap/gtwrap/matlab_wrapper.py
index aeaf221bd..b2d8dfe2d 100755
--- a/wrap/gtwrap/matlab_wrapper.py
+++ b/wrap/gtwrap/matlab_wrapper.py
@@ -591,7 +591,7 @@ class MatlabWrapper(object):
         file_name = self._wrapper_name() + '.cpp'

         wrapper_file = textwrap.dedent('''\
-            # include <gtwrap/matlab.h>
+            # include <wrap/matlab.h>
             # include <map>
         ''')

@@ -1476,7 +1476,7 @@ class MatlabWrapper(object):
         """Generate the c++ wrapper."""
         # Includes
         wrapper_file = textwrap.dedent('''\
-            #include <gtwrap/matlab.h>
+            #include <wrap/matlab.h>
             #include <map>
             #include <boost/archive/text_iarchive.hpp>
             #include <boost/archive/text_oarchive.hpp>
```

ProfFan, 2021-03-24:
@varunagrawal

carlmorgan, 2021-03-24:
Merged the PR for fix/wrapper-update ( 77a0fa476 )

Different errors

```[100%] Generating ../wrap/gtsam/gtsam_wrapper.cpp
cd <HOMEPATH>/gtsam/target_build/wrap/gtsam && /usr/bin/cmake -E env PYTHONPATH=<HOMEPATH>/gtsam/wrap /usr/bin/python3.8 <HOMEPATH>/gtsam/wrap/scripts/matlab_wrap.py --src <HOMEPATH>/gtsam/gtsam/gtsam.i --module_name gtsam --out <HOMEPATH>/gtsam/target_build/wrap/gtsam --top_module_namespaces gtsam --ignore gtsam::Point2 gtsam::Point3
Traceback (most recent call last):
  File "<HOMEPATH>/gtsam/wrap/scripts/matlab_wrap.py", line 54, in <module>
    module = parser.Module.parseString(content)
  File "<HOMEPATH>/gtsam/wrap/gtwrap/interface_parser.py", line 945, in parseString
    return Module.rule.parseString(s)[0]
  File "<HOME>/.local/lib/python3.8/site-packages/pyparsing.py", line 1955, in parseString
    raise exc
  File "<HOME>/.local/lib/python3.8/site-packages/pyparsing.py", line 3814, in parseImpl
    raise ParseException(instring, loc, self.errmsg, self)
pyparsing.ParseException: Expected stringEnd, found 'n'  (at char 276), (line:11, col:1)
make[2]: *** [matlab/CMakeFiles/gtsam_matlab_wrapper.dir/build.make:72: wrap/gtsam/gtsam_wrapper.cpp] Error 1
make[2]: Leaving directory '<HOMEPATH>/gtsam/target_build'
make[1]: *** [CMakeFiles/Makefile2:23351: matlab/CMakeFiles/gtsam_matlab_wrapper.dir/all] Error 2
make[1]: Leaving directory '<HOMEPATH>/gtsam/target_build'
make: *** [Makefile:163: all] Error 2```

varunagrawal, 2021-03-24:
@carlmorgan thanks for raising the issue. I am actively working on refactoring the wrapper and I'll have a PR ready in a bit. Please bear with us.

carlmorgan, 2021-03-24:
Cool - looks like major surgery - we are ok on 4.0.2 for the moment and can provide a good test case for develop with the same (and additional internal) matlab wrappers, all on 2020Bu1 / Focal.  

If different flavours of Ubuntu / Matlab are required let us know and I'll see what can be tested here.

varunagrawal, 2021-03-24:
Awesome! I'll have the PR ready in less than an hour, so if you can checkout the corresponding branch and test the bejesus out of it, I'll be a happy dev.

carlmorgan, 2021-03-25:
Resolved with 30c84eabe

closed 2021-03-25 by carlmorgan